realsenseai/librealsense #7066: I515 color camera ov2740 rolling shutter readout time

dysenustc, 2020-08-08:
* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { I515 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (10)  | 
| Kernel Version (Linux Only)    |                                                               | 
| Platform                                 | PC  |
| SDK Version                           |                          | 
| Language                               |                           | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
where can i get I515 color camera ov2740 rolling shutter readout time?

RealSenseSupport, 2020-08-10:
Hi,

You can query timestamp information as part of the frame's metadata.  See:

https://github.com/IntelRealSense/librealsense/blob/master/doc/frame_metadata.md

as well as comments for get_timestamp() here:

https://github.com/IntelRealSense/librealsense/blob/f7cdf6e8961e1709f6d864bdb33095c00a671ca7/include/librealsense2/hpp/rs_frame.hpp#L446

Thanks

dysenustc, 2020-08-16:
OK , thank you

dysenustc, 2020-08-16:
> Hi,
>
> You can query timestamp information as part of the frame's metadata. See:
>
> https://github.com/IntelRealSense/librealsense/blob/master/doc/frame_metadata.md
>
> as well as comments for [...]

dysenustc, 2020-08-16:
OK, thank you

closed 2020-08-16 by dysenustc