realsenseai/librealsense #7068: Unity error when using 515

chrisrogers3d, 2020-08-09:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | l515| 
| Firmware Version                   | 1.4.1.2 | 
| Operating System & Version |  win10  | 
| SDK Version                            | release, master branch, 50515 branch                   | 
| Language                            |   unity }                          | 
| Segment			|   VR  }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
I got 2.36, the 2.37 prerelease, built master from scratch, built 50_515 and for all of them in unity i get this error:

ExternalException: rs2_pipeline_start_with_config(pipe:000001B845E78650, config:000001B845E784B0)
Rethrow as Exception: Couldn't resolve requests
Intel.RealSense.ErrorMarshaler.MarshalNativeToManaged (System.IntPtr pNativeData) (at <782525f085c64debb7028293360b3d39>:0)
Intel.RealSense.Pipeline.Start (Intel.RealSense.Config cfg) (at <782525f085c64debb7028293360b3d39>:0)
RsDevice.OnEnable () (at Assets/RealSenseSDK2.0/Scripts/RsDevice.cs:71)

Several other people have posted similar issues in the last 2 weeks.
Ive spent a lot of time trying to get this to work because other comments here implies that it works

dorodnic, 2020-08-09:
Hi @chrisrogers3d 
As far as I know, "Couldn't resolve requests" should be resolved in 2.37, tested with Unity 2017.
Are you using the Unity package from 2.37 release? Perhaps there is other realsense2.dll somewhere in system path?

crogersarvr, 2020-08-28:
Hey @dorodnic Whats your best advice on getting this to work with Unity 2019.4?  i can compile a different github branch if i need to, I havent tried to actually debug the code other than that screenshot.  I am chrisrogers3d (personal account that started this thread) and now I am trying to get this to work for a work project (facebook reality labs). thanks!

chrisrogers3d, 2020-08-30:
This all works for me with the latest 2.38.1 update.
Thanks!

closed 2020-08-30 by chrisrogers3d