realsenseai/librealsense #7077: Is there any way to accelerate the spatial filter process time? Like gpu implementation.

tarekmuallim, 2020-08-10:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   |      | 
| Operating System & Version | Linux  | 
| Kernel Version (Linux Only)    |                         | 
| Platform                                 | Up board/ NVIDIA Jetson nano   |
| SDK Version                            |      2.36.0                     | 
| Language                            |  C++                           | 
| Segment			|                           | 

### Issue Description
I am trying to use a single board computer with librealsense sdk. such as Intel up board or NVIDIA Jetson Nano.
The problem is the spatial filter process `rs2::spatial_filter` time is long. On those platforms it can takes time more than 120 ms depending on the filter setting.

I have checked `rs2::gl` and I already use is to accelerate point cloud generation.
but there is no gpu accelerate for the spatial filter process.

If there is any acceleration could you lead me to it please.

MartyG-RealSense, 2020-08-10:
Hi @tarekmuallim   The official RealSense store stocks a USB device called the **Intel Neural Compute Stick 2 (NCS2)** that can be plugged into a USB port on the same computer as the RealSense camera to provide hardware acceleration for vision computing applications.

https://store.intelrealsense.com/buy-intel-neural-compute-stick-2.html

There is a RealSense SDK example program for detecting an object and measuring the distance to it that makes use of NCS2.

https://github.com/PINTO0309/MobileNet-SSD-RealSense

I do not have information though on whether it can provide a beneficial effect for spatial filtering though.   I will ask the RealSense SDK Manager about it.

@dorodnic  Do you know whether librealsense's spatial filtering can benefit from the NCS2, please?

tarekmuallim, 2020-08-10:
Hi @MartyG-RealSense 
Thank you for the quick replay.

I am not looking for a new hardware.
Intel up board or NVIDIA Jetson Nano already have gpu so no need for a new hardware.
and since `rs2::spatial_filter::process` can be applied using parallel processing, I expected to find a similar function to implement it on the gpu.

like the functions provided by `rs2::gl
`

MartyG-RealSense, 2020-08-10:
The Jetson's Nvidia GPU can assist with **depth to color alignment** and **color conversion** when CUDA support is enabled.

https://github.com/IntelRealSense/librealsense/pull/1866

tarekmuallim, 2020-08-11:
@MartyG-RealSense yes I already mentioned this problem year ago.
`rs2::gl` accelerate the point cloud calculation implementing the gpu. this some how solve the problem of the point cloud generation time.
I still have problem with ` rs2::spatial_filter::process`
and I assumed after one year I will have a solution.

MartyG-RealSense, 2020-08-11:
I ran tests with the spatial filter and it seemed to verify the previous advice of @dorodnic in https://github.com/IntelRealSense/librealsense/issues/4468 that it does not seem to make a significant difference to the image.  Using the **Hole-Filling** filter has a much greater instant impact on the cleanliness of an image.

I tested a range of options to improving the quality of edges, if that is your reason for using the spatial filter.  Using the **Medium Density** visual preset (a balance of image fill and accuracy) in combination with maximised **Laser Power** (to minimise the "sparseness" of the image) may provide similar image quality with less processing demand.

tarekmuallim, 2020-08-12:
@MartyG-RealSense thanks for the advice.

As you say hole filling filter improve the quality of edges a lot with low computation power needed.

Regarding the  spatial filter I admit it make a small deference. However it still improve the result.
I attached pictures for two results, one with applying spatial filter and one with not.

[image]
[image]

Here the laser power is set to max.
You can see that the box surface is noisy and wavy, while with spatial filter on it become flat. This can improve the measurement result.

MartyG-RealSense, 2020-08-12:
Point cloud "waviness" may increase as the camera moves further from an object and decrease if closer to the object.  It may be worth trying to move the camera closer to the box if you only need to capture the box and not the surface that it is sitting on.

tarekmuallim, 2020-08-12:
In our application the camera is fixed above the box with a fixed distant.

Any way, I think we have shifted from the title here.
So the answer to my question is till now there is no  acceleration for `rs2::spatial_filter::process` , and there is no intend to do some acceleration for it in the future.

Am I right?

MartyG-RealSense, 2020-08-12:
Yes, my aim was to see if there were ways other than using a spatial filter that your desired results could be achieved.

As far as I know, CUDA and GLSL are the main **software** ways to achieve offloading of work from the CPU onto the GPU to accelerate applications.  There is also **hardware** acceleration with the NCS2, though you do not want to opt for that route.

I'm not on the developer team (I am on the Support team) so I do not have knowledge to add to Dorodnic's previous statement a year ago about spatial filters and acceleration on your old case.

https://github.com/IntelRealSense/librealsense/issues/4468#issuecomment-513485662

In regard to OpenCL, there is an OpenCL integration for Python.  I would speculate that it might be within the realms of possibility to link into it via the RealSense Python wrapper but that is a subject outside of my programming knowledge.

https://github.com/inducer/pyopencl