realsenseai/librealsense #7078: install Realsense on Jetson Xavier agx

Fred3D-tech, 2020-08-10:
Is there a simple way to install Pyrealsense2 on Jetson Xavier agx ?
I tried ALL the solutions proposed on this forum but nothing worked: the problem is with the python wrapper.
The problem seems to be that the kernel has to been patched.
Their is a solution here (and a good explanation) : https://github.com/jetsonhacks/buildLibrealsense2Xavier
but the article seemed to be outdated...
Is their a workable solution ?

MartyG-RealSense, 2020-08-10:
Hi @Fred3D-tech   The link below has a method for building librealsense and the Python wrapper together by using the RSUSB backend method that does not require patching.   It worked for a Jetson Nano user.

https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-665650908

An Xavier NX user who had problems with using Python 3 also provided their solution that you could try if you encounter problems with this on your AGX:

https://github.com/IntelRealSense/librealsense/issues/6980#issuecomment-666858977

Fred3D-tech, 2020-08-11:
thanks but #6964 is not working for me (I have the last version of the jetpack and sdk manager)...
cmake --version
cmake version 3.18.1

CMake suite maintained and supported by Kitware (kitware.com/cmake).

cmake ../ -DFORCE_RSUSB_BACKEND=ON -DBUILD_PYTHON_BINDINGS:bool=true -DPYTHON_EXECUTABLE=...
-- Internet connection identified
-- Info: REALSENSE_VERSION_STRING=2.36.0
-- Setting Unix configurations
-- Info: Building with CUDA requires CMake v3.8+
-- CUDA_LIBRARIES: /usr/local/cuda-10.2/include /usr/local/cuda-10.2/lib64/libcudart_static.a;Threads::Threads;dl;/usr/lib/aarch64-linux-gnu/librt.so;/usr/local/cuda-10.2/lib64/libcusparse.so;/usr/lib/aarch64-linux-gnu/libcublas.so
-- using RS2_USE_LIBUVC_BACKEND
CMake Error at wrappers/python/third_party/pybind11/tools/FindPythonLibsNew.cmake:95 (message):
  Python config failure:

Call Stack (most recent call first):
  wrappers/python/third_party/pybind11/tools/pybind11Tools.cmake:16 (find_package)
  wrappers/python/third_party/pybind11/CMakeLists.txt:33 (include)

-- Configuring incomplete, errors occurred!
See also "/home/fred/librealsense/build/CMakeFiles/CMakeOutput.log".
See also "/home/fred/librealsense/build/CMakeFiles/CMakeError.log".

Fred3D-tech, 2020-08-11:
ok.. cmake works with :
cmake -DFORCE_RSUSB_BACKEND=ON -DBUILD_PYTHON_BINDINGS:bool=true -DPYTHON_EXECUTABLE=/usr/bin/python3 -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true -DBUILD_WITH_CUDA:bool=true ..

MartyG-RealSense, 2020-08-11:
Thanks very much for the above update about CMake @Fred3D-tech - did it solve your installation problems, please?   Thanks!

Fred3D-tech, 2020-08-11:
not really... it is still not working...

  File "/home/fred/librealsense/wrappers/python/examples/pyglet_pointcloud_viewer.py", line 91, in <module>
    pipeline = rs.pipeline()

AttributeError: module 'pyrealsense2' has no attribute 'pipeline'

MartyG-RealSense, 2020-08-12:
Typically, pyrealsense2 scripts begin with **import pyrealsense2 as rs** instead of **import pyrealsense2**.  So as pipeline uses the definition **rs.pipeline**, I would speculate that it may not be able to access the pipeline if you have not done **import pyrealsense2 as rs**

Fred3D-tech, 2020-08-12:
of course, I know that. I'm using this program to test it : https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/pyglet_pointcloud_viewer.py
and even if the import pyrealsense2 is done, I get this strange error on the Xavier.. of course it works without problem on a windows 10 machine and on an intel nuc working with ubuntu.

MartyG-RealSense, 2020-08-12:
@Fred3D-tech   Below is a link to recent methods submitted by Python 3 pyrealsense2 users that helped them to get it working with Python 3.  Read downwards from the linked-to comment please:

https://github.com/IntelRealSense/librealsense/issues/6296#issuecomment-672768158

Fred3D-tech, 2020-08-12:
ok... I tried python3 -m pip install pyrealsense2
here is the result:
ERROR: Could not find a version that satisfies the requirement pyrealsense2 (from versions: none)
ERROR: No matching distribution found for pyrealsense2
is their a link for a repository?

AndreV84, 2020-08-12:
it is only for x86_64 @Fred3D-tech 
for Jetson nx/xavier/nano I can see working the following approach:
cmake 3.18 &
`cmake -DFORCE_RSUSB_BACKEND=ON -DBUILD_PYTHON_BINDINGS:bool=true -DPYTHON_EXECUTABLE=/usr/bin/python3 -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true -DBUILD_WITH_CUDA:bool=true ..`
no there is no link to repository;
Moreover, after making install it might still require manual operations for copying two files to the folder from that a python file will be executed:
```
cp /home/nvidia/librealsense-2.38.0/build/wrappers/python/pyrealsense2.cpython-36m-aarch64-linux-gnu.so ~/script/pyrealsense2.so
cp /home/nvidia/librealsense-2.38.0/build/librealsense2.so ~/script/
```
then from the script folder I can run python3 import or execute python3 files

AndreV84, 2020-08-12:
e.g. for AGX it is
`cmake ../ -DFORCE_RSUSB_BACKEND=true -DBUILD_PYTHON_BINDINGS=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_WITH_CUDA=true -DPYTHON_EXECUTABLE=/usr/bin/python3 -DBUILD_GRAPHICAL_EXAMPLES=true -DCMAKE_CUDA_ARCHITECTURES=72`

AndreV84, 2020-08-12:
for 3.18 cmake and for 3.13 cmake the line differs slightly though
fo r3,13 is is 
`cmake -DFORCE_RSUSB_BACKEND=ON -DBUILD_PYTHON_BINDINGS:bool=true -DPYTHON_EXECUTABLE=/usr/bin/python3 -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true -DBUILD_WITH_CUDA:bool=true -DBUILD_WITH_CUDA:bool=true -DCMAKE_CUDA_ARCHITECTURES=72 ..`

AndreV84, 2020-08-12:
let me try your script
0 updating the cmake to 3.18
 ```
wget https://github.com/Kitware/CMake/releases/download/v3.18.1/cmake-3.18.1.tar.gz
   tar -xvf cmake-3.18.1.tar.gz 
    cd cmake-3.18.1/
./bootstrap --system-curl
make -j8
# sudo make install optional step; might also include removing current apt cmake version
echo 'export PATH=/home/nvidia/cmake-3.13.0/bin/:$PATH' >> ~/.bashrc
source ~/.bashrc
```
installing mlocate & updating the database
`sudo apt install mlocate -y && sudo apt updatedb`
locating the required file
` sudo updatedb`
`locate pyrealsense2.cpython-36m-aarch64-linux-gnu.so`
`/home/nvidia/librealsense/librealsense-2.38.0/build/wrappers/python/pyrealsense2.cpython-36m-aarch64-linux-gnu.so`
locate pyglet_pointcloud_viewer.py
/home/nvidia/librealsense/librealsense-2.38.0/wrappers/python/examples/pyglet_pointcloud_viewer.py
copying it to the folder containing the example pyglet_pointcloud_viewer.py
```
cp /home/nvidia/librealsense/librealsense-2.38.0/build/wrappers/python/pyrealsense2.cpython-36m-aarch64-linux-gnu.so /home/nvidia/librealsense/librealsense-2.38.0/wrappers/python/examples/pyrealsense2.so
then locating and copying the .so file librealsense.so
`locate librealsense*.so
``cp /home/nvidia/librealsense/librealsense-2.38.0/build/librealsense2.so /home/nvidia/librealsense/librealsense-2.38.0/wrappers/python/examples/`

```

first it will require to copy two .so files to the folder where the script is located:
1.
2.
then trying the script:
python3 pyglet_pointcloud_viewer.py 
Traceback (most recent call last):
  File "pyglet_pointcloud_viewer.py", line 39, in <module>
    import pyglet
ModuleNotFoundError: No module named 'pyglet'
adding the piglet with
pip3 install piglet;
trying again
```
python3 pyglet_pointcloud_viewer.py 
Traceback (most recent call last):
  File "pyglet_pointcloud_viewer.py", line 40, in <module>
    import pyglet.gl as gl
  File "/home/agx/.local/lib/python3.6/site-packages/pyglet/gl/__init__.py", line 243, in <module>
    import pyglet.window
  File "/home/agx/.local/lib/python3.6/site-packages/pyglet/window/__init__.py", line 1897, in <module>
    gl._create_shadow_window()
  File "/home/agx/.local/lib/python3.6/site-packages/pyglet/gl/__init__.py", line 220, in _create_shadow_window
    _shadow_window = Window(width=1, height=1, visible=False)
  File "/home/agx/.local/lib/python3.6/site-packages/pyglet/window/xlib/__init__.py", line 173, in __init__
    super(XlibWindow, self).__init__(*args, **kwargs)
  File "/home/agx/.local/lib/python3.6/site-packages/pyglet/window/__init__.py", line 585, in __init__
    display = pyglet.canvas.get_display()
  File "/home/agx/.local/lib/python3.6/site-packages/pyglet/canvas/__init__.py", line 94, in get_display
    return Display()
  File "/home/agx/.local/lib/python3.6/site-packages/pyglet/canvas/xlib.py", line 123, in __init__
    raise NoSuchDisplayException('Cannot connect to "%s"' % name)
pyglet.canvas.xlib.NoSuchDisplayException: Cannot connect to "None"
seems working but for the fact that I have display and camera disconencted

```

AndreV84, 2020-08-12:
@Fred3D-tech let us know if the solution above worked for you;
btw do you have jetpack version 4.4GA? DP? 4.3?
I can supply files for 4.4GA AGX; find them atatched
https://storage.googleapis.com/realsense/librealsense2.so
https://storage.googleapis.com/realsense/pyrealsense2.so
pitting this two into the folder will allow you tu pun python3 import or to run python 3 files from it

MartyG-RealSense, 2020-08-12:
@AndreV84  Thanks so much for all the effort you have gone to in developing your solution and your offer of help to @Fred3D-tech

Fred3D-tech, 2020-08-12:
ok thanks...
then what I'm trying;
cmake --version
cmake version 3.18.1

CMake suite maintained and supported by Kitware (kitware.com/cmake).
then cmake is already v 3.18.1

cmake ../ -DFORCE_RSUSB_BACKEND=true -DBUILD_PYTHON_BINDINGS=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_WITH_CUDA=true -DPYTHON_EXECUTABLE=/usr/bin/python3 -DBUILD_GRAPHICAL_EXAMPLES=true -DCMAKE_CUDA_ARCHITECTURES=72

then :
make -j6
sudo make install
then I copied 
/usr/lib/pyrealsense2.cpython-36m-aarch64-linux-gnu.so
/usr/lib/pyrealsense2.cpython-36m-aarch64-linux-gnu.so.2.36
/usr/lib/pyrealsense2.cpython-36m-aarch64-linux-gnu.so.2.36.0
to a script directory, I placed my code in this script directory and it finally worked.
Thanks a lot!

MartyG-RealSense, 2020-08-12:
@Fred3D-tech   Great news that it worked for you!  @AndreV84   Thank you again for your help.

AndreV84, 2020-08-12:
@Fred3D-tech Thank you for letitng us know!
@MartyG-RealSense thanks is good, but H1B work authorization is better. I heard that intel should be able to do H1B.

AndreV84, 2020-08-12:
the proposed method seems to allow to use with AGX Xavier the following approach https://ardupilot.org/rover/docs/common-vio-tracking-camera.html , initially defined for Rasp. Pi devices

AndreV84, 2020-08-18:
Moreover, there is nother project that seems to work with AGX/ nano Jetsons:
https://github.com/devshank3/JetScan
https://www.hackster.io/devshank/jetscan-16a521
That can be used with d400 cameras

MartyG-RealSense, 2020-08-27:
Hi @AndreV84   Do you still require assistance with this case, please?   Thanks!

AndreV84, 2020-08-27:
Hi @MartyG-RealSense 
Thhank you for following up.
We are able to install realsense sdk on Xavier AGX, also d435i, d435, d455 seem detectable.
Thhere are othher issues apart of the installation setup though.
@Fred3D-tech , do you require furether assistance on the issue or the issue could be closed as for now?

Fred3D-tech, 2020-08-27:
hello @AndreV84 ...
It is also ok for me and it is working with my d415, d435 and l515... then everything is ok for me and you can close the issue.
My last issue (as you I have seen) is with open3d! but nothing to do with realsense2.
Thanks a lot

MartyG-RealSense, 2020-08-27:
Thanks so much @Fred3D-tech and @AndreV84 for your updates.   @Fred3D-tech as this was originally your case and you are satisfied to close it, I will do so.  Thanks again!

closed 2020-08-27 by MartyG-RealSense

devyouship, 2023-03-23:
It is easier to add the path on the script:

sys.path.append("/usr/local/lib/python3.6/pyrealsense2/")
import pyrealsense2 as rs

MartyG-RealSense, 2023-03-23:
Thanks so much @devyouship for sharing your experience of this subject with the RealSense community!