realsenseai/librealsense #7080: Program terminated due to an unrecoverable SEH exception

gmaajid, 2020-08-10:
When running the pre-relaese version 2.37.0 I encountered the following exception.

| | |
|---|---|
|**librealsense**|2.37.0 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```

Program terminated due to an unrecoverable SEH exception:
Illegal CPU instruction!
Possibly newer CPU architecture is required
```

[DEV1.txt](https://github.com/IntelRealSense/librealsense/files/5052122/DEV1.txt)

MartyG-RealSense, 2020-08-10:
Hi @gmaajid   Your kindly provided log has the following details about your computer's CPU:

**Intel Core i7 3930K
Sandy Bridge-E
Intel(R) Core(TM) i7-3930K CPU @ 3.20GHz**

Sandy Bridge-E architecture is very old, being launched in late 2011.  Even the original RealSense camera in 2014 required a Haswell CPU that was 2 architecture generations newer than Sandy Bridge.  (Sandy Bridge > Ivy Bridge > Haswell).  Although RealSense 400 Series cameras (if that is what you are using) are meant to work with any Intel CPU, there may be a limit to that in terms of processors that are excessively aged.

Have you been able to successfully run the RealSense Viewer on any previous RealSense SDK versions such as the slightly older **2.36.0** (which is a stable version of the RealSense SDK rather than a pre-release). 

https://github.com/IntelRealSense/librealsense/releases/tag/v2.36.0

dorodnic, 2020-08-11:
Here is some data on CPU requirements - 
1. RealSense Viewer bundled inside SDK installer requires Skylake architecture CPU (since it includes OpenVINO and OpenVINO requires AVX instruction set)
2. Rest of the SDK, including depth quality tool or the standalone viewer you can download from the site should work on more or less any x86 and Arm CPUs, just slower. Another factor is OpenGL 3.0 and GLSL 1.3 compatibility via graphics driver.

gmaajid, 2020-08-11:
I tried 2.36.0  and got the same exception. I can't use an older build since I'm working with the L515 Lidar sensor and I don't think it's supported by older SDKs. 

Thanks for the information. I'll try a machine with a newer CPU when using the RealSense Viewer. It's good to know the rest of the SDK should function on older machines.

I'll close this issue since this issue is expected behavior.

closed 2020-08-11 by gmaajid

dorodnic, 2020-08-12:
Hi @gmaajid 
Please let us know if you were able to run [standalone viewer](https://github.com/IntelRealSense/librealsense/releases/download/v2.36.0/Intel.RealSense.Viewer.exe) and feel free to reopen / open new issue if you encounter any problem

wldevries, 2020-09-15:
I think a lot of people have these problems. Six year old i7 CPUs run just fine and this specific CPU generation requirement isn't listed anywhere as far as I can tell. You should seriously consider only loading the OpenVINO component if requirements are met and or show a more appropriate message and update the requirements for the SDK.