realsenseai/librealsense #7082: Unable to locate package librealsense2-dkms

MarcoAndrade26, 2020-08-10:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 & D435| 
| Firmware Version                   | NA | 
| Operating System & Version |  Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  5.3.0-1030-raspi2                                         | 
| Platform                                 | Raspberry Pi 4B 4Gb|
| SDK Version                            | 2.0                        | 
| Language                            |  NA                          | 
| Segment			|  Robotics }                          | 

### Issue Description
Hi, I am following the tutorial form: https://intel.github.io/robot_devkit_doc/pages/install.html

Executing the following commands:
git clone https://github.com/intel/robot_devkit.git
cd robot_devkit
./demo/rdk_install.sh

But as a result I got:
E: Unable to locate package librealsense2-dkms

RealSenseSupport, 2020-08-10:
Hi Marco,

While we do not provide support for things outside our repo we do recommend installing librealsense/realsense-ROS using our instructions here:

https://github.com/IntelRealSense/realsense-ros

Thanks

MarcoAndrade26, 2020-08-10:
Thanks! I will try, is there any release for ROS2 Dashing?

JimXu1989, 2021-02-28:
> Required Info	
> Camera Model	T265 & D435
> Firmware Version	NA
> Operating System & Version	Ubuntu 18.04
> Kernel Version (Linux Only)	5.3.0-1030-raspi2
> Platform	Raspberry Pi 4B 4Gb
> SDK Version	2.0
> Language	NA
> Segment	Robotics }
> ### Issue Description
> Hi, I am following the tutorial form: https://intel.github.io/robot_devkit_doc/pages/install.html
> 
> Executing the following commands:
> git clone https://github.com/intel/robot_devkit.git
> cd robot_devkit
> ./demo/rdk_install.sh
> 
> But as a result I got:
> E: Unable to locate package librealsense2-dkms
> 
> Also tried with manual installation and no result.

Hi, I have met the same problem, have you solved the problem?
Thanks

jhbirdchoi, 2021-04-15:
I have met the same problem, have you solved the problem?
?????????
thanks

BartVanderHaagen, 2021-04-21:
Same here on Jetson Nano. 

After successfully changing Kernel and installing Librealsense ROS via https://github.com/IntelRealSense/realsense-ros 

i get 

E: Unable to locate package librealsense2-dkms

the Application works partly and run a lot errors, also RTABMAP can not use the frames to generate a map.

Prasanna-icefire, 2021-07-29:
I have the same issue on Jetson Xaview NX as well!

sumitpaulde, 2022-09-16:
@MarcoAndrade26 do you have any idea how to solve this issue? Waiting for your reply.

paolof91, 2022-09-20:
same issue

ryanmaxwell96, 2022-11-09:
Same issue. Any update?

MartyG-RealSense, 2022-11-09:
Hi everyone, there are dedicated instructions for installing the librealsense SDK on Jetson boards here:

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md

Once the SDK has been installed (either with packages or source code) and tested as successfully working then the RealSense ROS2 wrapper can be installed afterwards on Jetson by building it from **source code**.

https://github.com/IntelRealSense/realsense-ros#step-3-install-intel-realsense-ros2-wrapper-from-sources

*******************

Regarding the original comment at the top of this case about using **robot_devkit**, the ROS2 section of the robot-devkit instructions are out of date as they were written in 2019 and refer to a ROS2 wrapper called **ros2_intel_realsense**.  However, that wrapper ceased to be updated around 2019.  

https://github.com/intel/ros2_intel_realsense

**ros2-development** is the current and actively updated RealSense ROS2 wrapper.

https://github.com/IntelRealSense/realsense-ros/tree/ros2-development

closed 2026-10-23 by MartyG-RealSense