realsenseai/librealsense #7084: Cannot run IR+Depth streams from paired D415's at 15 FPS

icefoxen, 2020-08-10:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 0.5.12.03.00 | 
| Operating System & Version |   Linux, Ubuntu 18.04 | 
| Kernel Version (Linux Only)    |  4.9.X                              | 
| Platform                                 | NVidia Jetson TX2 |
| SDK Version                            |  2.35.2                          | 
| Language                            |  C++  | 
| Segment			|  Robotics     | 

### Issue Description

Dear Intel,

Long story short: I have two Realsense D415's set up with hardware-synced clocks, one producing the time sync signal and the other consuming it..  I'm writing some software using the C++ `librealsense2` lib to do some image processing and generate point clouds and such, with slightly grim CPU/power requirements since the end result is going to be battery powered.  I am trying to get an IR camera image and depth image at the same time for each camera, at a rate 15 FPS.  However, whenever I try to open *both* the `IR1` and `Depth` video streams for both cameras, the IR1 stream is fine and rock-solid at 15 FPS, while the depth stream is incredibly slow and flaky, maxing out at 10-12 FPS even on a powerful computer and often dropping down to <1 FPS.  

So, I suppose I'm asking, is this a configuration that *should* work?  Profiling has been pretty unhelpful, and here's what I've tried so far:

 * I'm pretty certain I'm not CPU bound, since I see similar behavior on a very powerful computer.
 * I'm pretty certain I'm not limited by USB3 bus bandwidth, since the math doesn't add up and I can run one camera at 30 fps just fine.
 * I'm somewhat certain that it's not a problem with USB cables, since I've tried a few different ones and found a set that seems good
 * I'm *mostly* certain that it is supposed to be possible to do this, since I've tried replicating this configuration with `realsense-viewer` and gotten it to work at 15 FPS...  only sometimes though, since `realsense-viewer` crashed a lot and often seemed to have issues re-initializing the cameras until I pulled the power on them and tried again.

This is bad for me, and I'm having trouble diagnosing the problem.  This is a bit of a research project for me, and there's any number of thing going on that could be the problem.  Any advice you have for things to check would be very welcome.

MartyG-RealSense, 2020-08-11:
Hi @icefoxen  First, may I confirm please that on one camera you have set the **INTER CAM SYNC** option to '1' for the camera that will be synced with (the Master camera) and set the other camera's INTER CAM SYNC to '2' so that it follows the timing of the Master camera (i.e it is a Slave).

Historically, multicam on Jetson boards tends to more commonly experience problems such as the **failed to set power state** error when using more than one camera.  SDK 2.35.2 introduced improvements to how librealsense handles multicam, though the enhancements do not prevent problems that are caused by specific models of USB hub.  

Could you unplug both cameras please and then plug them back in using a different order (Cam A first and Cam B second on the first test, and CAM B first and Cam A second on the next test) and see if you have any improvement in reliability with one insertion pattern or the other?  I ask this because prior to the multicam enhancements being introduced, one workaround for multicam Jetson problems was to insert the cameras in a certain order.

I note though that you experienced the same problem on a powerful PC, so it may be worth looking at the common factors between the Jetson setup and the PC setup.  This could include the USB hub if you are using one.  If the problem was related to the camera firmware driver then any problem caused by the firmware on the Jetson may be carried over to the PC.  This is because the firmware driver is stored inside the camera.

icefoxen, 2020-08-13:
> Hi @icefoxen First, may I confirm please that on one camera you have set the INTER CAM SYNC option to '1' for the camera that will be synced with (the Master camera) and set the other camera's INTER CAM SYNC to '2' so that it follows the timing of the Master camera (i.e it is a Slave).

That is the configuration I am using... however I just double checked and it *appears* that setting both cameras to 0 make the framerate problems go away!  Thank you for the suggestion to double check, I will follow up on this side of the problem with more tests.  I've been bitten before by getting excited when something seems to fix it only to mess up again later, so I will do as much investigation of this as I can, but it's a angle of attack.

> Historically, multicam on Jetson boards tends to more commonly experience problems such as the failed to set power state error when using more than one camera.

I've run into some problems along those lines...  I ended up setting it up so I call `device::hardware_reset()` on all the devices each time the software starts and make sure all the sensor streams are shut down properly when it finishes, and that ended up solving a lot of problems.

The other suggestions are good, thank you.  I'll go down the full list and make sure I check everything again.  I have a couple different models of USB hub to try.  I THINK I'm using the same version of the firmware on both sets of cameras but I will verify.

MartyG-RealSense, 2020-08-13:
Setting INTER CAM SYNC to '0' on both cameras will basically mean that they are in the default unsynched state and operating independently (i.e no master or slave).

Thanks very much for sharing the methodology that worked for you.

icefoxen, 2020-08-14:
> Could you unplug both cameras please and then plug them back in using a different order (Cam A first and Cam B second on the first test, and CAM B first and Cam A second on the next test) and see if you have any improvement in reliability with one insertion pattern or the other?

Ok, I tried this, and it didn't make any difference.  I also tried two different pairs of Realsense cameras and it didn't make any difference.  Regardless of which pair I used or what order they were in, when attached to the Jetson TX2 board, setting one with `INTER_CAM_SYNC` to 1 and the other to 2 made the depth images not come through properly.  Changing USB hubs also didn't make any difference.

However, I DID discover a wiring flaw with the setup I was using connected to the powerful computer.  With that corrected, this works *fine* on the desktop!  So, one way or another the difference is something about the TX2.  Perhaps its USB input cannot handle the bandwidth (though it handles one camera at 30 FPS just fine...) or perhaps it is not putting out as much power (though I don't know why that would matter for camera sync, I can dig up a powered hub and check).

MartyG-RealSense, 2020-08-15:
Powered hubs do usually help, though on Jetson they do not seem to be as reliable a solution to multicam issues as they are when used with other computer types such as PC.  

Regarding the error **uvc streamer watchdog triggered on endpoint: 130**: another TX2 user who had this error found that it was resolved after they updated their TX2.

https://github.com/IntelRealSense/realsense-ros/issues/1015#issuecomment-564411882

MartyG-RealSense, 2020-08-27:
Hi @icefoxen  Do you require further assistance with this case please, or can it be closed?   Thanks!

icefoxen, 2020-08-29:
Been rummaging around on the NVidia side of things, and I think that's where the problem lies.  Closest thing I've been able to find is this issue: https://forums.developer.nvidia.com/t/realsense-camera-unstable/71798

Wasn't able to get the fix listed there to work on the TX2, but using the R32.4.3 version firmware increased the clock rate of their USB controller by 50%, and improved this problem a *little*... I can run two cameras with external sync with 15 FPS IR streams and 8-9 FPS depth streams.  So the only thing I can think of right now is that the TX2's USB controller is just too slow to get slammed with 4 frames all instantly at the same time, but can handle 2-3 of them fine, or all 4 of them if they're spaced out a little.  I don't know enough about USB3 to know if that is actually what's going on, but that's my working theory.  Using a PC with PCIe USB expansion card works great, next I guess I need to get the card working on a Jetson and see what happens.

Next time I'll just use MIPI or some other video solution than USB3, I guess.  ANY other, USB3 has just been the worst part of this project.  Or not use NVidia hardware.  Or both.

Thank you!  This can be closed.

MartyG-RealSense, 2020-08-29:
Thanks so much @icefoxen for your very detailed insights that should benefit others reading this case in future!

closed 2020-08-29 by MartyG-RealSense