realsenseai/librealsense #7086: Exception if setting the AutoExposureRoi on some RS D435

Sanelle2001, 2020-08-11:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | 5.12.06.00 | 
| Operating System & Version |   {Win 10 1903 18362.959  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | intel NUC |
| SDK Version                            |  2.36.0 }                          | 
| Language                            |  c#                           | 
| Segment			|  other                          | 

### Issue Description

Setting is 640x480 15Hz

If I try to set Auto exposure ROI with 

_rgbSensor.AutoExposureSettings.SetRegionOfInterest(new ROI() { minX = minX, maxX = maxX, minY = minY, maxY = maxY });

Some cameras are OK, other Cameras throw an Exception:

[ **System.ArgumentException: hwmon command 0x75( 18 1c8 e0 1a0 ) failed (response -6= Invalid parameter) ---> System.Runtime.InteropServices.ExternalException: rs2_set_region_of_interest(sensor:0000024B2848EF30, min_x:224, min_y:24, max_x:416, max_y:456)**
   --- End of inner exception stack trace ---
   at Intel.RealSense.ErrorMarshaler.MarshalNativeToManaged(IntPtr pNativeData) in C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\wrappers\csharp\Intel.RealSense\Helpers\ErrorMarshaler.cs:line 72
   at System.StubHelpers.MngdRefCustomMarshaler.ConvertContentsToManaged(IntPtr pMarshalState, Object& pManagedHome, IntPtr pNativeHome)
   at Intel.RealSense.NativeMethods.rs2_set_region_of_interest(IntPtr sensor, Int32 min_x, Int32 min_y, Int32 max_x, Int32 max_y, Object& error)
   at Intel.RealSense.AutoExposureROI.SetRegionOfInterest(ROI value) in C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\wrappers\csharp\Intel.RealSense\Types\AutoExposureROI.cs:line 25
  
If I open RealSenseViewer I see an error wits RS I2C ...

I have no idea where to search and what I can do?!

MartyG-RealSense, 2020-08-11:
Hi @Sanelle2001   Can you confirm whether you are using kernel 4.14.13 as indicated on your specification description at the top of this case, please?  Kernel 4.14 is not on the RealSense SDK's list of officially supported kernels (it goes 4.13, then 4.15).  

[image]

Sanelle2001, 2020-08-11:
Sorry I corrected this, I'm using Windows 10 and not Linux!

MartyG-RealSense, 2020-08-11:
No problem, thanks for the correction!

Do you experience the I2C error every time that you open the RealSense Viewer with the affected cameras?

Sanelle2001, 2020-08-11:
Only after I try to set the ROI in my software and afterwards opening RealSenseViewer, in about 8 of 10 cases.

MartyG-RealSense, 2020-08-11:
And you have shut down your own program before launching the RealSense Viewer to make sure that your own software is not trying to claim access to the camera?

Sanelle2001, 2020-08-11:
Yes, for sure.

MartyG-RealSense, 2020-08-11:
If you are able to use the Viewer with the affected camera, do you also get an error if you enable auto-exposure, start a stream (to make the Set ROI button appear aside the Enable Auto-Exposure option) and then click the Set Roi button and draw a ROI box on the Viewer's center panel with the mouse?  If there is a problem with the ROI function, I would expect an error to appear when releasing the left mouse button after dragging out a box.

Sanelle2001, 2020-08-11:
I tried it out and the ROI works in RealSense Viewer without error. Afterwards with my software the error came again. After changing the camera, both programs runs as expected. Changing back, the camera with the problem throws the exception again with my software, RealSenseViewer works.

MartyG-RealSense, 2020-08-11:
You could eliminate the possibility that the affected cameras have a damaged firmware by downgrading the firmware version in the Viewer and then re-upgrading to the latest firmware version.

In the Viewer, you can downgrade using the following steps:

1.  Download firmware file **5.12.5.0** from the firmware releases page at the link below.

https://dev.intelrealsense.com/docs/firmware-releases

2.  Open the RealSense Viewer and go to the **More** icon at the top of the Viewer's options side-panel.  Select **Update Firmware** from its menu and select the firmware file that you downloaded.

3.  Test whether the ROI problem still occurs with the camera when using firmware 5.12.5.0.  You can then re-update to **5.12.6.0** without a firmware file by going back to the More option and this time selecting **Install Recommended Firmware** from the menu.

Sanelle2001, 2020-08-11:
OK, I try this tomorrow, other stuff has prio. Thanks

MartyG-RealSense, 2020-08-11:
You are very welcome.   I look forward to your test results later.   Good luck!

closed 2020-08-26 by Sanelle2001