realsenseai/librealsense #7088: Wrong options values during net_device initialization

podshivalov-ilya, 2020-08-11:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.12.07.100 | 
| Operating System & Version |  Linux (Manjaro 20.0.3)  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC, Raspberry Pi  |
| SDK Version                            |  2.38.0                          | 
| Language                            |  C++                          | 
| Segment			|  others                          | 

I'm trying to configure network device according to the [article](https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras). Everything works fine when I add device through realsense viewer. But in my project `rs2::net_device` constructor throws an exception `out of range value for argument "def"`. I got a realsense log:
`RealSense network logging initialized`
`Created new TCP socket 21 for connection`
`continueAfterOPTIONS 0 RGB Camera[0{0,1,0,1};1{-64,64,0,1};2{0,100,50,1};3{1,10000,166,1};4{0,128,64,1};5{100,500,300,1};6{-180,180,0,1};7{0,100,64,1};8{0,100,50,1};9{2800,6500,4600,10};10{0,1,1,1};11{0,1,1,1};19{0,32,16,1};22{0,3,3,1};30{0,1,1,1};53{0,1,1,1};]Stereo Module[3{1,165000,8500,1};4{16,248,16,1};10{0,1,1,1};12{0,6,0,1};13{0,360,150,30};18{0,2,1,1};19{0,32,16,1};24{0,1,0,1};26{0,1,0,1};28{1e-06,0.01,0.001,1e-06};40{50.1787,50.1787,50.1787,0};42{0,258,0,1};46{0,1,0,1};53{0,1,1,1};71{0,1,0,1};]`
`Init sensor 1, option 'Backlight Compensation', value 4.34403e-44`
`Init sensor 1, option 'Brightness', value 4.34403e-44`
`is_valid(...) failed! step is not properly defined. (7.70714e-44)`
`is_valid(...) failed! step is not properly defined. (7.70714e-44)`
`Init sensor 1, option 'Contrast', value 4.34403e-44`
`out of range value for argument "def"`
`null pointer passed for argument "device"`
By the way, in my dev laptop I haven't got this exception and continueAfterOPTIONS are the same but device initializes successfully. What can I do with it?

podshivalov-ilya, 2020-08-12:
There's just locale issue in (sscanf)[https://github.com/IntelRealSense/librealsense/blob/master/src/ethernet/RsRtspClient.cpp#L623] because for some locales LC_NUMERIC specifies comma as decimal separator for floats instead of dot.

RealSenseSupport, 2021-04-06:
@podshivalov-ilya I saw you've submitted PR https://github.com/IntelRealSense/librealsense/pull/7096 and waiting to be merged. Can we close this issue here? Looking forward to your update. Thanks!

podshivalov-ilya, 2021-04-07:
> @podshivalov-ilya I saw you've submitted PR #7096 and waiting to be merged. Can we close this issue here? Looking forward to your update. Thanks!

There is definitely the way to solve the problem. I don't mind at all because I solved it for myself and just intended to share it for the other users. Regardless of the solution, this PR is an example of the support quality for me, even if will be closed or opened for the next couple of years. Unfortunately, I need to support projects based on the product but there is a clear sign to consider alternatives.

RealSenseSupport, 2021-04-07:
@podshivalov-ilya Thanks for your quick reply! Closed this ticket accordingly and thanks again for your contribution!

closed 2021-04-07 by RealSenseSupport