realsenseai/librealsense #7089: disable histogram euqalization lead depth map to void

DragonDeux, 2020-08-11:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

DragonDeux, 2020-08-11:
Hello,

i have recounted a strange issue with new Realsense L515 device: if "histogram_equalization_enabled is disable" then the output depth map would be just a black file while using colorizer to extract this depth map. the depth visualization min and max distance is configured correctly.

Is there anyone could help? 
regards

dorodnic, 2020-08-11:
Hi @DragonDeux 
Sounds like a bug, we will try to look into it

DragonDeux, 2020-08-11:
Hello @dorodnic,

very thanks for reponse so quickly, but i have found the reason for causing this "bug(actually it's not the bug i thought)". It's because i didn't configure the "visual preset" to fixed, so the min and max distance setting didn't make any effects. and we also should take care to configure "depth visulization" with a propre order, for me like this is working(version python):

`colorMap = rs.colorizer()
colorMap.set_option(rs.option.visual_preset, 1)
colorMap.set_option(rs.option.histogram_equalization_enabled, 0.0) #disable histogram equalization
colorMap.set_option(rs.option.color_scheme, float)
colorMap.set_option(rs.option.min_distance , float)
colorMap.set_option(rs.option.max_distance , float) `
ps: with realsense D435 doesn't have the same "issus" because i have two of them, and i tested it.

regards

dorodnic, 2020-08-11:
Glad you found a solution. This seems like a bug on our side because of the different behaviour between L515 and D400.

DragonDeux, 2020-08-11:
I'm sorry, actually, the issue is awlays there. please keep this issue in mind .

DragonDeux, 2020-09-26:
hello @dorodnic 
Have you resolved this issue?

aangerma, 2020-10-20:
Hello @DragonDeux , I tried to reproduce this issue with our rs-capture example and L515, can you send me the code that reproduce it?

DragonDeux, 2020-10-20:
Hello @aangerma,
Yes sure,
`import pyrealsense2 as rs
import numpy as np
import cv2
import json as js

#coloriz depth map 
colorMap = rs.colorizer()
colorMap.set_option(rs.option.visual_preset, 1)
# colorMap.set_option(rs.option.histogram_equalization_enabled, 1.0) #disable histogram equalization
colorMap.set_option(rs.option.histogram_equalization_enabled, 0.0) #disable histogram equalization
colorMap.set_option(rs.option.color_scheme, 3.0)
colorMap.set_option(rs.option.min_distance , 0.1)
colorMap.set_option(rs.option.max_distance , 3)

#postprocess

temp_filter = rs.temporal_filter()
temp_filter.set_option(rs.option.holes_fill, 8)

#Create a pipeline
pipeline = rs.pipeline()
config = rs.config()

config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 1920, 1080, rs.format.bgr8, 30)
config.enable_stream(rs.stream.infrared, 640, 480, rs.format.y8, 30)
profile = pipeline.start(config)

    
#Getting the depth sensor's depth scale (see rs-align example for explanation)

depth_sensor = profile.get_device().first_depth_sensor()
depth_sensor.set_option(rs.option.invalidation_bypass , 1.0) #activate invalisation bypass

depth_scale = depth_sensor.get_depth_scale()
print("Depth Scale is: " , depth_scale)

#We will be removing the background of objects more than
#clipping_distance_in_meters meters away
clipping_distance_in_meters = 1 #1 meter
clipping_distance = clipping_distance_in_meters / depth_scale

#Create an align object
#rs.align allows us to perform alignment of depth frames to others frames
#The "align_to" is the stream type to which we plan to align depth frames.
align_to = rs.stream.color
align = rs.align(align_to)

try:
    while True:

        # Get frameset of color and depth
        frames = pipeline.wait_for_frames()
        infrared_frame = frames.first(rs.stream.infrared)
        IR_image = np.asanyarray(infrared_frame.get_data())
        # frames.get_depth_frame() is a 640x360 depth image

        # Get aligned frames
        aligned_depth_frame = frames.get_depth_frame() # aligned_depth_frame is a 640x480 depth image
        color_frame = frames.get_color_frame()
        
        
        
        
        # Validate that both frames are valid
        if not aligned_depth_frame or not color_frame:
            continue
        
        
        colorizedDepth = colorMap.colorize(aligned_depth_frame)
        
        depth_image = np.asanyarray(colorizedDepth.get_data())
        color_image = np.asanyarray(color_frame.get_data())

        
        cv2.imshow('Align color', color_image)
        cv2.imshow('Align depth', depth_image)
        cv2.imshow('infram depth', IR_image)
        
        key = cv2.waitKey(1)
        # Press esc or 'q' to close the image window
        if key & 0xFF == ord('q') or key == 27:
            cv2.destroyAllWindows()
            cv2.imwrite("depth.png", depth_image)
            cv2.imwrite("image.png", color_image)
            break
            
finally:
    pipeline.stop()`

DragonDeux, 2020-10-20:
Hello @aangerma, Yes sure, please download this attached python file
[l515.zip](https://github.com/IntelRealSense/librealsense/files/5407860/l515.zip)

dorodnic, 2020-10-26:
Thanks @DragonDeux

DragonDeux, 2020-11-04:
@dorodnic 
Hello Dorodnic, 
Did this issue has been solved ?  

regards,
Deux Xie

RealSenseSupport, 2021-01-19:
Hi @DragonDeux 

This issue should be fixed based on updates here in this PR:  https://github.com/IntelRealSense/librealsense/pull/7645

*If we don’t hear from you in 7 days, this issue will be closed.

closed 2021-01-29 by RealSenseSupport