realsenseai/librealsense #7091: Camera Configuration not loaded on Linux

sans-ltd, 2020-08-11:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   |05.12.05.00 | 
| Operating System & Version |  Linux (Build env: Ubuntu 16.04 + gcc-9 target: SuSE Leap 42.2  | 
| Kernel Version (Linux Only)    |                    4.4.27-2-default                   | 
| Platform                                 | PC (x64) |
| SDK Version                            | 2.35.2                          | 
| Language                            | C++                          | 
| Segment			|  Robot/machinery                          | 

We currently port a RealSense based application from Windows 10 to Linux. We use the load_json() functionality of the advanced_mode to initialize the camera. The JSON file itself itself loads fine and initialisation is working on Windows. On Linux we get the following exception in our application:
`
rs2_load_json(dev:0x224ff40, json_content:0x29e8290, content_size:3418):
    Operation failed with error code=1
Using RS2_RS400_VISUAL_PRESET_HIGH_ACCURACY as fallback!
`
The same json file loads fine within the RealSense viewer on the same Linux system running with the same user.
We have installed `99-realsense-libusb.rules` to get rid of the warning in the RealSense Viewer.
Error code 1 means permission denied within POSIX, but what's the difference between the viewer and our code?

Here is our relevant initialisation code (again, this is working on Windows):

```
rs2::device_list devices = ctx.query_devices(RS2_PRODUCT_LINE_DEPTH); 
// check if one device is connected else return.
const auto& device = *(devices.begin()); 
loadAvancedCameraConfigJson(device); 
```
The loader looks like this:
```
    void loadAvancedCameraConfigJson(const rs2::device& device)
    {
        if (device.is<rs400::advanced_mode>() && config.has_camera_settings_json_file())
        {
        // Open file, read into std::string jsonString
        if (ok)
        {
                try
                {
                    advancedDevice.load_json(jsonString);
                    std::cout << "Initialized the camera from " << config.camera_settings_json_file() << std::endl;
                    return;
                }
                catch (const rs2::error & e)
                {
                    std::cerr << "Error parsing advanced camera settings from " << config.camera_settings_json_file() << ": " << std::endl
                              << e.get_failed_function() << "(" << e.get_failed_args() << "):\n    " << e.what() << std::endl;
                }
         }
   }
}   
```

MartyG-RealSense, 2020-08-11:
Hi @sans-ltd    Reading your script carefully and comparing it to other programming references about loading a json with C++, I note that you check the status of Advanced Mode with **if (device.is<rs400::advanced_mode>()**

After that **device.is** statement though,  there does not seem to be a **device.as** statement.

A simple example of use of **device.as** is provided by the simple json loading C++ snippet on the link below:

https://github.com/IntelRealSense/librealsense/issues/3037#issuecomment-452732168

Whilst the Advanced Mode documentation has an example script for checking Advanced Mode status that is more similar to yours:

https://github.com/IntelRealSense/librealsense/blob/master/doc/rs400/rs400_advanced_mode.md#advanced-mode-c-api-1

To quote from the above script:

  if (dev.is<rs400::advanced_mode>())
    {
        **auto advanced_mode_dev = dev.as<rs400::advanced_mode>();**
        // Check if advanced-mode is enabled
        if (!advanced_mode_dev.is_enabled())
        {
            // Enable advanced-mode
            advanced_mode_dev.toggle_advanced_mode(true);
        }
    }
    else
    {
        cout << "Current device doesn't support advanced-mode!\n";
        return EXIT_FAILURE;
    }

sans-ltd, 2020-08-12:
@MartyG-RealSense  thanks for the fast reply. 
We changed our code accordingly to the given examples:
```
        if (device.is<rs400::advanced_mode>() && config.has_camera_settings_json_file())
        {
            auto advancedDevice = device.as<rs400::advanced_mode>();
            std::cout << "Could access the device in advanced mode: is_enabled: " << advancedDevice.is_enabled() << std::endl;
            if (!advancedDevice.is_enabled())
            {
                // Enable advanced-mode
                advancedDevice.toggle_advanced_mode(true);
            }
            std::cout << "After evetually toggle the advance_mode: is_enabled: " << advancedDevice.is_enabled() << std::endl;
            std::ifstream cfgJson(config.camera_settings_json_file());
            if (cfgJson.is_open() && cfgJson.good())
            {
                std::string jsonString((std::istreambuf_iterator<char>(cfgJson)), std::istreambuf_iterator<char>());
                try
                {
                    advancedDevice.load_json(jsonString);
                    std::cout << "Initialized the camera from " << config.camera_settings_json_file() << std::endl;
                    return;
                }
                catch (const rs2::error & e)
                {
                    std::cerr << "Error parsing advanced camera settings from " << config.camera_settings_json_file() << ": " << std::endl
                              << e.get_failed_function() << "(" << e.get_failed_args() << "):\n    " << e.what() << std::endl;
                }
            }
            else
            {
                std::cerr << "Cannot read advanced camera settings from " << config.camera_settings_json_file() << std::endl;
            }
        }
```
But it still does not work:
Here is the output:
```
After evetually toggle the advance_mode: is_enabled: 1
Error parsing advanced camera settings from camera_settings.json: 
rs2_load_json(dev:0x26b0eb0, json_content:0x28529d0, content_size:3418):
    Operation failed with error code=1
Using RS2_RS400_VISUAL_PRESET_HIGH_ACCURACY as fallback!
```
Do you have any other suggestions?
Thanks in advance,
Florian

sans-ltd, 2020-08-12:
We have a new insight about the issue: 
```loadAvancedCameraConfigJson()``` is called twice during the startup. On Windows, this works, on Linux this leads to IO errors on the second call.
It might be even called again during runtime, if a change in the configuration was detected.

See the attached strace.log for details:
[strace.log](https://github.com/IntelRealSense/librealsense/files/5061786/strace.log)
Some remarks about the attacted strace.log:
Line 9359: beginning of the working setting transmission
Line 10456: Transmission finished.
Line 16235: Begin of the failing tansmission
Line 16838: End of the failing transmission
Line 16823: The mentioned IO error.

So now the question: What are the preconditions to do this on Linux?
Regards
Florian

MartyG-RealSense, 2020-08-12:
I considered your question very carefully.  It is best if I refer it to a programming specialist on the RealSense team though.  @dorodnic Would you be able to kindly provide suggestions about the question above please?  Thanks!

MartyG-RealSense, 2020-09-03:
Reminder:  @dorodnic Can you take a look at the question above by @sans-ltd please? Thanks!

MartyG-RealSense, 2020-09-18:
I apologise that you did not receive a furthwr response.  Did you solve the problem in the meantime please?

MartyG-RealSense, 2020-09-26:
Case closed due to no further comments received.

closed 2020-09-26 by MartyG-RealSense

sans-ltd, 2020-09-29:
Hello again,
no we did not find a way to technically fix the issue, no work around, nothing. We fixed it in an organisatoric way because this functionality is only needed during development and not in production use.
Regards
Florian