realsenseai/librealsense #7093: Error flashing firmware backup

jb455, 2020-08-11:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Firmware Version                   | 05.12.05.00 | 
| Operating System & Version |   Win10  | 
| Platform                                 | PC |
| SDK Version                            |  2.36.0}                          | 

Hi, I'm trying to build firmware updating into our app. The updating itself seems fine but I thought it would be safe to take a backup before updating, just in case. The backup seems to work ok but if I try to install the backup either from the viewer or rs-fw-update.exe it fails with an error:
```
C:\Program Files (x86)\Intel RealSense SDK 2.0\tools>rs-fw-update.exe -b D:\backup.bin

backing-up device flash:
flash backup progress: 100[%]
C:\Program Files (x86)\Intel RealSense SDK 2.0\tools>rs-fw-update.exe -f D:\backup.bin

updating device:
Name: Intel RealSense D435, serial number: 818312071113, update serial number: 822213020241, firmware version: 05.12.05.00, USB type: 3.2
 11/08 17:04:24,458 ERROR [5452] (types.h:308) hr returned: HResult 0x8007001f: "A device attached to the system is not functioning."

firmware update started

RealSense error calling rs2_update_firmware_cpp(device:000001EB604B9100, fw_image:000001EB610F4060):
    Device: 822213020241 failed to download firmware
Please verify that no other librealsense application is running
```
I get the same if I first flash a different FW image between backup and flashing the backup (so it's not caused by trying to apply the same FW as is currently on the device).
Is this a bug, or is it expected and there's another way to revert to the backed up image?
Thanks

MartyG-RealSense, 2020-08-11:
hi @jb455  In the librealsense file **rs_device.hpp** there is a note about creating a flash backup that says:

**Create backup of camera flash memory.  Such backup does not constitute valid firmware image, and cannot be loaded back to the device, but it does contain all calibration and device information**

https://github.com/IntelRealSense/librealsense/blob/master/include/librealsense2/hpp/rs_device.hpp#L189

There is a case that discusses downloading firmwares from an Amazon AWS S3 link, though this may not be practical if your application cannot be dependent on an internet connection:

https://github.com/IntelRealSense/librealsense/issues/5114

dorodnic, 2020-08-11:
Hi @jb455 
To expand on @MartyG-RealSense answer - the reason it doesn't work is because end-customer devices require any firmware file to be digitally signed by Intel before flashing. There are two reasons for the back-up:
1. Have backup of all calibration tables, in case there is some kind of bug in the future
2. Some customers may have experimental devices that do accept unsigned firmware (in case your company is in-touch with us)

This is probably not very helpful, but at least the reasons are communicated

jb455, 2020-08-12:
Thanks for your help both.
What do you recommend our app does if flashing a new image fails for some reason?

MartyG-RealSense, 2020-08-12:
Maybe have a copy of a firmware bin file in your project and use an adaptation of the recovery code in the **rs-fw-update** program to perform recovery using that bin file?

https://github.com/IntelRealSense/librealsense/blob/master/tools/fw-update/rs-fw-update.cpp#L161

jb455, 2020-08-12:
Thanks!
One small further thing: I don't have an L515 to hand at the moment, am I safe to assume it follows the same format as D400 and shows up as "L5xx Recovery" when in recovery mode?

MartyG-RealSense, 2020-08-12:
@jb455   I'm not currently handling L515 related questions, so I will have to defer to @dorodnic on that one.    :)

dorodnic, 2020-08-13:
Yes, L500 firmware update flow is exactly the same and uses same APIs

jb455, 2020-08-13:
Ok thanks, I'm now using `device.Is(Extension.UpdateDevice)` to determine if the device is in recovery mode, though I still need to use the name to determine which FW to flash (D400 or L500). I notice in the Viewer code that it uses `RS2_CAMERA_INFO_PRODUCT_LINE` for this purpose, though this isn't currently available in the C# wrapper.
Thanks for all your help both!

closed 2020-08-13 by jb455

rupak-d, 2021-05-20:
Got a similar message during the update process on one of our cameras:
`Current Version:  5.12.6.0
FW Version mismatch. Updating camera.
Device: 935523025577 failed to download firmware
Please verify that no other librealsense application is running`

It was trying to update to 5.12.10. Is there a reason this happens?

MartyG-RealSense, 2021-05-20:
Hi @rupak-d   As mentioned earlier in this discussion, the firmware that is being updated to should be a digitally signed driver and not a firmware backup file.

https://github.com/IntelRealSense/librealsense/issues/7093#issuecomment-672186337

If you are using your own firmware update script and are updating to **5.12.10.0** using a firmware file, was the file obtained from the official firmware releases page please?

https://dev.intelrealsense.com/docs/firmware-releases#section-d400-series-firmware-downloads

rupak-d, 2021-05-20:
Yes the firmware file was obtained from the official release pages.

MartyG-RealSense, 2021-05-20:
There was a case where this error occurred due to performing the firmware update on a virtual machine (VM).  Does this possibility apply to your own situation please?

https://support.intelrealsense.com/hc/en-us/community/posts/360052995513-L515-Unable-to-update-firmware

rupak-d, 2021-05-20:
No it is on a Linux based system.

MartyG-RealSense, 2021-05-20:
VMs can be based around Linux too, as mentioned in the link below.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#linux-ubuntu-installation

You are confirming though that a Virtual Machine was definitely not being used in this particular case?

rupak-d, 2021-05-20:
yes, there is no virtual machine involved here.

MartyG-RealSense, 2021-05-20:
There are few programming references available for scripting firmware update into an application, as it is recommended that an official firmware update solution (the RealSense Viewer or the **rs-fw-update** tool) is used if possible.  

If updating firmware from within your own application is essential, the link below is the main other coding reference for firmware updating that I know of other than this case here.

https://github.com/IntelRealSense/librealsense/issues/4416