realsenseai/librealsense #7095: update firmware of D430

aitazhixin, 2020-08-12:
OS: Android O
Device: Two D430
Firmware: 5.11.11.100

Problem: Update firmware for these two device continuously, the second will stay in recovery mode after finished updating. If re-update in recovery mode on the time, it will throw: 
`rs2_create_device(info_list:0x782ce1b700, index:1):
        DFU detach failed!`

MartyG-RealSense, 2020-08-12:
Hi @aitazhixin  When you say that the firmware is being updated continuously, do you mean that you are deliberately updating it continuously, please?  For example, as a 'stress test' to confirm that a firmware update will work consistently.

If you are performing continuous updates, I would advise against it.  This is because flash memory (the memory storage inside the camera where firmware is written to) degrades a little each time that is is written to.  Under normal circumstances it may take a long time to degrade but the more 'write cycles' that occur, the faster that it degrades.   I have broken quite a few USB flash drive sticks within a few weeks of purchasing them by writing 30 GB worth of files per day to them.

In the case of the particular **DFU deatched failed!** error that you encountered, it looks as though it is thrown by the librealsense file **fw-update-device.cpp** when the DFU is detected not to be in an idle state.

This section of the firmware code received some bug-fix changes in February 2020.  So it was probably merged into the SDK around the time of SDK version **2.34.0**.

https://github.com/IntelRealSense/librealsense/pull/5826/files/af9afa9f3e8eb4e11aa9ede6f67edee1f2c96482

Which version of the RealSense SDK are you using, please?   I ask this because I note that you are using firmware **5.11.11.100**, which is seven firmware releases old at the time of writing this.

dorodnic, 2020-08-12:
Hi @aitazhixin 
Couple of things to consider -
1. Firmware update has a counter of sequential downgrades (including same version) after which it will only allow to update to a newer firmware version. This is done to protect the flash, see @MartyG-RealSense's comment.
2. On Android, you need to be careful when working with multiple devices. USB-Host (Android library similar to libusb) requires exclusive lock on any device as far as I know, and this may cause issues with multi-camera connection and disconnection scenarios, like this one. I think that the application needs to open the device after going to update state to read its serial number.

aitazhixin, 2020-08-20:
@MartyG-RealSense Thanks for your answer. My 'continuously' means: I upgrade firmware for second RealSense device after the first device upgraded. The second device is upgraded, it keeps in recovery mode.

MartyG-RealSense, 2020-08-20:
Hi @aitazhixin   I note that you did not confirm the method that you used to update the firmware.  Did you use the firmware updater in the RealSense Viewer, or did you use the **rs-fw-update** tool please?

If you upgraded the firmware in the RealSense Viewer, then the **rs-fw-update** tool has a recovery method for cameras that are in a recovery state.

https://support.intelrealsense.com/hc/en-us/community/posts/360034110333/comments/360008716253

aitazhixin, 2020-08-24:
Here is my codes to update two device which updated from rs-fw-update.cpp in librealsense.

    SwitchArg all_arg("a", "allupdate", "update all connected devices");

    cmd.add(all_arg);

    if (all_arg.isSet()) {
        std::vector<std::string> serials;
        auto devices = ctx.query_devices(RS2_PRODUCT_LINE_DEPTH);
        for (auto&& dvs : devices) {
            serials.push_back(dvs.supports(RS2_CAMERA_INFO_SERIAL_NUMBER) ? dvs.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER) : "unknown");
        }
        for (auto && serial : serials) {

            rs2::device loc_new_device;
            rs2::update_device loc_new_fw_update_device;
            bool updated = false;
            // Update device
            ctx.set_devices_changed_callback([&](rs2::event_information& info)
            {
                if (info.get_new_devices().size() == 0)
                {
                    return;
                }

                for (auto&& d : info.get_new_devices())
                {
                    std::lock_guard<std::mutex> lk(mutex);
                    if (d.is<rs2::update_device>() && (d.get_info(RS2_CAMERA_INFO_FIRMWARE_UPDATE_ID) == update_serial_number))
                        loc_new_fw_update_device = d;
                    else
                        loc_new_device = d;
                }
                if(loc_new_fw_update_device || loc_new_device)
                    cv.notify_one();
            });

            auto devs = ctx.query_devices(RS2_PRODUCT_LINE_DEPTH);

            bool device_found = false;

            for (auto&& d : devs)
            {
                if (!d.is<rs2::updatable>() || !(d.supports(RS2_CAMERA_INFO_SERIAL_NUMBER) && d.supports(RS2_CAMERA_INFO_FIRMWARE_UPDATE_ID)))
                    continue;

                if (d.supports(RS2_CAMERA_INFO_USB_TYPE_DESCRIPTOR))
                {
                    std::string usb_type = d.get_info(RS2_CAMERA_INFO_USB_TYPE_DESCRIPTOR);
                    if (usb_type.find("2.") != std::string::npos) {
                        std::cout << std::endl << "Warning! the camera is connected via USB 2 port, in case the process fails, connect the camera to a USB 3 port and try again" << std::endl;
                    }
                }

                update_serial_number = d.get_info(RS2_CAMERA_INFO_FIRMWARE_UPDATE_ID);

                auto sn = d.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER);
                if (sn != serial && devs.size() != 1)
                    continue;
                device_found = true;
                auto fw = d.get_info(RS2_CAMERA_INFO_FIRMWARE_VERSION);

                if (backup_arg.isSet())
                {
                    std::cout << std::endl << "backing-up device flash: " << std::endl;

                    auto flash = d.as<rs2::updatable>().create_flash_backup([&](const float progress)
                    {
                        printf("\rflash backup progress: %d[%%]", (int)(progress * 100));
                    });
[... 68 lines cut ...]
                        if (update_serial == update_serial_number) {
                            find_update_device = true;
                        }
                        else {
                            continue;
                        }
                        if (sn != serial)
                            continue;

                        auto fw = d.supports(RS2_CAMERA_INFO_FIRMWARE_VERSION) ? d.get_info(RS2_CAMERA_INFO_FIRMWARE_VERSION) : "unknown";
                        std::cout << std::endl << "device " << sn << " successfully updated to FW: " << fw << std::endl;
                    }

                    if (!find_update_device) {
                        std::this_thread::sleep_for(std::chrono::milliseconds(400));
                    }
                    else {
                        break;
                    }
                }
            }
            std::this_thread::sleep_for(std::chrono::seconds(4));
        }
        return EXIT_SUCCESS;
    }

MartyG-RealSense, 2020-08-24:
Hi @aitazhixin   Was the script above the method that you originally used to update your two cameras please? (the one that worked and the one that went into recovery mode)

MartyG-RealSense, 2020-09-03:
Hi @aitazhixin    Do you require further assistance with this case please?   Thanks!

aitazhixin, 2020-09-11:
> Hi @aitazhixin Was the script above the method that you originally used to update your two cameras please? (the one that worked and the one that went into recovery mode)

Yes, using the script, First Device can work after updated firmware. The second device into recovery mode.

MartyG-RealSense, 2020-09-11:
What might put a camera into recovery mode during a firmware update is if some event occurs to interrupt the update process before it is 100% completed.   I would refer back to the comment of @dorodnic earlier in this discussion about being careful when doing multi-camera connection and disconnection on Android.

https://github.com/IntelRealSense/librealsense/issues/7095#issuecomment-672818385

The **rs-fw-update** tool has an "-r" condition that recovers all connected cameras that are in recovery mode.  Perhaps after the first firmware update attempt has been made, an adaptation of that recovery code from **rs-fw-update** could be run to try to re-attempt the process on any cameras left in recovery (it should ignore all the cameras that were successfully upgraded and not in recover mode).  This would be like a failsafe process.

MartyG-RealSense, 2020-09-29:
Case closed due to no further commente received.

closed 2020-09-29 by MartyG-RealSense

etorbenson, 2021-07-14:
@MartyG-RealSense  - We're experiencing this same issue on Windows systems updating two D435s or D435is sequentially. Using Intel.RealSense.FW.Update.exe and a signed FW file, the first camera upgrades fine, but the second camera is put into recovery mode. Using the "-r" switch does eventually work but it takes several tries. Any hints on how to account for this during an automated upgrade job?

MartyG-RealSense, 2021-07-14:
Hi @etorbenson  As far as I am aware, the update tool was designed for updating one camera at a time instead of batch updating.  It looks as though the multiple-camera aspect of this case revolves around an adaptation of the rs-fw-update tool that was created by @aitazhixin in https://github.com/IntelRealSense/librealsense/issues/7095#issuecomment-678954542 - so the code changes would not be included in the official pre-built **Intel.RealSense.FW.Update.exe** version of the tool.

My understanding of this case is that the above script adaptation of rs-fw-update was also experiencing problems with updating the firmware of more than one camera, but there was not a conclusion to the case and so a fix was not posted here.

Whilst rs-fw-update or the RealSense Viewer are the recommended methods of updating firmware, the RealSense SDK does also support writing your own firmware update script, like the one in the link below.

https://github.com/IntelRealSense/librealsense/issues/4416#issuecomment-511481279

etorbenson, 2021-07-14:
Thanks @MartyG-RealSense - what I'm wondering is why it happens at all. All of the different methods used to flash the firmware are accessing the same libraries and such. Has anyone been able to reproduce this issue on Intel's side? We could work around it but it would be great if we didn't need to. (i.e. if we could identify a problem and get a fix.) I've got an issue where I have tons of field devices running Windows and each having 2 RealSenses that need an unattended way to upgrade firmware without having to worry about recovery mode.

MartyG-RealSense, 2021-07-14:
I will seek advice from Intel about your question.  Thanks for your patience!

RealSenseSupport, 2021-07-14:
Hi, I ran some quick tests on SDK 2.47 with two cameras and perform the update on my Win 10 system, no problem so far. 
Does the second camera consistently end up in the recovery state and on which version of the SDK.

MartyG-RealSense, 2021-07-15:
@RealSenseSupport   Thanks for your advice  :)

@etorbenson  Thanks very much for your very detailed report in https://github.com/IntelRealSense/librealsense/issues/9395 of your issue.  I recommend using that issue as a reference but continuing the support discussion here so that it takes place in a single location.