realsenseai/librealsense #7100: What is the real-world xyz value of the center pixel?

jpenaloza1211, 2020-08-13:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | (05.12.03.00)  | 
| Operating System & Version |   {Win 10}  | 
| Kernel Version (Linux Only)    |  -                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.33.1                          | 
| Language                            |  {python }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
Simply put, if I calculate the center pixel's xyz using the deproject method, should this xyz value be [0, 0, depth] ? 
I believe I have seen this to be approximately true, but recently this is not what I am getting. 

To demonstrate this issue, I am using the simple code at the following link: https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/opencv_viewer_example.py

The only changes I made to it are that between converting the images to numpy arrays and stacking the two images, I have the following:

        # Apply colormap on depth image (image must be converted to 8-bit per pixel first)
        colorizer = rs.colorizer()
        depth_colormap = np.asanyarray(colorizer.colorize(depth_frame).get_data())

        size = 1
        t = 1
        increment = 25
        color = (255, 255, 255)

        depth_intrinsics = depth_frame.profile.as_video_stream_profile().intrinsics
        center_pt = (424, 240)
        center_xyz = rs.rs2_deproject_pixel_to_point(depth_intrinsics, [center_pt[1], center_pt[0]],
                                                     depth_image[center_pt[1], center_pt[0]])
        center_xyz = np.array(center_xyz)
        # center_xyz = center_xyz.astype('int')

        cv2.circle(depth_colormap, center_pt, 4, color, -1)
        mstring = "Center pt xyz: "
        cv2.putText(color_image, mstring, (5, increment * 6), cv2.FONT_HERSHEY_SIMPLEX, size, color, t)
        cv2.putText(color_image, str(center_xyz), (5, increment * 7), cv2.FONT_HERSHEY_SIMPLEX, size, color, t)

All this does is find the center pixel on the depth image and determines the xyz using the deproject approach while also displaying a bunch of information the image window for me to debug as shown in the following image. (the center pixel is represented by the white dot on the depth image)

As seen on the following image, the center pixel on my (848x480) depth image is not approximately 0,0,depth
I don't know if I am missing something or if there is something wrong with my camera. Any help would be appreciated!

ps. the depth value is accurate from my measurements
pss. the camera is pointed at mostly flat wall (with an object a bit in the way at the bottom right of the depth image)

[image]

ev-mp, 2020-08-13:
@jpenaloza1211  hello, 
The (i,j) coordinates passed to the function appear to be swapped and not point to the center:
>center_pt = (424, 240)  <- This is correct, Column 424, Row 240
center_xyz = rs.rs2_deproject_pixel_to_point(depth_intrinsics, [**center_pt[1], center_pt[0]**],... - ?

Can you check?

jpenaloza1211, 2020-08-13:
hi @ev-mp ,

wow, yup, that seems to be the problem. I've been going crazy trying to figure this out.

so just to confirm then, the input point to deproject is [x,y] and not [row, column]?

ev-mp, 2020-08-13:
That's correct: (x,y)->(column,row).

jpenaloza1211, 2020-08-13:
thanks a ton!

I have another question similar to this if you happen to know. 

If I want to get the xyz of every pixel, do I have to run this function on loop for every pixel or is there a simpler function call that I can use to get back a complete xyz array?

ev-mp, 2020-08-14:
@jpenaloza1211 , absolutely.
The use-case of calculating [x,y,z] for each pixel of depth frame is known as `3D-pointcloud` generation.
You can achieve it by using the following flow (adapted from [rs-pointcloud demo](https://github.com/IntelRealSense/librealsense/tree/master/examples/pointcloud)).
The 3D pointcloud can further can be enhanced by calculating the RGB texture map, though I removed it from the snippet below for clarity
```cpp
    // Declare pointcloud object, for calculating pointclouds and texture mappings
    rs2::pointcloud pc;
    // Declare RealSense pipeline, encapsulating the actual device and sensors
    rs2::pipeline pipe;
    // Start streaming with default recommended configuration
    pipe.start();

    while (app) 
    {
        // Wait for the next set of frames from the camera
        auto frames = pipe.wait_for_frames();

        if (auto depth = frames.get_depth_frame())
        {
            // Generate the pointcloud and texture mappings
            points = pc.calculate(depth);

            // vertices_buffer points to a buffer of 3D vertices [x,y,z] for each pixel in the original frame.
            // For areas with no depth data available the vertices will be set to [0,0,0]
            auto vertices_buffer = points.get_data();  
```

The pointcloud class also has SSE and CUDA implementations that can be opted in at CMake configuration stage.

jpenaloza1211, 2020-08-14:
@ev-mp 

Does this functionality exist in the python wrapper? 
If so, would it follow the same flow as above?

RealSenseSupport, 2020-08-19:
@jpenaloza1211 For python wrapper, please refer to https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/opencv_pointcloud_viewer.py#L292  Thanks!

RealSenseSupport, 2020-09-04:
@jpenaloza1211 Any other questions about this? Looking forward to your update. Thanks!

jpenaloza1211, 2020-09-04:
@RealSenseSupport no more questions, forgot to close the issue. Thank you!

closed 2020-09-04 by jpenaloza1211