realsenseai/librealsense #7101: Health-Check of On-Chip Calibration shows the negative value

IanChou617, 2020-08-13:
Hi, I do the on-chip calibration for my kit, and the health-check shows the negative value as below:
[image]
but the indication represents that the value is larger than 0
[image]

agrunnet, 2020-08-13:
The thing that matters is the absolute value (which should have been made more clear). So you unitsnin range -0.15 to +0.15 are fine.

IanChou617, 2020-08-13:
But it still shows good when the value is -0.35
[image]

agrunnet, 2020-08-13:
That is not good. That was a bug that was fixed a while ago. Are you using latest version of FW and SDK?

IanChou617, 2020-08-13:
> That is not good. That was a bug that was fixed a while ago. Are you using latest version of FW and SDK?

Thanks a lot!!! The result is correct now

agrunnet, 2020-08-13:
Great

closed 2020-08-13 by agrunnet