realsenseai/librealsense #7102: Changing RGB Options in MATLAB Wrapper

bauder96, 2020-08-13:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.12.06.00| 
| Operating System & Version |   Windows 10  | 
| Kernel Version (Linux Only)    |  N/A                                         | 
| Platform                                 | PC |
| SDK Version                            |                            | 
| Language                            | MATLAB                        | 
| Segment			|                          | 

### Issue Description
Good afternoon,

I am using the MATLAB wrapper to control and take in data from my D415. I am currently taking in a frame from the RGB camera, however the image is too dark to see anything. It is the same as when I use the RealSense Viewer and turn auto-exposure off. So, I am trying to change the options of the RGB camera. However, I cannot figure out how to do this. Here is my method:

`dev = profile.get_device();`
`sensor = dev.first('SENSOR NAME');`
`sensor.set_option(realsense.option.enable_auto_exposure,0);`
`sensor.set_option(realsense.option.exposure,156);`

The problem is, the only sensor types in the wrapper appear to be "sensor", "roi_sensor", "depth_sensor", and "depth_stereo_sensor". So when I run this, the exposure of the camera remains unchanged. 

Does anyone have a solution?

Thanks!

MartyG-RealSense, 2020-08-13:
Hi @bauder96    The best scripting reference that I could find for setting sensor options in the MATLAB wrapper is information provided by a RealSense team member in the link below:

https://github.com/IntelRealSense/librealsense/issues/3358#issuecomment-475489478

bauder96, 2020-08-13:
Hi Marty,

Thanks for your help. I did end up finding a solution (sort of). I am able to edit the settings of the RGB sensor. However, enabling auto exposure does not fix the dark image issue, so I had to set it manually. Not sure why auto exposure is not working. Here is my solution:
     `pipe = realsense.pipeline();`
     `profile = pipe.start();`
    `dev = profile.get_device();`
    `sensors = dev.query_sensors();`
    `sname = sensors{1}{2}.get_info(realsense.camera_info.name);`
    `disp(sname);`
    `sensors{1}{2}.set_option(realsense.option.enable_auto_exposure,0);`
    `sensors{1}{2}.set_option(realsense.option.exposure,500);`

closed 2020-08-13 by bauder96

MartyG-RealSense, 2020-08-13:
Hi @bauder96   Great news that you made progress!

Assuming that auto-exposure is active, you may be able to increase brightness using a setting called **Mean Intensity Set Point**.  Increasing the value should increase brightness.

You can test this in the RealSense Viewer program by expanding open the **Advanced Controls** section of the Viewer's options side panel and expanding open its **AE Control** sub-category.