realsenseai/librealsense #7104: Unable to execute this command:./scripts/patch-realsense-ubuntu-lts.sh

yy0626, 2020-08-13:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

yy0626, 2020-08-13:
**Required Info |**  
Camera` Model                     | { D435i }
Operating System & Version | {Ubuntu 16.04 LTS}
Kernel Version (Linux Only)  | (4.14.13-tegra)
Platform                                | NVIDIA Jetson TX2(JetPack3.3.4)
SDK Version                         | { legacy / 2.. }

**Issue Description**
I installed the D435i sdk driver on TX2 step by step via https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md

When reaching this step, /scripts/patch-realsense-ubuntu-lts.sh, the terminal always report an error and cannot continue the installation:

nvidia@tegra-ubuntu:~/librealsense$ ./scripts/patch-realsense-ubuntu-lts.sh

Remove all RealSense cameras attached. Hit any key when ready

Reading package lists... Done
Building dependency tree       
Reading state information... Done
git is already the newest version (1:2.7.4-0ubuntu1.9).
linux-headers-generic is already the newest version (4.4.0.186.192).
bc is already the newest version (1.06.95-9build1).
build-essential is already the newest version (12.1ubuntu2).
0 upgraded, 0 newly installed, 0 to remove and 277 not upgraded.
Package required libusb-1.0-0-dev:  - found
Package required libssl-dev:  - found
Create patches workspace in ubuntu-xenial-master folder

Fetching Ubuntu LTS tag   to the local kernel sources folder
fatal: You need to specify a tag name.

ev-mp, 2020-08-13:
@yy0626 hello,
The mentioned guide applies to Ubuntu/x86. The Jetsons installation guide is available at documentation folder  - https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md

yy0626, 2020-08-13:
> @yy0626 hello,
> The mentioned guide applies to Ubuntu/x86. The Jetsons installation guide is available at documentation folder - https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md

Thank you very much for your comment.
I re-installed according to the guide you mentioned. When I was running "realsense-viewer", I found that the camera could not be connected with a USB3.0 hub. Repeatedly removing and plugging in the camera had no effect. 

"nvidia@tegra-ubuntu:~$ sudo apt-key adv --keyserver keys.gnupg.net --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE || sudo apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE
Executing: /tmp/tmp.bZ6gajiXk5/gpg.1.sh --keyserver
keys.gnupg.net
--recv-key
F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE
gpg: requesting key 6F3EFCDE from hkp server keys.gnupg.net
gpg: key 6F3EFCDE: ""CN = Intel(R) Intel(R) Realsense", O=Intel Corporation" not changed
gpg: Total number processed: 1
gpg:              unchanged: 1
nvidia@tegra-ubuntu:~$ sudo add-apt-repository "deb http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial main" -u
nvidia@tegra-ubuntu:~$ sudo apt-get install librealsense2-utils
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following additional packages will be installed:
  librealsense2 librealsense2-gl librealsense2-udev-rules
The following NEW packages will be installed:
  librealsense2 librealsense2-gl librealsense2-udev-rules librealsense2-utils
0 upgraded, 4 newly installed, 0 to remove and 277 not upgraded.
Need to get 16.5 MB of archives.
After this operation, 45.5 MB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial/main arm64 librealsense2 arm64 2.28.1-5ubuntu5~1.gbpb2a8ce [7,191 kB]
Get:1 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial/main arm64 librealsense2 arm64 2.28.1-5ubuntu5~1.gbpb2a8ce [7,191 kB]
Get:2 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial/main arm64 librealsense2-gl arm64 2.28.1-5ubuntu5~1.gbpb2a8ce [1,010 kB]
Get:3 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial/main arm64 librealsense2-udev-rules arm64 2.28.1-5ubuntu5~1.gbpb2a8ce [8,072 B]
Get:4 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial/main arm64 librealsense2-utils arm64 2.28.1-5ubuntu5~1.gbpb2a8ce [8,322 kB]
Get:4 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial/main arm64 librealsense2-utils arm64 2.28.1-5ubuntu5~1.gbpb2a8ce [8,322 kB]
Fetched 8,915 kB in 30min 45s (4,830 B/s)                                      
Selecting previously unselected package librealsense2:arm64.
(Reading database ... 221519 files and directories currently installed.)
Preparing to unpack .../librealsense2_2.28.1-5ubuntu5~1.gbpb2a8ce_arm64.deb ...
Unpacking librealsense2:arm64 (2.28.1-5ubuntu5~1.gbpb2a8ce) ...
Selecting previously unselected package librealsense2-gl:arm64.
Preparing to unpack .../librealsense2-gl_2.28.1-5ubuntu5~1.gbpb2a8ce_arm64.deb ...
Unpacking librealsense2-gl:arm64 (2.28.1-5ubuntu5~1.gbpb2a8ce) ...
Selecting previously unselected package librealsense2-udev-rules:arm64.
Preparing to unpack .../librealsense2-udev-rules_2.28.1-5ubuntu5~1.gbpb2a8ce_arm64.deb ...
Unpacking librealsense2-udev-rules:arm64 (2.28.1-5ubuntu5~1.gbpb2a8ce) ...
Selecting previously unselected package librealsense2-utils:arm64.
Preparing to unpack .../librealsense2-utils_2.28.1-5ubuntu5~1.gbpb2a8ce_arm64.deb ...
Unpacking librealsense2-utils:arm64 (2.28.1-5ubuntu5~1.gbpb2a8ce) ...
Processing triggers for libc-bin (2.23-0ubuntu11) ...
Setting up librealsense2:arm64 (2.28.1-5ubuntu5~1.gbpb2a8ce) ...
Setting up librealsense2-gl:arm64 (2.28.1-5ubuntu5~1.gbpb2a8ce) ...
Setting up librealsense2-udev-rules:arm64 (2.28.1-5ubuntu5~1.gbpb2a8ce) ...
Setting up librealsense2-utils:arm64 (2.28.1-5ubuntu5~1.gbpb2a8ce) ...
Processing triggers for libc-bin (2.23-0ubuntu11) ...
nvidia@tegra-ubuntu:~$ sudo apt-get install librealsense2-dev
[sudo] password for nvidia: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following NEW packages will be installed:
  librealsense2-dev
0 upgraded, 1 newly installed, 0 to remove and 277 not upgraded.
Need to get 5,203 kB of archives.
After this operation, 14.8 MB of additional disk space will be used.
Get:1 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial/main arm64 librealsense2-dev arm64 2.28.1-5ubuntu5~1.gbpb2a8ce [5,203 kB]
Get:1 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial/main arm64 librealsense2-dev arm64 2.28.1-5ubuntu5~1.gbpb2a8ce [5,203 kB]
Get:1 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial/main arm64 librealsense2-dev arm64 2.28.1-5ubuntu5~1.gbpb2a8ce [5,203 kB]
Get:1 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial/main arm64 librealsense2-dev arm64 2.28.1-5ubuntu5~1.gbpb2a8ce [5,203 kB]
Get:1 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial/main arm64 librealsense2-dev arm64 2.28.1-5ubuntu5~1.gbpb2a8ce [5,203 kB]
Get:1 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial/main arm64 librealsense2-dev arm64 2.28.1-5ubuntu5~1.gbpb2a8ce [5,203 kB]
Fetched 1,170 kB in 56min 1s (348 B/s)                                         
[... cut ...]
 13/08 23:55:22,300 WARNING [548000329728] (ds5-factory.cpp:1119) DS5 group_devices is empty.
 13/08 23:55:22,504 WARNING [548000329728] (ds5-factory.cpp:1119) DS5 group_devices is empty.
 13/08 23:55:22,725 WARNING [548000329728] (ds5-factory.cpp:1119) DS5 group_devices is empty.
 13/08 23:55:22,938 WARNING [548000329728] (ds5-factory.cpp:1119) DS5 group_devices is empty.
"
Is it because I have installed Realsense-viewer V2.36 before, so do I need to uninstall the previous one before reinstalling?
What should I do?

ev-mp, 2020-08-13:
@yy0626  hello,
In case of multiple versions verify that the version you're running with `$which realsense-viewer`.
D435i should be recognized and run with the `viewer` from the Debian installer you've downloaded with no further modifications.

As for the`empty group` warning - if this message is encountered when building v2.36 from sources it make indicate a non-compatible OS-interface module was selected.  Make sure to apply `-DFORCE_RSUSB_BACKEND=true` and  "-DBUILD_WITH_CUDA=true" cmake flag and rebuild if needed.

yy0626, 2020-08-14:
> @yy0626 hello,
> In case of multiple versions verify that the version you're running with `$which realsense-viewer`.
> D435i should be recognized and run with the `viewer` from the Debian installer you've downloaded with no further modifications.
> 
> As for the`empty group` warning - if this message is encountered when building v2.36 from sources it may indicate a non-compatible OS-interface module being selected. Make sure to apply "-DFORCE_RSUSB_BACKEND=true" and "-DBUILD_WITH_CUDA=true" cmake flag and rebuild if needed.

Running `$which realsense-viewer` and output '/usr/local/bin/realsense-viewer', which is V.2.36 installed for the first time. 
So how can I run the real-viewer from the Debian installer I've downloaded ?

ev-mp, 2020-08-14:
The standard installation path for Debians is `/usr/bin`, therefore imo `$/usr/bin/librealsense-viewer` should be sufficient.
Alternatively the path can be established with `$dpkg -L librealsense2-utils`

yy0626, 2020-08-14:
> The standard installation path for Debians is `/usr/bin`, therefore imo `$/usr/bin/librealsense-viewer` should be sufficient.
> Alternatively the path can be established with `$dpkg -L librealsense2-utils`

Thank you for your patient answers, now I can use it.

ev-mp, 2020-08-16:
@yy0626 , thanks for the confirmation.
Closing as resolved.

closed 2020-08-16 by ev-mp