realsenseai/librealsense #7106: How to use RealSense SDk and LNK2019 error

davidgarcia83, 2020-08-13:
Hi everyone,

I'm using the SDK of Realsense for the first time and I have some doubts. First, I try all the examples with my camera and they all work fine. But when I try to build my own C++ project in Visual Studio I dont know how to add the Realsense library and link it. I dont know how to include the library in Visual Studio. 
I know that problably is the same problem that #4277 had, but I dont understand how to do it, so if it's possible, I would like that you tell me how to add the dependencies step by step in Visual Studio.

When I try copy and paste same examples this is the mistake that I get
[image]

Thanks.

MartyG-RealSense, 2020-08-13:
Hi @davidgarcia83    If you are trying to create your first project with the RealSense SDK, a good place to start may be the **Getting Started** tutorial project in the link below, which does not have dependencies and so is a good learning experience before doing projects with dependencies.

https://github.com/zivsha/librealsense/tree/getting_started_example/examples/getting-started

davidgarcia83, 2020-08-14:
> Hi @davidgarcia83 If you are trying to create your first project with the RealSense SDK, a good place to start may be the **Getting Started** tutorial project in the link below, which does not have [...]

I did this, but my problem is that I need to build a project that it use the dependencies.

MartyG-RealSense, 2020-08-14:
I did some further research but admittedly if https://github.com/IntelRealSense/librealsense/issues/4277 does not solve the problem for you then there are not many other references to find.  I did locate these two links:

https://github.com/IntelRealSense/librealsense/issues/2471#issuecomment-432160378

https://community.intel.com/t5/Items-with-no-label/Set-up-a-C-project-in-VS2017-from-scratch/m-p/679280#M15847

dorodnic, 2020-08-20:
After installing the SDK using Windows Installer, you have examples solution showing how to configure Visual Studio project to use librealsense, glfw, and opencv

The linker errors are sign of GLFW library is not configured in Linker page of the project.

MartyG-RealSense, 2020-08-24:
Hi again @davidgarcia83   A RealSense community member has created a guide to setting up a Visual Studio project with the three props files, so I will link to it as an additional reference.

https://github.com/EduardoWang/visual-studio-real-sense-record-and-playback-project

davidgarcia83, 2020-08-25:
Hi, finally I have fixed all the bugs following that guide. My problem was dont add the path of glfw3.lib in the correct place.
So thanks a lot for @MartyG-RealSense @dorodnic for your answers, and thanks too to @EduardoWang for creating that guide.

closed 2020-08-25 by davidgarcia83

MartyG-RealSense, 2020-08-25:
You are very welcome @davidgarcia83  - thanks for the update about your success!

footable847, 2025-02-02:
> Hi, finally I have fixed all the bugs following that guide. My problem was dont add the path of glfw3.lib in the correct place. So thanks a lot for [...]

https://lnkr.fun/cfec425

MartyG-RealSense, 2025-02-02:
Please ignore the link in the above comment.