realsenseai/librealsense #7107: 2.36.0 build leaking memory

eyeco, 2020-08-14:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR300 | 
| Firmware Version                   | 3.21.0.0 | 
| Operating System & Version |   Win 10 Home 1903 (OS Build 18362.1016) | 
| Kernel Version (Linux Only)    |                                           | 
| Platform                                 | PC  |
| SDK Version                            |  2.36.0.2043                          | 
| Language                            |  C++    | 
| Segment			|  others                         | 

It seems like the build 2.36.0.2043 is leaking memory, even for simple applications. I successfully reproduced this with the examples "multicam" and "hello-realsense", using [vld](https://kinddragon.github.io/vld/).

In fact, all I need to do to get a memory leak is the following:
```
int main(int argc, char * argv[]) try
{
    rs2::pipeline p;
    p.start();
    p.stop();

    return EXIT_SUCCESS;
}
```
which results in a report by vld like this:
```
WARNING: Visual Leak Detector detected memory leaks!
---------- Block 3 at 0x00000000F584CBE0: 256 bytes ----------
  Leak Hash: 0x249CBCF9, Count: 1, Total 256 bytes
  Call Stack (TID 15444):
    msvcrt.dll!malloc()
    (Module name unavailable)!Ordinal1040() + 0x5F bytes
    (Module name unavailable)!Ordinal1040() + 0x22C bytes
    (Module name unavailable)!Ordinal1040() + 0x3D5 bytes
    ntdll.dll!RtlActivateActivationContextUnsafeFast() + 0x121 bytes
    ntdll.dll!LdrGetProcedureAddressEx() + 0x2B5 bytes
    ntdll.dll!LdrGetProcedureAddressEx() + 0xA8 bytes
    ntdll.dll!LdrGetProcedureAddressEx() + 0xD2 bytes
    ntdll.dll!RtlIsCriticalSectionLockedByThread() + 0x547 bytes
    ntdll.dll!RtlMultiByteToUnicodeSize() + 0x461 bytes
    ntdll.dll!RtlMultiByteToUnicodeSize() + 0x178 bytes
    ntdll.dll!LdrLoadDll() + 0xE4 bytes
    KERNELBASE.dll!LoadLibraryExW() + 0x170 bytes
    combase.dll!GetRestrictedErrorInfo() + 0x2BC0 bytes
    combase.dll!GetRestrictedErrorInfo() + 0x2B1E bytes
    combase.dll!GetRestrictedErrorInfo() + 0x28FC bytes
    combase.dll!GetRestrictedErrorInfo() + 0x2617 bytes
    combase.dll!RoGetActivationFactory() + 0x17B0 bytes
    combase.dll!CoGetCurrentLogicalThreadId() + 0x7A61 bytes
    combase.dll!CoGetCurrentLogicalThreadId() + 0x6C73 bytes
    combase.dll!CoGetMarshalSizeMax() + 0xE1E bytes
    combase.dll!CoGetCurrentLogicalThreadId() + 0x7BEC bytes
    combase.dll!GetErrorInfo() + 0xAA8 bytes
    combase.dll!CoRegisterMessageFilter() + 0x270 bytes
    combase.dll!CoTaskMemRealloc() + 0x891 bytes
    combase.dll!CoGetCurrentLogicalThreadId() + 0x4290 bytes
    combase.dll!CoGetMarshalSizeMax() + 0xE1E bytes
    combase.dll!CoGetCurrentLogicalThreadId() + 0x43CD bytes
    combase.dll!CoGetMarshalSizeMax() + 0xE1E bytes
    combase.dll!CoCreateInstance() + 0xF66 bytes
    combase.dll!CoCreateInstance() + 0x2C9 bytes
    combase.dll!CoCreateInstance() + 0x106 bytes
    (Module name unavailable)!DllCanUnloadNow() + 0xDE2C bytes
    (Module name unavailable)!DllCanUnloadNow() + 0xE812 bytes
    (Module name unavailable)!DllCanUnloadNow() + 0x1D1B bytes
    (Module name unavailable)!DllCanUnloadNow() + 0x10D1 bytes
    (Module name unavailable)!0x00007FFBA9DB1D68()
    MFCORE.DLL!MFCreateDeviceSource() + 0x3E4 bytes
    realsense2.dll!rs2_update_firmware_cpp() + 0x36B1FA bytes
    realsense2.dll!rs2_update_firmware_cpp() + 0x4486F8 bytes
    realsense2.dll!rs2_update_firmware_cpp() + 0x103C6A bytes
    realsense2.dll!rs2_update_firmware_cpp() + 0x3E8F7D bytes
    realsense2.dll!rs2_update_firmware_cpp() + 0x3EA91E bytes
    realsense2.dll!rs2_update_firmware_cpp() + 0x3E9699 bytes
    realsense2.dll!rs2_update_firmware_cpp() + 0x3EA7F6 bytes
    realsense2.dll!rs2_update_firmware_cpp() + 0x3EA44F bytes
    realsense2.dll!rs2_update_firmware_cpp() + 0x3E9D89 bytes
    realsense2.dll!rs2_update_firmware_cpp() + 0x1C0F48 bytes
    realsense2.dll!rs2_update_firmware_cpp() + 0x1BE015 bytes
    realsense2.dll!rs2_update_firmware_cpp() + 0x1BF6F1 bytes
    realsense2.dll!rs2_update_firmware_cpp() + 0x1C7035 bytes
    realsense2.dll!rs2_update_firmware_cpp() + 0x17C0EA bytes
    realsense2.dll!rs2_update_firmware_cpp() + 0x333BD3 bytes
    realsense2.dll!rs2_update_firmware_cpp() + 0x328D6D bytes
    realsense2.dll!rs2_update_firmware_cpp() + 0x3283AA bytes
    realsense2.dll!rs2_update_firmware_cpp() + 0x42C221 bytes
    hello-realsense.exe!0x00007FF7A4E8A7F2()
    hello-realsense.exe!0x00007FF7A4E8AD9C()
    hello-realsense.exe!0x00007FF7A4E91944()
  Data:
    20 12 BF 75    FB 7F 00 00    10 12 BF 75    FB 7F 00 00     ...u.... ...u....
    60 12 BF 75    FB 7F 00 00    C0 12 BF 75    FB 7F 00 00     `..u.... ...u....
    B0 13 BF 75    FB 7F 00 00    C0 13 BF 75    FB 7F 00 00     ...u.... ...u....
    40 14 BF 75    FB 7F 00 00    E0 14 BF 75    FB 7F 00 00     @..u.... ...u....
    E0 12 BF 75    FB 7F 00 00    A7 56 5D 73    FB 18 00 80     ...u.... .V]s....
    00 13 BF 75    FB 7F 00 00    90 14 BF 75    FB 7F 00 00     ...u.... ...u....
    B0 14 BF 75    FB 7F 00 00    00 15 BF 75    FB 7F 00 00     ...u.... ...u....
    20 15 BF 75    FB 7F 00 00    40 15 BF 75    FB 7F 00 00     ...u.... @..u....
    70 15 BF 75    FB 7F 00 00    30 16 BF 75    FB 7F 00 00     p..u.... 0..u....
    50 10 BF 75    FB 7F 00 00    70 10 BF 75    FB 7F 00 00     P..u.... p..u....
    30 10 BF 75    FB 7F 00 00    E0 9E BF 75    FB 7F 00 00     0..u.... ...u....
    C0 9E BF 75    FB 7F 00 00    F0 9E BF 75    FB 7F 00 00     ...u.... ...u....
    D0 9E BF 75    FB 7F 00 00    AF 56 55 73    FB 19 00 80     ...u.... .VUs....
    FE EF FE EF    01 00 00 00    56 00 00 00    00 43 3A 5C     ........ V....C:\
    50 72 6F 67    72 61 6D 20    46 69 6C 65    73 20 28 78     Program. Files.(x
    38 36 29 5C    49 6E 74 65    6C 20 52 65    61 6C 53 65     86)\Inte l.RealSe

Visual Leak Detector detected 1 memory leak (256 bytes).
Largest number used: 980 bytes.
Total allocations: 2496 bytes.
Visual Leak Detector is now exiting.
```

I'm using Visual Studio Community 2017, Version 15.9.17.

MartyG-RealSense, 2020-08-14:
Hi @eyeco  **hello-realsense** may be a better test-bed for testing for memory leak than your test snippet.  I think it is better to use a "proper" SDK program for testing than one that is purpose-designed for the sole purpose of detecting a leak, otherwise you may not get a result that is relevant to real-world performance.

There was a report of a mysterious memory leak on the Intel Support RealSense forum last month that did not seem to fit into known-about memory leaks on the Known Issues section of the SDK release notes.  There was not a definitive resolution in that particular case, but the information in it may provide some useful insights for your own situation.

https://support.intelrealsense.com/hc/en-us/community/posts/360048505293-out-of-memory

eyeco, 2020-08-17:
Hi @MartyG-RealSense ,
thanks. Like I wrote, this is also the case with the "proper" SDK programs, such as _hello-realsense_ and _multicam_, with no modifications whatsoever. So you're saying, you're not able to reproduce this?

MartyG-RealSense, 2020-08-17:
Rather than test for a memory leak with the **vld** program that you linked to, I would recommend testing the old fashioned way, by observing the memory usage of the computer over time and seeing if the available memory reduces over time as the application is left running.

closed 2020-09-03 by MartyG-RealSense