realsenseai/librealsense #7108: Tiny and very small objects compatible Realsense product

HatemHosam, 2020-08-14:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
Hello there, I want to ask what is the best product of realsense products which is compatible with tiny objects 3D scanning with high accuracy, the objects desired to be scanned are about 50x20x0.2 mm cubic. I want the most possible details of the scanned object. 
Thanks.

MartyG-RealSense, 2020-08-14:
Hi @HatemHosam   If you need the best quality image, the D415 model is likely to be the best choice.  It is suited to scanning small obects and has an optimum depth sensing resolution of 1280x720 (compared to 800x480 on the D435 models).  

If you are planning on scanning 3D models with the camera then LIPS Corp's **LIPScan 3D** software may be suitable, as you can place a small object on a turntable and scan in the model as the turntable rotates.

https://www.youtube.com/watch?v=N14Pi6z-MkE

HatemHosam, 2020-08-14:
Hello @MartyG-RealSense , Thank you for your reply. I have another question, what is the minimum Z distance of D415, as The object is very small , it need to be much closer to the camera to get more details because there are small holes on the surface. Is there any other recommendations in scanning process.

MartyG-RealSense, 2020-08-14:
The default minimum distance of the D415 is 0.3 meters, compared to 0.1 meters on the D435 models.  There is a function in the RealSense SDK called **Disparity Shift** though which can reduce the camera's minimum depth sensing distance so that the camera can get closer to the object.  Reducing th eminimum distance with Disparity Shift also reduces the camera's maximum observable distance, though this should not be a problem if you are scanning at very close range.

Depending on whether you are scanning objects with reflective surfaces, you may also consider applying a professional 3D scanning aerosol spray to dampen reflections on the object.

HatemHosam, 2020-08-14:
Is the Disparity Shift easy to access through the SDK or through LIPscan 3D? because the camera is to be placed in a small box and the working area will be small as 30x30x30 cm3 or 50x50x30 cm3. So I need to be sure that Disparity shift is handy, easy to be changed and ensure working in small space. Thank you for your reply and information.

MartyG-RealSense, 2020-08-14:
I would think that it would have to be accessed through the RealSense SDK.  Is there going to be a computing board inside the box please, like a sealed unit?

HatemHosam, 2020-08-14:
I and my team did not agree yet, if inside the box we might use a recommended intel board, if outside it is going to be a PC with windows 10. Is there any recommendation for that?

MartyG-RealSense, 2020-08-14:
The official RealSense Store promotes use of the 400 Series cameras with the **Up Squared** board, with the board available from the store on its own or bundled with a camera.  Up Boards have a long history of being partnered with RealSense cameras.

https://store.intelrealsense.com/

There are certainly many other options for development boards that can be paired with RealSense: Nvidia Jetson, Raspberry Pi, Odroid, to name just a few.  The 400 Series cameras can work with any Intel or Arm processor.

There is not a recommendation for full Windows 10 PCs.  In theory, any PC with Windows 10 should be usable with the cameras.  If you prefer to go with Intel, you could consider an Intel NUC mini-PC.   They are very small and come in a variety of configurations and price-points: circuit boards, partial kits where you supply some of the components such as memory yourself, and complete machines that work out of the box.

https://www.intel.co.uk/content/www/uk/en/products/boards-kits/nuc.html

MartyG-RealSense, 2020-08-27:
Hi @HatemHosam   Do you require further assistance with this case, please, or can it be closed?   Thanks!

HatemHosam, 2020-08-27:
Thank you very much for your help. You can close the case.

MartyG-RealSense, 2020-08-27:
Thanks very much for the update!   :)

closed 2020-08-27 by MartyG-RealSense