realsenseai/librealsense #7111: T265 :Enumeration after boot on Jetson Xavier NX

jack-wlj, 2020-08-15:
| Camera Model | T265 |
| Operating System & Version | Linux nvdia-desktop 4.9.140-tegra #1 SMP PREEMPT Wed Apr 8 18:15:20 PDT 2020 aarch64 aarch64 aarch64 GNU/Linux |
| Platform | Jetson Xavier NX  |
| SDK Version | 2.35.2 |

Issue Description
Can anyone please provide any update on this issue. It seems like the camera is not detected after initial boot and one has to plug out and plug in to make it work. This is a serious problem that prevents this camera from being used in any of my products. Customers certainly cant plug out and plug in every time they want to use this device. I am really hoping Engineering team can provide some workaround for this issue.

nvdia@nvdia-desktop:~$ lsusb
Bus 002 Device 006: ID 2b03:f682  
Bus 002 Device 005: ID 05e3:0620 Genesys Logic, Inc. 
Bus 002 Device 004: ID 0bda:0489 Realtek Semiconductor Corp. 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 13d3:3549 IMC Networks 
Bus 001 Device 014: ID 2b03:f681  
Bus 001 Device 013: ID 0403:6001 Future Technology Devices International, Ltd FT232 USB-Serial (UART) IC
Bus 001 Device 016: ID 093a:2510 Pixart Imaging, Inc. Optical Mouse
Bus 001 Device 015: ID 04f3:0103 Elan Microelectronics Corp. ActiveJet K-2024 Multimedia Keyboard
Bus 001 Device 012: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 001 Device 011: ID 03e7:2150  
Bus 001 Device 010: ID 0bda:5489 Realtek Semiconductor Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

nvdia@nvdia-desktop:~$ lsusb -d 03e7:2150 -v | grep bcdUSB
  bcdUSB               1.10

nvdia@nvdia-desktop:/usr/bin$ uname -a
Linux nvdia-desktop 4.9.140-tegra #1 SMP PREEMPT Wed Apr 8 18:15:20 PDT 2020 aarch64 aarch64 aarch64 GNU/Linux
nvdia@nvdia-desktop:/usr/bin$ ./rs-pose
RealSense error calling rs2_pipeline_start_with_config(pipe:0x55bcafe4d0, config:0x55bcae7730):
    No device connected

use uhubctl get data:
nvdia@nvdia-desktop:~/Downloads/USBHubReset_265/uhubctl$ sudo ./uhubctl -e -l 1-1.2 -a 1 -R
[sudo] password for nvdia: 
No compatible smart hubs detected at location 1-1.2!

csr-kick, 2020-08-18:
This is a well-known issue with this device, with no evidence of any support or fix forthcoming. 

To work around, install a tool like uhubctl and use it to powercycle your usb ports on boot. On the Rpi you would do:
```
uhubctl -a cycle -l 1 -p 1-4
uhubctl -a cycle -l 2 -p 1-4
```
So I assume something like that will work for you too.

AndreV84, 2020-09-01:
we have the same issue on jetson
t265 will not get after reboot.
we also got uhubctl compatible hub;
it doesn't seem to save the off state after reboot;
@csr-kick do you suggest to try the commands above given external usb hub?

Petrox, 2020-09-01:
This is clearly a bug, support should handle this as such, not with "have you tried turn it off and on again?"

AndreV84, 2020-09-01:
seems we got it with specific usbctl compatible hub with command
```
uhubctl -a cycle -l 1 -p 1-4
uhubctl -a cycle -l 2 -p 1-4
```

jack-wlj, 2020-09-02:
First, check the USB port information through  lsusb -t

(Note: You can plug and unplug the USB you want to control to determine which BUS is which port.)
Here I'm going to operate port 6 under port 1 of BUS 02.

Execute the following command to unbind the USB: 
echo '2-1.6' | sudo tee /sys/bus/usb/drivers/usb/unbind
to rebind the command, just change the unbind in the command above to BIND.
echo '2-1.6' | sudo tee /sys/bus/usb/drivers/usb/bind

I solved this problem by controlling USB binding/unbinding before enabling the program;
But this does not fundamentally solve the product problem;

AndreV84, 2020-09-15:
there is no unbind entry on jetson

yikir, 2022-02-07:
yes, I meet this problem too!!! does anyone knows how to bind-rebind the realsense device on nvidia-nx?

manyxu, 2022-10-07:
also got the problem ;-(

closed 2026-10-23 by MartyG-RealSense