realsenseai/librealsense #7113: Realsense frame issue

Zach0806, 2020-08-15:
Hi all,

I recently recorded a video using Realsense d435 with 30fps. The recording time was about 16 minutes. But when I converted the .bag file to PNG pictures, it only generated around 10000 pictures. I think there should be 16*60*30 = 28800 pictures. Does anyone know why? Thank you!

MartyG-RealSense, 2020-08-15:
Hi @Zach0806   The RealSense SDK has a tool called **rosbag inspector**.  It can be used to check the count of the frames in a bag by reading the topics and messages of the bag.

https://github.com/IntelRealSense/librealsense/tree/master/tools/rosbag-inspector

Calculating the expected frame count by other methods, such as multiplying the time length of the bag by the FPS, tends to be an inexact science, so a tool such as rosbag-inspector can provide a much more accurate value for the total number of frames in a bag.  You can then reliably compare this frame count to the number of frames that were extracted from the bag.

If you are a Windows user then you can find a pre-built version of the tool in the **Tools** directory of the SDK.  The Tools directory can be found quickly by right-clicking on the RealSense Viewer launch icon on the Windows desktop and selecting **Open file location** from its menu.

If rs-convert is dropping frames during extraction, the link below offers an explanation for why this may occur.

https://github.com/IntelRealSense/librealsense/issues/3500#issuecomment-473624881

Zach0806, 2020-08-15:
Hi @MartyG-RealSense ,

Thank you for your help! I will try the rosbag inspector at first.

Zach0806, 2020-08-15:
Hi @MartyG-RealSense，

Sorry for bothering you again. I want to know how to open the rosbag inspector application? Thank you!

MartyG-RealSense, 2020-08-15:
If you are using Windows, there is not a pre-built version of the rosbag inspector tool in the Tools directory of the SDK, unfortunately.  So the tool will likely need to be built from source code to create an executable program file.

An easier way to read the frame count will be to drag and drop your bag file into the central panel of the RealSense Viewer program to load the bag into the Viewer.  You can then click the pause button and drag the timeline to the end to see the value of the final frame.

[image]

[image]

Zach0806, 2020-08-15:
Thank you so much for helping!

MartyG-RealSense, 2020-08-27:
Case closed due to no further comments received.

closed 2020-08-27 by MartyG-RealSense

Zach0806, 2020-09-11:
Hi, @MartyG-RealSense, 

I have used your method to check the total frames of my bag file, and it showed around 39000 frames. But when I use ROS to check the information of the same bag file, it shows that there are only 9700 pictures in the image topic. Do you know why?

[image]
[image]

MartyG-RealSense, 2020-09-12:
Hi @Zach0806   Were you recording from the beginning to the end of the 38989 frame long sequence?  Or did you start the recording later or stop it early?  That might account for why there is much less frames recorded than the length of the camera session would seem to indicate.

Zach0806, 2020-09-12:
Hi, @MartyG-RealSense , Thank you for your reply.

I did record the sequence from beginning to the end. The situation was that I was doing my experiment upon one of the subjects and I needed to record the whole experiment using both Gopro and Realsense. I started both cameras at the beginning of the experiment and stopped them at the end, and there was no stop during the whole experiment.
When I loaded the sequence in Realsense Viewer, it showed me that the sequence has 38989 frames and lasts 19min23sec. But when I use ROS to check the information about the same sequence, it showed me that there are only 9700 messages in the color image topic. I was really confused, because I did not stop a single time when I recorded this sequence.

[image]

MartyG-RealSense, 2020-09-13:
I conducted an experiment by recording a bag with the same settings in the Viewer (1280x720, 30 FPS, RGB only).  After 1 minute I had 1988 frames.  Multiplying that by 20 (minutes) and assuming that the frame rate accumulation would have remained relatively constant, that equates to 39,760 frames (compared to the 38989 you got over a 20 minute period).  So it sounds as though the Viewer is recording the bag correctly.

I then extracted the color frames using the **rs-convert** bag frame extraction tool to convert the bag frames to PNG image files.  At the end of the conversion process of the 1988-frame bag file it said '1988 **color** frames processed'.   So this tool was also reading the bag's contents consistently.

https://github.com/IntelRealSense/librealsense/tree/master/tools/convert

So I cannot account for the difference that you experienced in the ROS tol that you used.

Zach0806, 2020-09-13:
Hi, @MartyG-RealSense , I had used the rs-convert bag frame extraction tool to convert my bag file too, but it only extracted 9700 PNG images. So i was so confused and did not know why the left frames were not extracted.

MartyG-RealSense, 2020-09-13:
There is an alternative bag to image converter program called **bag2image** that you could try to confirm whether there is a concsistent problem with extraction on your computer.  It extracts the color frames as JPEG rather than PNG though.

https://github.com/UnaNancyOwen/rs_bag2image

Zach0806, 2020-09-13:
Thank you! I will try it.

Zach0806, 2020-09-13:
Hi, @MartyG-RealSense , I think I found the problem. The problem is that I did not use the original USB cable to connect Realsense camera. When I did my experiment last month, i found that the original USB cable was not long enough, so I bought another USB 3.1 cable from amazon and used the new cable to conduct my experiment. So the recording file missed so many frames. BTW thank you so much for your time and help!

MartyG-RealSense, 2020-09-13:
You are very welcome.  I'm very pleased that you found a solution!