realsenseai/librealsense #7115: Python 3.8

Leo16112, 2020-08-16:
Hello,

are there any plans for implementation of Python 3.8 support? It is native Python in Ubuntu 20.04 so it would be nice to have. Certainly there is an option of installation of Python 3.7 and virtual environments, but still.

Kind Regards,
Anton.

MartyG-RealSense, 2020-08-16:
Hi @Leo16112   You can use Python 3.8 with librealsense already, though the **pyrealsense2** Python compatibility wrapper needs to be built from source code as there is not currently a PyPi (pip) installation package for 3.8.

 https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python#building-from-source

Leo16112, 2020-08-16:
Thank you for information.
Kind Regards,
Anton.

closed 2020-08-16 by Leo16112

yanickrochon, 2020-11-23:
Any news about this? Ubuntu 20.04 LTS comes by default with Python 3.8 and it's arleady a struggle to convince project managers to use Linux instead of Windows for new projects. This doees not help. Far from me to push contributors on this project, I'm a contributor myself on quite a few projects, I was just looking for updates since there does not seem to have much about this recently.

MartyG-RealSense, 2020-11-23:
Hi @yanickrochon   The most recent pip packages are still Python 3.7.  I do not have any further news about pip support for 3.8.

The pyrealsense2 wrapper can be built from source code for Python 3,8.  Alternatively, if building librealsense from source with CMake, the pyrealsense2 wrapper can be built at the same time by including Python wrapper flags in the CMake build instruction and setting it to target a specific Python version (making the pyrealsense2 installation process much simpler).

yanickrochon, 2020-11-23:
Hi @MartyG-RealSense but the question is : if I can build from source for Python 3.8, why isn't the package available through pip3 for Python 3.8? Why is the requirement not listing Python 3.8 as target platform?

MartyG-RealSense, 2020-11-23:
As mentioned earlier in this discussion, there is a feature request filed with Intel for adding Python 3.8 pip packages for pyrealsense2.  I have no information to provide other than that though or a time-schedule estimate.

yanickrochon, 2020-11-23:
Fair enough, but until the matter is actually resolved, why was this issue closed? Who is responsible to compile this project and publish it on Python's repository?

MartyG-RealSense, 2020-11-23:
The feature request is linked to the case below, so it is not necessary to keep this case open too.

https://github.com/IntelRealSense/librealsense/issues/6402