realsenseai/librealsense #7116: pose arriltag on windows?

AndreV84, 2020-08-16:
is ther eany way to run pose-apriltag on windows?

RealSenseSupport, 2020-08-21:
@AndreV84 I think this question depends on Apriltag project. I saw there's PR in that project. You might have a try. Thanks! https://github.com/AprilRobotics/apriltag/pull/27

RealSenseSupport, 2020-09-04:
@AndreV84 Any other questions about this? Looking forward to your update. Thanks!

AndreV84, 2020-09-04:
we were not able yet to implement the apriltag on windows.
we are trying to bet a mount of L515 & t265 right now with calibration extrinsics. Maybe you have solutions for such fixed mount implermentation?

RealSenseSupport, 2020-09-07:
@AndreV84  Sorry that we don't have such fixed mount solution for L515&T265.

RealSenseSupport, 2020-09-15:
@AndreV84 Any other questions about this? Thanks!

AndreV84, 2020-09-15:
we were not able to implement windows apriltag [ I do not have windows/realsense to test with locally]
thus the project got postponed.
regarding the combo l515 with t265 we will be looking forward to get combos that has components such as listed below but not limited to t265, l515, d435i, d435, d455

RealSenseSupport, 2020-09-15:
@AndreV84 If your project postponed, can we close this issue at this point? And please feel free to create new one when you restart the project if you still have issues or questions. Thanks!

closed 2020-09-15 by AndreV84