realsenseai/librealsense #7117: Depth value interpolated at edges

shiyoung77, 2020-08-16:
| Required Info |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 01.04.01.02 | 
| Operating System & Version |  Ubuntu18.04  | 
| Kernel Version (Linux Only)    |  5.4.0 | 
| Platform                                 | PC |
| SDK Version                            |  2.36 | 
| Language                            |  python  | 
| Segment			|  Robot  | 

### Issue Description
Hi, I just got a new L515 sensor and quickly tested it with some table-top objects. The overall depth quality seems good. But the depth value at the object edges seems to be interpolated and gives me "tails" of objects. I would rather get a zero instead of those interpolated values. I'm just wondering if this is normal and if there is a way to resolve this issue (perhaps by tuning some parameters/filters?). Attached is a sample point cloud from two viewpoints. Thanks.

[image]
[image]

RealSenseSupport, 2020-08-19:
Hi @shiyoung77 

We have quite a bit of documentation about tuning and understanding the L515 camera technology.  Here are links:

https://support.intelrealsense.com/hc/en-us/articles/360051646094-Intel-RealSense-LiDAR-Camera-L515-User-Guide

https://www.intelrealsense.com/optimizing-the-lidar-camera-l515-range/

https://dev.intelrealsense.com/docs/projection-texture-mapping-and-occlusion-with-intel-realsense-depth-cameras

tarekmuallim, 2020-08-21:
I also noticed the same problem with L515.
I read all the documents related to L515 and I didn't find any think related to this problem. 
I tired to change all the available control parameters, but the edges still have this problem.
I think Instead of making the edges interpolated like that it should give empty gab here.
This issue will introduce a lot of problems for some applications such as volume measurement.
Are there any direct instructions to solve this?

tarekmuallim, 2020-08-28:
Hi @RealSenseSupport @dorodnic 
is there any answer to my question?

RealSenseSupport, 2020-08-28:
Hi @tarekmuallim 

I understand what you mean.  One stream I would suggest to look at is the confidence stream.  That is one of the features added here on the L515 that can provide further information as to the confidence of the system in providing depth value for that pixel.  Changing camera perspective/moving the camera to a different position also can give a better understanding of the edges.  There isn't a way to zero out depth data for any given region or pixel.

FYI - Upgraded to the latest SDK and latest FW on L515 camera.

tarekmuallim, 2020-08-31:
Hi @RealSenseSupport 
Thank you for the support.

I tried to check the confidence map as you suggest.
It looks that this edges give a very high confidence despite being wrong edges!
it sill connect the edges with the background and give this points a high confidence.

This picture demonstrate the case:
I just put two boxes above each other in the view.
[image]

Here is the the result point cloud.
[image]
You can see the wrong edges

Her with confidence map texture
[image]
you can see the the edges have high confidence level!

Now if I change the confidence threshold to maximum "3"  I get this:
[image]
I still get the wrong edge with max confidence level.

RealSenseSupport, 2020-09-17:
Hi @tarekmuallim 

It looks like you're trying to do some type of box measurement usage, is that correct?  Or you're just using the box on top of box as an example?  I only say this cause we have box measurement with our DIM Weight Software (https://www.intelrealsense.com/dimensional-weight-software/).

This edge detection example is something that is difficult for a TOF based device.  At times the confidence stream would be useful in this situation but also the IR stream can be used to help determine edges as well.  The IR and Depth are calibrated together, so there is no difference between the two.

tarekmuallim, 2020-09-18:
Hi @RealSenseSupport  

Thanks for your replay.

What I want to say in my example, that it looks like  the confidence map is some how non relative, because it gives high confidence for false edges. and I suppose it should give low confidence for this edges.

Regarding Intel Realsense DIM Weight Software, I would like to try it if there is Linux version available.
Yes we are trying to build a Dimensioning System. we already manged to build one using Realsense D415.
You can check our system: [EasyCube](https://www.youtube.com/watch?v=lcVDIBGE5Hk) we are now finalizing the system design.

We are now testing L515 for longer range applications.

Anyway, I am testing some filtering techniques to improve the scene which we got from  L515.

this the original scene.
[image]
[image]

and this what we got after filtering.
[image]
[image]

Thanks again.

shiyoung77, 2020-09-20:
Hi @tarekmuallim,

Do you mind sharing the filtering techniques you are using? The results seem to be very good.

RealSenseSupport, 2020-10-02:
Hi @tarekmuallim 

Yes your filtering techniques seem really good based on the images provided.  I cannot provide a timeline as to when we may have Linux version of DIM Weight SW, right now we only have Windows version.  Monitor our DIM Weight SW site (https://www.intelrealsense.com/dimensional-weight-software/) for any updates.

*If we don’t hear from you in 7 days, this issue will be closed.

wangsff, 2020-10-12:
And I notice some noises on depth after align to color map.[#7541 ] I guess the noises showed in depth map maybe caused by "tails" of objects (mentioned above).  Am I right ? And how can I solve it? thanks @dorodnic @RealSenseSupport @shiyoung77 @tarekmuallim

RealSenseSupport, 2020-11-12:
Hi @wangsff 

There can always be some part of "tails" from objects when you're talking about capturing boxes or objects at various angles with the depth cameras.  Not sure if you've tried the DWS itself for box measurements but I'd suggest giving it a try and seeing if works in your usage.  

*If we don’t hear from you in 7 days, this issue will be closed.

closed 2020-12-23 by RealSenseSupport