realsenseai/librealsense #7118: The align " rs2::align align_to_depth(RS2_STREAM_DEPTH)" doesn't work

zhuhe1113, 2020-08-17:
realsense D435,C++

I want to align color image to depth image , it's doesn,t work.But I can align depth image to color image,
when I run:
    rs2::frameset frames = pipe.wait_for_frames();
    rs2:: align align2(RS2_STREAM_COLOR);
    auto processed = align2(frames);
it works.
But when I run:
    rs2::frameset frames = pipe.wait_for_frames();
    rs2:: align align2(RS2_STREAM_DEPTH);
    auto processed = align2(frames);
it doesn't work, and causes a interruption in the function:
    void invoke(frame f) const
   {
        rs2_frame* ptr = nulltr;
        std::swap(f.frame_ref,ptr);
       rs2_error* e = nullptr;
       rs2_process_frame(get(),ptr,&e)
       error::handle(e);
    }

dorodnic, 2020-08-17:
Hi @zhuhe1113 
Do you observe the same behaviour in rs-align example?

zhuhe1113, 2020-08-17:
> Hi @zhuhe1113
> Do you observe the same behaviour in rs-align example?
Yes ,I observe the same behaviour in rs-align example.
and I do the same thing in another computer,cause the interruption:
case RS2_EXCEPTION_TYPE_NOT_IMPLEMENTED:
                throw not_implemented_error(e);

zhuhe1113, 2020-08-17:
I don’t know if it’s due to different driver versions

zhuhe1113, 2020-08-17:
this is the code:
`int main()
{
	rs2::context ctx;
	rs2::colorizer color;
	rs2::config cfg1;
	pipeline pipeline0;
	cfg1.enable_stream(RS2_STREAM_DEPTH,1280, 720, RS2_FORMAT_Z16, 30);
	cfg1.enable_stream(RS2_STREAM_COLOR, 1280, 720, RS2_FORMAT_BGR8, 30);
	pipeline0.start(cfg1);
	while (1)
	{
		rs2::frameset frameset1 = pipeline0.wait_for_frames();
		rs2::align align2(RS2_STREAM_DEPTH);
		auto frameset_2 = align2.process(frameset1);
		rs2::frame frame_2 = frameset_2.get_depth_frame().apply_filter(color);
		rs2::frame frame_2_c = frameset_2.get_color_frame();
		Mat Color_Image2(Size(WIDTH, HEIGHT), CV_8UC3, (void*)frame_2.get_data(), Mat::AUTO_STEP);
		Mat Color_Image2_c(Size(WIDTH, HEIGHT), CV_8UC3, (void*)frame_2_c.get_data(), Mat::AUTO_STEP);
		cv::imshow("D435_2", Color_Image2);
		cv::imshow("D435_4", Color_Image2_c);
		waitKey(1);
	}
	return 0;

}`

RealSenseSupport, 2020-08-21:
@Zhuhe1113 Could you please tell which SDK version you used when get this issue? Thanks!

RealSenseSupport, 2020-08-21:
@zhuhe1113 And one minor suggestion - please note that creating align object is an expensive operation that should not be performed in the main loop. So please move "rs2::align align2(RS2_STREAM_DEPTH)" out of the loop. Please refer to https://github.com/IntelRealSense/librealsense/blob/master/examples/align/rs-align.cpp#L51

onceforall, 2020-08-21:
Hi, i  met same issue, as you set cfg1.enable_stream(RS2_STREAM_DEPTH,1280, 720, RS2_FORMAT_Z16, 30),  RS2_FORMAT_Z16 means depth value is 16 bits, but Mat Color_Image2(Size(WIDTH, HEIGHT), CV_8UC3, (void*)frame_2.get_data(), Mat::AUTO_STEP) is CV_8UC3, that's the problem, you should keep it consisent, set CV_8UC3 to CV_16UC1 will  solve it. Hope it helpful!

RealSenseSupport, 2020-08-24:
@onceforall Thanks very much for your sharing! 
@zhuhe1113  Could you please try as @oneceforall shared to see if it can solve your issue? Thanks!

RealSenseSupport, 2020-09-04:
@zhuhe1113 Did you get through the issue? Looking forward to your update. Thanks!

RealSenseSupport, 2020-09-15:
@zhuhe1113 Any update from your side? Looking forward to your reply. Thanks!

RealSenseSupport, 2020-09-21:
@zhuhe1113 Sorry that we didn't get response from you for weeks. Will close the ticket at this point. Please feel free to create another one if you still have issues or questions.Thanks!

closed 2020-09-21 by RealSenseSupport