realsenseai/librealsense #7120: D435 resolution ratio:640*360 x,y,z Coordinate error

wuhaoqi-hub, 2020-08-17:
hello,i want to understand the x,y,z Coordinate error in D435,the resolution ratio is 640*360,how can i get this data?

MartyG-RealSense, 2020-08-17:
Hi @wuhaoqi-hub   Could you provide more detail about your problem, please?

On D435, the optimal depth resolution is 848x480.  Use of lower resolutions reduces depth precision.

wuhaoqi-hub, 2020-08-17:
Because our imaging distance is relatively close，is about 18cm,so we choose the lower resolutions

MartyG-RealSense, 2020-08-17:
Reducing the resolution can reduce the minimum distance of the camera and allow it to get closer to an object.  There is a more refined way of achieving this goal though.  You can also increase the value of a setting called **Disparity Shift** to reduce minimum distance.  This should enable you to get closer to an object whilst using the optimal 848x480 resolution.

The link below may also be useful.  In it, the Chief Technical Officer of the RealSense Group at Intel (agrunnet) provides advice on depth accuracy at very close range.

https://github.com/IntelRealSense/librealsense/issues/4550

wuhaoqi-hub, 2020-08-17:
How big is the error of 18cm? If it is within 3mm, it is acceptable, so I want to know whether it can meet the requirements?

wuhaoqi-hub, 2020-08-17:
Or do I need to measure myself？

MartyG-RealSense, 2020-08-17:
The depth accuracy of the D400 series cameras is less than one percent of the distance from the object.  So if the camera is 1 meter from the object, the expected accuracy is between 2.5mm to 5mm.

18 cm would convert to 0.18 meters.  So it is likely that if you are at a 0.18 m range and your acceptable target is 3 mm, you should be fine.

The RealSense SDK's **Depth Quality Tool** can provide real-time feedback about depth quality, including RMS Error.

wuhaoqi-hub, 2020-08-17:
OK，thank you

wuhaoqi-hub, 2020-12-21:
The depth accuracy of the D400 series cameras is less than one percent of the distance from the object. Can I understand that if the distance is 18 cm and the error is less than 1%, then the error is 1.8 mm, that is + - 0.9 mm?

MartyG-RealSense, 2020-12-21:
Accuracy can depend on a range of factors, including environmental and lighting conditions.  Instead of calculating 1% of a distance, from an object it would be more reliable to calculate **RMS Error** (error that increases linearly over distance) using the formula in point 5 of the document linked to below:

https://dev.intelrealsense.com/docs/tuning-depth-cameras-for-best-performance#section-verify-performance-regularly-on-a-flat-wall-or-target

[image]

An easier way to get feedback about the camera's current RMS error is to use the SDK's **Depth Quality Tool** program.

wuhaoqi-hub, 2020-12-21:
The RMS error represents the depth noise for a localized plane fit to the depth values，Can I understand that if the RMS error is 10 mm and the real value is 180 mm, then the value measured by the camera is in the range of 175mm ~ 185mm?

MartyG-RealSense, 2020-12-21:
RMS error can be different depending on what camera model is being used (D435 has around 2x the RMS Error over distance of D415, for example).  I would therefore refer you to Intel's own chart for RMS over distance to estimate RMS at 180 mm with a D435.  As you can see from the chart though, as RMS is near zero at the camera lens, RMS error should be close to zero at 180 mm.

[image]

closed 2021-01-03 by MartyG-RealSense