realsenseai/librealsense #7122: Runtime Error when reading advanced mode settings from recorded device

xHotch, 2020-08-17:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Operating System & Version |   Win 10 | 
| Platform                                 | PC |
| SDK Version                            |  2.36.0                        | 
| Language                            |  python                          | 

### Issue Description

I am trying to read the advanced mode settings from a recorded .bag file.

To achieve this i want to use [serialize_json](https://intelrealsense.github.io/librealsense/python_docs/_generated/pyrealsense2.rs400_advanced_mode.html?highlight=advanced#pyrealsense2.rs400_advanced_mode.serialize_json)

My code so far looks something like this:

    config = rs.config()

    rs.config.enable_device_from_file(config, args.input)

    pipeline = rs.pipeline()

    pipeline.start(config)

    pipeline.get_active_profile().get_device().as_playback().set_real_time(False)

    device = pipeline.get_active_profile().get_device()

    advnc_mode = rs.rs400_advanced_mode(device)

    print(advnc_mode.is_enabled())

    serialized_string = advnc_mode.serialize_json()`

This however results in a _RuntimeError: null pointer passed for argument "dev"_ 

Is is not possible to read advanced mode settings from a recorded device or am i doing something wrong?

Reading other infos such as sensor options works fine.

MartyG-RealSense, 2020-08-17:
Hi @xHotch   json files are typically manually created and edited in a text editor program or can be created by exporting settings from the **RealSense Viewer** program as a custom json file using the json export button on the top of the options side-panel.    So the Advanced Mode settings would not be incorporated into a bag during the bag recording process.

It is straightforward to apply Advanced Mode settings to the bag data by loading a json file though.

xHotch, 2020-08-18:
Hello @MartyG-RealSense,
thank you for your quick answer.

You are correct in your edit, i want to only read out the values of Advanced Mode, so i can find out what values were used to record a specific bag file, as i have used different Advanced Mode settings(presets) for my recodings. I was also able to read out values under the **Controls** section.

MartyG-RealSense, 2020-08-18:
A way to tell which set of Advanced Mode settings were used may be to export a json file from the Viewer for each different set of Advanced Mode values that you use, and reference the name of the json in the filename of the bag.   I can appreciate though that this would not be useful for bag files that have already been recorded.

As Advanced Mode is part of the ASIC hardware of the camera, I do not know of a way to retrieve the Advanced Mode settings that were used during the recording once the recording has been completed.

closed 2020-09-03 by MartyG-RealSense