realsenseai/librealsense #7124: A query regarding T265 re localization

zainmehdi, 2020-08-18:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265| 
| Firmware Version                   | Latest| 
| Operating System & Version |  Ubuntu 18.04 | 
| Kernel Version (Linux Only)    |      5.4                                    | 
| Platform                                 | PC/Raspberry  |
| SDK Version                            |   2.37                          | 
| Language                            |  C++                        | 
| Segment			|  {Robot   | 

### Issue Description
I have been playing around with AR advances example provided for T265 in order to test landmark creation and map saving. I was successfully able to create a landmark and save the map. Later the map could also be retrieved with the correct position of landmark in global space. What I want to ask is that is it possible to relocalize the robot/camera after a reboot with absolute pose. By absolute pose I mean the pose at which the robot/camera was shutdown. Right now when I start the camera again after a shutdown down the landmark is localized well but the pose I get for the robot is with respect to that landmark not the absolute origin. 

[image]

In the above figure when I relocalize my robot it makes the landmark as origin and my robot pose is provided relative to the landmark. Is it possible to retrieve the absolute pose of the robot i.e. (3,2) .
Thanks

zainmehdi, 2020-09-04:
Can I please get some feed on this issue ?

arjunsengupta98, 2020-09-21:
Hey @zainmehdi 
I'm working on re localization witht the T265 as well. Were you able to have any success with the package mentioned here: https://github.com/IntelRealSense/realsense-ros/issues/1155#issuecomment-675755855 ?

Apparently the realsense team will not address issues related to t265 anymore 
https://github.com/IntelRealSense/librealsense/issues/7258#issuecomment-692964955

zainmehdi, 2020-09-23:
Hi @arjunsengupta98 Sorry for the late reply. I actually was able to make it work. Like you can set landmarks and it would localize to those landmarks when you revisited the place. It want too robust I would say however.Feel free to let me know if you have any more queries.

arjunsengupta98, 2020-10-05:
Hi @zainmehdi 
Thanks for the update! Sorry, I missed replying to you earlier. I've temporarily gotten rid of the T265 from my robot since I need to understand and evaluate other VIO algorithms on my board. I probably won't be using the t265 again anytime soon. If I do, and manage to make any progress, I shall share an update. 
Thanks again!

zainmehdi, 2020-10-05:
@arjunsengupta98 Looking forward to your updates. Just for the heads up I tried basalt vio using fish eye stream and it was working better than other open source solutions. I am also looking for an alternative VIO solution for embedded platform. Looking forward to hearing from you

Thanks

arjunsengupta98, 2020-10-06:
@zainmehdi 
Ohh I see! I'm working on Nvidia's Isaac SDK for now and am directly using their SVIO GEM https://docs.nvidia.com/isaac/isaac/packages/perception/doc/visual_odometry.html

closed 2026-10-23 by MartyG-RealSense