realsenseai/librealsense #7125: T265 Keeps conneting and disconnecting (keeps booting) and not found Intel Realsense Viewer 2.36

MarcoAndrade26, 2020-08-18:
---------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 & D435| 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Windows 10 & Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  5.3.0-1030-raspi2                                      | 
| Platform                                 | PC/Raspberry Pi  |
| SDK Version                            |  2.0                           | 
| Language                            |  C++ }                          | 
| Segment			|  Robotics                          | 

### Issue Description
I downloaded Intel Realsense SDK and when opening Intel Realsense Viewer, I have tried three OS:

- Windows 10 (with Intel Core i7, 16 Gb of Ram)
- Ubuntu 18.04 on a Virtual Machine (VM Ware with 8 Gb of RAM)
- Ubuntu 18.04 on Raspberry Pi 4 (4Gb)

Camera D435 is detected and works fine. I tried to connect T265 and this camera keeps booting (connecting and disconnecting). I have already updated USB drivers. Both cameras are being connected on USB 3.0 ports. I purchased a USB 3.0 hub that is powered externally to provide more power. First I connect the Hub to a USB 3.0 port on Windows 10 computer, then T265 camera, it is detected but when opening Realsense Viewer is when error starts. Suddenly it works fine but like 95% of the time the error happens.

Any advice? Thanks!

RealSenseSupport, 2021-01-30:
Hi,

Will you be needing further help with this?

*If we don’t hear from you in 7 days, this issue will be closed.*

Thanks

MarcoAndrade26, 2021-02-16:
Hello! @RealSense-Customer-Engineering  @RealSenseCustomerSupport Now I am having this issue. #8374

closed 2026-10-23 by MartyG-RealSense