realsenseai/librealsense #7129: RealSense error calling rs2_load_json(dev:05E01D48, json_content:05E7A680, content_size:1539):     hr returned: HResult 0x80070079: "信号灯超时时间已到"

xiebei, 2020-08-19:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D415 } | 
| Firmware Version                   | 05.12.06.00 | 
| Operating System & Version |   Win10  | 
| SDK Version                           |  2.36.0                          | 
| Language                               |  {C++ }                          | 

### Issue Description
I used 2 USB ports to connect 4 D415 cameras via USB hubs, sometimes this error. USB hubs support external power supplies.
It seems to be working well for two months, and I'm not sure if it's because of what I've changed. may be a win10 system update?

RealSense error calling rs2_load_json(dev:05E01D48, json_content:05E7A680, content_size:1539):
    hr returned: HResult 0x80070079: "信号灯超时时间已到"

my code like this:
    try
    {
            if (dev_it->is<rs400::advanced_mode>())
            {
                auto advanced_mode_dev = dev_it->as<rs400::advanced_mode>();
                if (!advanced_mode_dev.is_enabled())
                    advanced_mode_dev.toggle_advanced_mode(true);

                std::ifstream t(config.preset, std::ifstream::in);
                std::string preset_json((std::istreambuf_iterator<char>(t)), std::istreambuf_iterator<char>());
                advanced_mode_dev.load_json(preset_json);
            }
    }
    catch (const rs2::error& e)
    {
        LOG(ERROR) << "RealSense error calling " << e.get_failed_function() << "(" << e.get_failed_args() << "):\n    " << e.what();
        return 0;
    }
    catch (const std::exception& e)
    {
        LOG(ERROR) << e.what();
        return 0;
    }

MartyG-RealSense, 2020-08-19:
Hi @xiebei  The last part of the error translates to "Semaphore timeout period has expired".  This is an error that does not typically appear during use of librealsense.  It is apparently most commonly encountered when transferring / copying files (for example, to / from an SD card).

https://answers.microsoft.com/en-us/windows/forum/all/error-0x80070079-the-semaphore-timeout-period-has/6cd88126-6bc5-460e-80f6-1bcaa017e422

Since the error message makes specific reference to json loading throughout: if you temporarily remove the line **advanced_mode_dev.load_json(preset_json);** then that should help to diagnose whether the problem is triggered by the reading of the json file.

MartyG-RealSense, 2020-08-27:
Hi @xiebei  Do you require further assistance with this case, please?   Thanks!

xiebei, 2020-08-27:
@MartyG-RealSense Hi,  I did a lot of testing and found that as long as Manual exposure was set, I will get this error. After debugging the realsenese source code, I found that it was an error when setting the gain value. I don't understand what's going on, Now I can only use automatic exposure...

MartyG-RealSense, 2020-08-27:
Hi again @xiebei   Are you disabling auto-exposure before you set the manual exposure, please?  If auto-exposure is active then the manual values are not updated.  A RealSense team member discusses the mechanics of this in more detail in the link below.

https://github.com/IntelRealSense/librealsense/issues/1624#issuecomment-389085387

xiebei, 2020-08-27:
Yes,  and I loaded the HighResHighAccuracy first, and then disabled auto-exposure,  set exposure value,  set gain value.
Now my system is a little complex and I have 8 d415 cameras connected,  One usb controller for every two camera, running in 15fps,  I will do more tests using one camera.

MartyG-RealSense, 2020-08-27:
I went back to the start of the case and reviewed all the details so far.  

It is not impossible that a Windows update in the past two months could have affected the camera, as it has certainly occurred numerous times in the past.  Unless you roll back Windows to the last good build that you knew worked (and make updates require user permission to install so camera-breaking updates do not install again automatically), the only remedy in such cases may be to wait til Microsoft release a new build that corrects the problem somehow.

Windows build versions may be one reason why a camera works fine on one Windows PC and not another.  Other factors such as the USB ports on a particular computer model may also have an effect.  Not all ports are built equal, despite USB being a standard.  An example would be devices with built-in USB ports that handle two ports on one controller like a hub instead of having a dedicated USB controller for every port.  

It is a good idea to divide the cameras across more than one hub like you are doing.  A powered hub can supply around 12V to all the devices attached to it, and you should budget for around 2W power for each attached camera.  The USB standard allows up to five hubs to be linked together on a single computer.

xiebei, 2020-09-03:
I didn't solve this problem, I think the power is enough, I decided to use auto exposure now...
Thank you for your patience help.

closed 2020-09-03 by xiebei

MartyG-RealSense, 2020-09-03:
Thanks too for the update!