realsenseai/librealsense #7130: set_region_of_interest too slow

ChenXinThales, 2020-08-19:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400 | 
| Firmware Version                   | 05.12.01.00 | 
| Operating System & Version |  Win7/10  | 
| Kernel Version (Linux Only)    |                                     | 
| Platform                                 | PC |
| SDK Version                            |  2.22                      | 
| Language                            |  C++                         | 
| Segment			|                      | 

### Issue Description
I set ROI for color sensor, it worked, but the set_region_of_interest function is a bit slow for me (140ms) , what should I do to improve it?
Here is part of my code:

pipeline *pipe;
rs2::config *s_cfg = new rs2::config;
s_cfg->enable_stream(RS2_STREAM_COLOR, 640, 480, RS2_FORMAT_BGR8, 30);
s_cfg->enable_stream(RS2_STREAM_DEPTH, 640, 480, RS2_FORMAT_Z16, 30);
rs2::pipeline_profile profile = pipe->start(s_cfg);
auto sensors = profile.get_device().query_sensors();
sensors[1].set_option(rs2_option::RS2_OPTION_ENABLE_AUTO_EXPOSURE, 1);
region_of_interest org_region= ((roi_sensor)sensors[1]).get_region_of_interest();
DOWRD startTime = GetTickCount();
((roi_sensor)ros).set_region_of_interest(org_region);
DOWRD endTime = GetTickCount();
printf("set ROI used time %d\n",endTime-startTime);

MartyG-RealSense, 2020-08-19:
Hi @ChenXinThales   Your stated librealsense version is very old (SDK 2.22.0 from May 2019.  You may get some benefit if you update your project to a modern librealsense version if possible so that you can benefit from advancements made to the SDK since May 2019.    

Also, I note that you use the term DOWRD.  Do you mean **DWORD** ?

ChenXinThales, 2020-08-19:
> Hi @ChenXinThales Your stated librealsense version is very old (SDK 2.22.0 from May 2019. You may get some benefit if you update your project to a modern librealsense version if possible so that you can benefit from advancements made to the SDK since May 2019.
> 
> Also, I note that you use the term DOWRD for the timer variables. Do you mean **DWORD** ?

Yes, I mean DWORD, I wrote it wrong. Besides updating SDK version, is there any other way to reduce setting ROI time?

MartyG-RealSense, 2020-08-19:
I'm not sure that the method that you are using to measure start to finish processing time is the best one.  I would recommend trying **std::chrono::high_resolution_clock**.  The link below has a C++ example that demonstrates measuring the time of a process using this method.

https://github.com/IntelRealSense/librealsense/issues/4633

Also, if you are building librealsense from source code with CMake instead of a Windows SDK installer file, I would recommend using the term  **-DCMAKE_BUILD_TYPE=release** in the build statement if not doing so already, so that you build librealsense as a **Release** build that has optimizations instead of a Debug build.

dorodnic, 2020-08-19:
Hi @ChenXinThales 
Updating Auto-Exposure ROI will take couple of frames to kick-in. There is nothing I know you can do to speed it up. @agrunnet - do you have any suggestion?

ChenXinThales, 2020-08-20:
> I'm not sure that the method that you are using to measure start to finish processing time is the best one. I would recommend trying **std::chrono::high_resolution_clock**. The link below has a C++ [...]

I have updated SDK version to 2.36 and used std::chrono::high_resolution_clock to measure time, but the result was still about 142ms.

MartyG-RealSense, 2020-08-20:
@ChenXinThales   Okay, thanks very much for trying the **std::chrono::high_resolution_clock** method.  At least you can have confidence in the accuracy of the time measurement, then.

closed 2020-09-03 by MartyG-RealSense