realsenseai/librealsense #7132: TypeError: enable_device() missing 2 required positional arguments: 'device_serial' and 'enable_ir_emitter'

sw14928, 2020-08-19:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 01.03.15.100 | 
| Operating System & Version |   Win10 | 
| Platform                                 | PC |
| SDK Version                            |  2.0                          | 
| Language                            |  python                          | 
| Segment			|  Robot                         | 

I am having some issues using my intel realsense L515. I am trying to set it up to take a depth image and I receive error:

TypeError: enable_device() missing 2 required positional arguments: 'device_serial' and 'enable_ir_emitter'

Here is my code:
    c = rs.config()
    c.enable_stream(rs.stream.depth, 1024, 768, rs.format.z16, 30)
    c.enable_stream(rs.stream.color, 1280, 720, rs.format.bgr8, 30)
    device_manager = rsdm.DeviceManager(rs.context(), c)
    #device_manager.enable_emitter(True)
    
    device_manager.enable_device()
    (serial, device) = device_manager._enabled_devices.items()[0]
    [depthStream, videoStream] = device.pipeline_profile.get_streams()
    camera = {'serial': serial, 'device': device, 'manager': device_manager, 
               'video': videoStream, 'depth': depthStream}

    refImage = getFrame(camera, 'video')

dorodnic, 2020-08-19:
Hi @sw14928 
What is `rsdm`?

sw14928, 2020-08-19:
Hi @dorodnic rsdm is real sense device manager. I import this at the top of the code: 

import realsense_device_manager as rsdm

many thanks!

dorodnic, 2020-08-19:
You mean from the box_dimensioner_multicam example? 
I don't think this sample would work out-of-the-box with L515. Before the implementation specifics, the sample relies on multicamera calibration, which cannot be done without hardware synchronisation with the L515. 
If you do want to use rsdm with L515, some code changes are required. L515 doesn't have emitter_enabled option (you cannot operate Lidar camera in passive mode, unlike stereo). Commenting out the two lines using emitter_enabled should help.

sw14928, 2020-08-20:
Ok this makes sense, thanks for the help! I tried to alter my code I had for a realsense stereo camera, but I will probably need to instead rebuild the code.

Commenting the emitter enabled line did not work unfortunately. I will try a different method entirely, many thanks!

RealSenseSupport, 2020-09-11:
@sw14928 

Is there anything else needed on this one?

*If we don’t hear from you in 7 days, this issue will be closed.

closed 2020-09-18 by RealSenseSupport

AndreV84, 2020-10-16:
@dorodnic could you confirm both?   any? neither of the two  to work with l515, please? (https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/box_dimensioner_multicam/box_dimensioner_multicam_demo.py) 
(https://github.com/IntelRealSense/librealsense/tree/master/examples/measure)