realsenseai/librealsense #7135: Measure distance using net_device

podshivalov-ilya, 2020-08-19:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.12.07.100 | 
| Operating System & Version |  Linux (Manjaro 20.0.3) | 
| Kernel Version (Linux Only)    |  5.8.0                                        | 
| Platform                                 | PC, Raspberry Pi  |
| SDK Version                            |  2.38.0                          | 
| Language                            |  C++                          | 
| Segment			|  others                         | 

I'm trying to measure distance using example rs-measure via camera connected using the [article](https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras). So I changed rs-measure respectively (changes below) but measurements aren't valid. In case of validation, I put a 15cm ruler under the camera and got approximately 800,000cm length for it using modified rs-measure. It's nonsense because everything works properly via usb. Does anybody have ideas how to fix it or from which I could start debugging?

<details>
<summary>rs-measure.cpp changes</summary>

```diff --git a/examples/measure/rs-measure.cpp b/examples/measure/rs-measure.cpp
index d14170e23..1149e94e9 100644
--- a/examples/measure/rs-measure.cpp
+++ b/examples/measure/rs-measure.cpp
@@ -3,6 +3,7 @@
 
 #include <librealsense2/rs.hpp> // Include RealSense Cross Platform API
 #include <librealsense2/rsutil.h>
+#include <librealsense2-net/rs_net.hpp>
 #include "example.hpp"          // Include short list of convenience functions for rendering
 
 // This example will require several standard data-structures and algorithms:
@@ -129,14 +130,17 @@ int main(int argc, char * argv[]) try
     //   b. We don't want to introduce new holes
     rs2::align align_to(RS2_STREAM_DEPTH);
 
+    rs2::context ctx;
+    rs2::net_device dev("10.0.0.175");
+    dev.add_to(ctx);
     // Declare RealSense pipeline, encapsulating the actual device and sensors
-    rs2::pipeline pipe;
+    rs2::pipeline pipe(ctx);
 
     rs2::config cfg;
     cfg.enable_stream(RS2_STREAM_DEPTH); // Enable default depth
     // For the color stream, set format to RGBA
     // To allow blending of the color frame on top of the depth frame
-    cfg.enable_stream(RS2_STREAM_COLOR, RS2_FORMAT_RGBA8);
+    cfg.enable_stream(RS2_STREAM_COLOR, RS2_FORMAT_RGB8);
     auto profile = pipe.start(cfg);
 
     auto sensor = profile.get_device().first<rs2::depth_sensor>();
```
</details>

<details>
<summary>CMakeLists.txt changes</summary>

```diff --git a/examples/measure/CMakeLists.txt b/examples/measure/CMakeLists.txt
index 66201dd4f..148d451ce 100644
--- a/examples/measure/CMakeLists.txt
+++ b/examples/measure/CMakeLists.txt
@@ -11,7 +11,7 @@ find_package(Threads REQUIRED)
 if(BUILD_GRAPHICAL_EXAMPLES)
     add_executable(rs-measure rs-measure.cpp ../example.hpp)
     set_property(TARGET rs-measure PROPERTY CXX_STANDARD 11)
-    target_link_libraries(rs-measure ${DEPENDENCIES} Threads::Threads)
+    target_link_libraries(rs-measure ${DEPENDENCIES} Threads::Threads realsense2-net)
     include_directories(rs-measure ../)
     set_target_properties (rs-measure PROPERTIES FOLDER "Examples")
```
</details>

MartyG-RealSense, 2020-08-19:
Hi @podshivalov-ilya   You have an uncommon Linux setup (at least, uncommon in terms of librealsense setups).  Would it be possible please to try using the modified **rs-measure.cpp** example with the pre-made Pi 4 SD card image that Intel provide, to confirm whether the problem that you are experiencing is related to your Linux configuration? 

https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras#section-2-3-preparing-the-sd-card

podshivalov-ilya, 2020-08-20:
@MartyG-RealSense I use exactly the same image on the Raspberry pi without any significant modifications. After setup the image I only configured network. So rs-server started on a common linux setup. But I got an idea. I'll try to reproduce the bug on different clients with a common linux or windows setup.

MartyG-RealSense, 2020-08-20:
Ok @podshivalov-ilya I look forward to your test results with different clients.  Thanks!

dorodnic, 2020-08-20:
Hi @podshivalov-ilya 
I suspect this may be due to a bug that was addressed lately (#7017). It seems that Depth-Units of Net Device were not being reported correctly. The fix should be part of upcoming release.

podshivalov-ilya, 2020-08-20:
@MartyG-RealSense I've reproduced the bug on a windows machine. But as @dorodnic suspected it's just a duplication of the issue #7017. Thank you for your support

closed 2020-08-20 by podshivalov-ilya