realsenseai/librealsense #7137: RealSense SDK2.0 Pre-Relase (2.37.0.2094) version doesn't work with D455 Camera

Jackchen234, 2020-08-20:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { RealSense  D455 }                                | 
| Firmware Version                   |                                                               | 
| Operating System & Version |   {Win (7/10) / Linux (Ubuntu 18.04)      | 
| Kernel Version (Linux Only)    |                                                               | 
| Platform                                 | PC/ NVIDIA Jetson Nano                       |
| SDK Version                            |  { 2.0 }                                                    | 
| Language                            |  {C/C#/labview/nodejs/opencv/ }             | 
| Segment			|  {Robot                                }                          | 

### Issue Description

RealSense D455 Camera doesn't recognized and work after SDK2.0 installed ( both 2.36.0 & Pre-Relase 2.37.0.2094 version)

Windows 7:
In Windows Device manager show => Device can't Enable 
In RealSense Viewer => Doesn't see any device 

Ubuntu 18.04 :
Same problem in Jetson Nano Platform 

Does any newly Driver or SDK support for RealSense D455 ....?

Pls help Tks !

-Jack

dorodnic, 2020-08-20:
Hi @Jackchen234 
Windows 7 is highly outdated and no longer supported by Microsoft, and for that reason we do not guaranty support for our new products on that platform on launch. We may add it in the future given sufficient continued interest from the community.  
Jetson Nano should work with the D455. Are you following [this](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md) installation guide?

Jackchen234, 2020-08-21:
hi  @dorodnic

I followed the installation guide to install on Jetson Nano again, but it still doesn't work.
I tried these two installation guide ( 1. https://www.jetsonhacks.com/2019/05/16/jetson-nano-realsense-depth-camera/      2.https://github.com/intelRealSense/librealsense/blob/master/doc/installation_jetson.md  ) and got the same result in realsense-viewer  show:

Notification from Hardware
Multiple realsense udev-rules were found!
1:/etc/udev/rules.d/99-realsense-libusb.rules
2:/lib/udev/rules.d/60-librealsense2-udev-rules.rules
Make sure to remove redundancies!

I tried to remove 1. and run again but it's told me  /lib/udev/rules.d/60-librealsense2-udev-rules.rules  is not up-to date!

I just wonder which step i missed ? ( ./patchUbuntu.sh ...? I can't find this patch file when i followed the instruction of jetsonhacks , even it's Note: If you have a D435i, the camera will not be detected without the patches.)  or ( the Step 5 of your instruction => "you can compile an application with librealsense using g++ -std=c++11 filename.cpp -lrealsense2 or an IDE of your choice. To get started with RealSense using CMake check out librealsense/examples/cmake" )

-Jack

cole-myers, 2020-08-21:
Exact same setup (win7 + nano), exact same issue here on both platforms! Following closely

RealSenseSupport, 2020-08-24:
@Jackchen234 , please try to manually remove the 2nd udev-rules and then relaunch the realsense viewer. 
Thank you. 
`sudo rm /lib/udev/rules.d/60-librealsense2-udev-rules.rules`

Jackchen234, 2020-08-24:
@RealSenseSupport

I tried to renamed before 

$ Sudo rm /lib/udev/rules.d/60-librealsense2-udev-rules.rules  /lib/udev/rules.d/60-librealsense2-udev-rules.0rules

The Realsense-view can launch ed , but it still can't connect to D455 , even those rs-samples code are the same !

RealSenseSupport, 2020-08-25:
Hi @Jackchen234, 
Just to check, please ensure the launched RealSense Viewer is actually the one built with RSUSB_BACKEND enabled.  (build/tools/realsense-viewer/realsense-viewer)

> librealsense version v2.36.0 / Pre-Release v2.37.0.2094
> Realsense-Viewer v2.31.0

It seems like three different versions being installed?

Jackchen234, 2020-08-25:
Hi@RealSenseSupport 

How can I ensure the launched RealSense-Viewer is the one built with RSUSB_BACKEND enabled ...?

I launched RealSense-Viewer from 
   ~$ librealsense/build/tools/realsense-viewer/realsense-viewer
but it's can't see D455 on left side , and the versions of realsense-viewer & librealsense are show v2.31.0

RealSenseSupport, 2020-08-25:
>  librealsense are show v2.31.0

This may be the issue. The D455 was added to the RealSense SDK support starting from [v2.36.0](https://github.com/IntelRealSense/librealsense/releases/tag/v2.36.0).

Jackchen234, 2020-08-25:
@RealSenseSupport 

How can I remove the librealsense v2.31.0 , and reinstall RealSense SDK v2.36.0 ...?
I tried to remove before with command :
       $ sudo dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge
but it shows dpkg: error --purge need one package name as least

RealSenseSupport, 2020-08-25:
Please checkout v2.36.0 first before building and re-installing  v2.36.0 

```
~/librealsense$ git checkout v2.36.0
~/librealsense$ mkdir build && cd build
~/librealsense$ cmake ../ -DBUILD_EXAMPLES=true -DCMAKE_BUILD_TYPE=Release -DBUILD_WITH_TM2=false -DBUILD_WITH_CUDA=false -DFORCE_RSUSB_BACKEND=true
~/librealsense$  sudo make uninstall && make clean && make && sudo make install
```

I hope this helps.

Jackchen234, 2020-08-25:
Hi @RealSenseSupport 

When I waiting for your reply, I tried to

1. Download RealSense SDK 2.0 (V2.36.0) Source code again.
2. Rebuild it with your instruction before :

             $ cmake ../ -DBUILD_EXAMPLES=true -DCMAKE_BUILD_TYPE=Release -DBUILD_WITH_TM2=false -DBUILD_WITH_CUDA=false -DFORCE_RSUSB_BACKEND=true
             $ make -j4
3. Launch the realsense-viewer that built with RSUSB_BACKEND enabled.

finally, It shows the realsense view v2.36.0 but also shows /etc/udev/rules.d/99-realsense-libusb.rules is not up-to date !
I followed the instruction to manual update 

          $ sudo cp ~/.99-realsense-libusb.rules /etc/udev/rules.d/99-realsense-libusb.rules && sudo udevadm control --reload-rules && udevadm trigger

4. Launch realsense-viewer again.

It's workes .....realsense-viewer can connect to D455 , and exsamples also can run !

Thanks for your support !

-Jack

xdwgood, 2020-08-26:
@Jackchen234 

> D455 is not recognized by RealSense-Viewer , but it workes in Cheese webcam viewer application.

It works fine on my system.

RealSense ROS v2.2.16
[ INFO] [1598420121.376173516]: Built with LibRealSense v2.36.0
[ INFO] [1598420121.376185236]: Running with LibRealSense v2.37.0
ubuntu 18.04
5.4.0-42-generic
D455 firmware 5.12.07.100
Realsense-Viewer v2.38.1

Jackchen234, 2020-08-26:
Thanks guys !

closed 2020-08-26 by Jackchen234