realsenseai/librealsense #7138: error when run "roslaunch realsense2_camera rs_camera.launch"

xdwgood, 2020-08-20:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D455 } | 
| Firmware Version                   | (5.12.06.00) | 
| Operating System & Version |   { Linux (Ubuntu 18.04lts)  | 
| Kernel Version (Linux Only)    |  (5.4.0-42-generic)                                         | 
| Platform                                 | PC|
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
dw@dw:~$ roslaunch realsense2_camera rs_camera.launch 
... logging to /home/dw/.ros/log/9dde61a8-e2c3-11ea-a993-98fa9bae5cf3/roslaunch-dw-17511.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://dw:41819/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
.......
 * /camera/realsense2_camera/usb_port_id: 
 * /rosdistro: melodic
 * /rosversion: 1.14.7

NODES
  /camera/
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [17522]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 9dde61a8-e2c3-11ea-a993-98fa9bae5cf3
process[rosout-1]: started with pid [17533]
started core service [/rosout]
process[camera/realsense2_camera_manager-2]: started with pid [17540]
process[camera/realsense2_camera-3]: started with pid [17541]
[ INFO] [1597914038.986163739]: Initializing nodelet with 8 worker threads.
[ INFO] [1597914039.075240575]: RealSense ROS v2.2.16
[ INFO] [1597914039.075269090]: Built with LibRealSense v2.36.0
[ INFO] [1597914039.075279809]: Running with LibRealSense v2.37.0
[ WARN] [1597914039.075290693]: ***************************************************
[ WARN] [1597914039.075299076]: ** running with a different librealsense version **
[ WARN] [1597914039.075306412]: ** than the one the wrapper was compiled with!   **
[ WARN] [1597914039.075313756]: ***************************************************
[ INFO] [1597914039.084389931]:  
[ INFO] [1597914039.451255339]: Device with serial number 029522250107 was found.

[ INFO] [1597914039.451308045]: Device with physical ID 2-3-9 was found.
[ INFO] [1597914039.451326783]: Device with name Intel RealSense D455 was found.
[ INFO] [1597914039.451936297]: Device with port number 2-3 was found.
[ INFO] [1597914039.454365173]: getParameters...
[ INFO] [1597914039.500001567]: setupDevice...
[ INFO] [1597914039.500030430]: JSON file is not provided
[ INFO] [1597914039.500041372]: ROS Node Namespace: camera
[ INFO] [1597914039.500052697]: Device Name: Intel RealSense D455
[ INFO] [1597914039.500063113]: Device Serial No: 029522250107
[ INFO] [1597914039.500073487]: Device physical port: 2-3-9
[ INFO] [1597914039.500085814]: Device FW version: 05.12.06.00
[ INFO] [1597914039.500105338]: Device Product ID: 0x0B5C
[ INFO] [1597914039.500122325]: Enable PointCloud: Off
[ INFO] [1597914039.500140457]: Align Depth: Off
[ INFO] [1597914039.500162653]: Sync Mode: Off
[ INFO] [1597914039.500222880]: Device Sensors: 
[ INFO] [1597914039.536214551]: Stereo Module was found.
[ INFO] [1597914039.566331872]: RGB Camera was found.
[ INFO] [1597914039.566590316]: Motion Module was found.
[ INFO] [1597914039.566630526]: num_filters: 0
[ INFO] [1597914039.566645242]: Setting Dynamic reconfig parameters.
[ INFO] [1597914044.956075186]: Done Setting Dynamic reconfig parameters.
[ INFO] [1597914044.957996114]: depth stream is enabled - width: 640, height: 480, fps: 30, Format: Z16
[ INFO] [1597914044.959490249]: color stream is enabled - width: 640, height: 480, fps: 30, Format: RGB8
[ INFO] [1597914044.959586521]: setupPublishers...
[ INFO] [1597914044.966149718]: Expected frequency for depth = 30.00000
[ INFO] [1597914045.041348192]: Expected frequency for color = 30.00000
[ INFO] [1597914045.067840065]: setupStreams...
[ INFO] [1597914045.171890682]: insert Depth to Stereo Module
[ INFO] [1597914045.172022331]: insert Color to RGB Camera
[ INFO] [1597914045.403407872]: SELECTED BASE:Depth, 0
[ INFO] [1597914045.433190559]: RealSense Node Is Up!
 20/08 17:00:45,543 WARNING [140714708702976] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 20/08 17:00:45,626 WARNING [140714708702976] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
.....

xdwgood, 2020-08-20:
How should I solve the `WARNING [140714708702976] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61` error?

MartyG-RealSense, 2020-08-20:
Hi @xdwgood   If you only encounter the warning **control_transfer returned** at the start of launch, or at a rate of about once a minute, then the RealSense ROS wrapper developer has said that it can be safely ignored so long as the topics are being correctly published.

If the warning is being generated in the log continuously however then this can indicate that there is a serious communication problem with the device.

May I first confirm the roslaunch instruction that is being used, please?  The title of the case says **rs_launch** (suggesting that live camera data is being used) and the first comment in the case says that **rs_from_file** (data from a recorded bag) is being used.  As the topic **/camera/realsense2_camera/rosbag_filename** is published, this suggests to me that it is a bag.  Thanks!

xdwgood, 2020-08-21:
@MartyG-RealSense  

> May I first confirm the roslaunch instruction that is being used, please

yes, I use roslaunch

I checked using `rostopic list` and all topics were published correctly. So this may not be a problem, but it will mislead newcomers like me.

May I ask, where is a better place to start learning realsense? 

@MartyG-RealSense Thanks again for your reply, good team! :heart:

closed 2020-08-21 by xdwgood

MartyG-RealSense, 2020-08-21:
Thanks for your kind words @xdwgood     :)

The official RealSense YouTube channel is a good place to start for a new RealSense user.

https://www.youtube.com/c/IntelRealSense/videos

The official RealSense blog is also excellent at writing about RealSense subjects in a beginner friendy way:

https://www.intelrealsense.com/blog/

Exploring the RealSense Viewer program will expose you naturally to a lot of the camera's operational aspects in a user-friendly format that is easy to experiment with.  The guide document for the Viewer is quite out of date now but is still good for explaining the basics.

https://www.intel.com/content/www/us/en/support/articles/000028593/emerging-technologies/intel-realsense-technology.html

After that, you could move on to the excellent RealSense camera tuning guide.

https://dev.intelrealsense.com/docs/tuning-depth-cameras-for-best-performance

The best way to learn RealSense ROS is to learn about the basics of librealsense first, as having knowledge of the fundamentals of how the camera works is vey useful when understanding what is happening during aspects of ROS such as launching.