realsenseai/librealsense #7141: How can I turn a video feed into point cloud?

sw14928, 2020-08-20:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515| 
| Firmware Version                   | 01.03.15.100 | 
| Operating System & Version |   Win (10) | 
| Platform                                 | PC |
| SDK Version                            |  2.0                         | 
| Language                            |  python}                          | 
| Segment			|  Robot                          | 

### Issue Description

How can I turn a video feed into a point cloud using python. I currently collect some frames and turn this into a point cloud using the following code:

import cv2                                # state of the art computer vision algorithms library
import numpy as np                        # fundamental package for scientific computing
import matplotlib.pyplot as plt           # 2D plotting library producing publication quality figures
from pyntcloud import PyntCloud # open source library for 3D pointcloud visualisation
import pyrealsense2 as rs                 # Intel RealSense cross-platform open-source API
import open3d as o3d

print("Environment Ready")

# Setup:
pipe = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.depth)
pipe.start(config)
colorizer = rs.colorizer()

# Skip 5 first frames to give the Auto-Exposure time to adjust
for x in range(5):
  pipe.wait_for_frames()
  
# Store next frameset for later processing:
frames = pipe.wait_for_frames()
#color_frame = frameset.get_color_frame()
#depth_frame = frameset.get_depth_frame()

colorized = colorizer.process(frames)

# Create save_to_ply object
ply = rs.save_to_ply("1.ply")

# Set options to the desired values
# In this example we'll generate a textual PLY with normals (mesh is already created by default)
ply.set_option(rs.save_to_ply.option_ply_binary, False)
ply.set_option(rs.save_to_ply.option_ply_normals, True)

print("Saving to 1.ply...")
# Apply the processing block to the frameset which contains the depth frame and the texture
ply.process(colorized)
print("Done")

input_file = "1.ply"
pcd = o3d.read_point_cloud(input_file)
o3d.draw_geometries([pcd])

What I really want to do is combine multiple frames into a point cloud. Is this possible? If so how please?

Many thanks

BIueMan, 2020-08-31:
i think there is a labrery on the SDK. "pyrealsense2.pointcloud"
but im not sure how to use it, would like an help with that as well

https://intelrealsense.github.io/librealsense/python_docs/_generated/pyrealsense2.pointcloud.html?highlight=pointcloud#pyrealsense2.pointcloud

RealSenseSupport, 2020-10-22:
Hi @sw14928 

Will you be needing further assistance with this? If so can you provide a little more detail on exactly what you're trying to accomplish?

Much Thanks

closed 2020-12-21 by RealSenseSupport