realsenseai/librealsense #7142: Timestamp: 1597937561226.082520 Severity: Error Description: RT IC2 Config error on 415 with latest firmware

davidkinetisense, 2020-08-20:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 } | 
| Firmware Version                   | 010423022664 | 
| Operating System & Version |   {Win (10) | 
| Kernel Version (Linux Only)    |                                  | 
| Platform                                 |  |
| SDK Version                            |                  | 
| Language                            |                            | 
| Segment			|                     | 

### Issue Description
Timestamp: 1597937561226.082520 Severity: Error Description: RT IC2 Config error on 415 with the latest firmware

When opening the realsense viewer this error is intermittent showing up.

davidkinetisense, 2020-08-20:
The software that we developed to use with the camera is hanging intermittently.  I am wondering if this error would be causing the issue.

MartyG-RealSense, 2020-08-20:
Hi @davidkinetisense   I researched past cases about the RT IC2 error being experienced when using the RealSense Viewer and sorted through the information provided in those cases.  The evidence of users' tests suggested that the problem was not related to USB cables (since the problem occurred whether using the official cable or an own choice of cable that was longer).  The popular conclusion reached by users was that the USB chipset on some computer models may make them more likely to have an RT IC2 error than other computer models.

Here is an example comment from one of the cases that I reviewed:

https://github.com/IntelRealSense/librealsense/issues/2045#issuecomment-533926710

davidkinetisense, 2020-08-20:
We are seeing this on all new Lenovo and Dells.  I am not sure if it is causing the freezing with our application.  Has there been any investigation to see what usb chipsets cause this and is there any fix coming?

MartyG-RealSense, 2020-08-20:
I am not aware of an official position regarding a cause of the RT IC2 Config error.  The suspicion of the USB chipset was based on the opinions of affected RealSense users who discussed the problem.  I would therefore not make recommendations for products without having very high confidence that they would provide a solution.

Browsing through reports of RT IC2 Config on the RealSense forums, a noticeable trend through is that Windows 10 is the OS involved rather than Linux.

MartyG-RealSense, 2020-09-03:
Hi @davidkinetisense   Do you require further assistance with this case please?   Thanks!

davidkinetisense, 2020-09-03:
HI, we are struggling with our unity program intermittently hanging and the
memory filling and eventually crashing.  It is a random problem and to this
point have not been able to determine the cause.  We were looking into this
error but no success.

David

MartyG-RealSense, 2020-09-03:
Thank you very much for the update.  Do you experience this memory issue only with librealsense in Unity, please?  For example, if you create a new empty scene and press play in the editor and monitor memory, do you experience the same memory leak?

davidkinetisense, 2020-09-03:
Hi, it is not a leak cause the hang,, it is the hang that causes the memory
to grow.  We are really struggling to pinpoint why it hangs and causes the
memory to grow.

One thing I noticed when Kinetisense releases the camera the RTC error
always happens.

[image: image.png]

Thanks, David

MartyG-RealSense, 2020-09-04:
Your image that you kindly provided by email was not displayed.  Could you post the image into this case using the web-page please?  Thanks!

I re-read your case from the start.  Some error messages that appear in the RealSense Viewer can also appear in other applications that the camera is used with, such as the RealSense ROS wrapper's log.    If you experience a **RT IC2 Config error** in the Viewer, there may be an underlying cause that causes your Unity project to freeze if the project is dependent on live camera data and the camera's operation is disrupted in some way.

How long has it been since the RealSense DLL files in your Unity project were updated, please?  They would not get automatically updated when upgrading to a newer version of the RealSense SDK.  So If you set up your Unity project a while ago, the RealSense DLLs in your project may have become outdated compared to the recent firmware driver version stored inside the camera.

nanotuxi, 2020-09-10:
Hi, the same RT IC2 Config error here on  a linux machine with latest librealsense installed from source and a D435i. Everything works fine (even the rs-server from a jetson nano. Thanks.)  It doesn't seem to be a problem with the usb cable as I used the original one which is working o.k. with all other examples and realsense-viewer for the D435i.

nanotuxi, 2020-09-10:
Got it working. Reading helps:-) The Readme.md in the librealsense/tools/rs-imu-calibration/ folder says it all. 
Instead saving the calibration data directly after having the calibration process finished I 
1. saved the data as accel_01.txt and gyro01.txt by answering the question, if to 

> save the calibration data with `Y` and entering `01`
and if to 
> write the data to eeprom with `N.`
Then I restarted the rs-imu-calibration tool again with the -i parameter like this
`python rs-imu-calibration.py -i accel_01.txt gyro_01.txt` 
The tool answered with the expected behaviour and said
`Would you like to write the results to the camera? (Y/N)Y
Writing calibration to device.
  Device PID:  0B3A
  Device name:  Intel RealSense D435I
  Serial number:  XXXXXXXXXXXX
  Firmware version:  05.12.07.100
SUCCESS: saved calibration to camera.`

MartyG-RealSense, 2020-09-10:
I was just about to reply to your case.  I'm very pleased to hear that you found a solution yourself.  Thanks for the update and for sharing the details of how you did it!

nanotuxi, 2020-09-10:
@MartyG-RealSense 
Sorry for asking befor reading. Thank you.

MartyG-RealSense, 2020-09-18:
Case closed due to solution found by customer and no further comments received.

closed 2020-09-18 by MartyG-RealSense