realsenseai/librealsense #7143: openvino-face-detection.cpp- Changing the thickness and transparency of bounding box

jackgrovery, 2020-08-20:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |D415 | 
| Firmware Version                   |05.12.06.00| 
| Operating System & Version |   ubuntu 18.04 | 
| Kernel Version (Linux Only)    |  (5.3.0-53-generic)                                         | 
| Platform                                 | HPZ4G4..  |
| SDK Version                            |  2.36.0                 | 
| Language                            |  {opencv/C++ }                          | 
| Segment			|  {Robot }                          | 

### Issue Description

Openvino face detection -Changing the bounding box to a solid shape /opaque

Hi All,

 

I was curious to know how to change the line thickness and transparency of the bounding box in https://github.com/IntelRealSense/librealsense/blob/master/tools/realsense-viewer/openvino-face-dete...

 
I tried changing the line thickness 1.4f to a larger value but dint succeed  cv::Mat face_image = image(openvino_helpers::adjust_face_bbox( rect, 1.4f )& cv::Rect( 0, 0, image.cols, image.rows ) );

 

**I want to  know how to draw a solid bounding box on the identified facial region**

 

// License: Apache 2.0. See LICENSE file in root directory.
// Copyright(c) 2020 Intel Corporation. All Rights Reserved.

// NOTE: This file will be compiled only with INTEL_OPENVINO_DIR pointing to an OpenVINO install!

#include "post-processing-filters-list.h"
#include "post-processing-worker-filter.h"
#include <rs-vino/object-detection.h>
#include <rs-vino/age-gender-detection.h>
#include <rs-vino/detected-object.h>
#include <cv-helpers.hpp>

/* We need to extend the basic detected_object to include facial characteristics
*/
class detected_face : public openvino_helpers::detected_object
{
float _age;
float _male_score, _female_score; // cumulative - see update_gender()

public:
using ptr = std::shared_ptr< detected_face >;

explicit detected_face( size_t id,
cv::Rect const& location,
float male_prob,
float age,
cv::Rect const& depth_location = cv::Rect{},
float intensity = 1,
float depth = 0 )
[... 296 lines cut ...]
}
catch( const std::exception & e )
{
LOG(ERROR) << get_context(fs) << e.what();
}
catch( ... )
{
LOG(ERROR) << get_context(fs) << "Unhandled exception caught in openvino_face_detection";
}
}

void on_processing_block_enable( bool e ) override
{
post_processing_worker_filter::on_processing_block_enable( e );
if( !e )
{
// Make sure all the objects go away!
std::lock_guard< std::mutex > lock( _objects->mutex );
_objects->clear();
}
}

};

static auto it = post_processing_filters_list::register_filter< openvino_face_detection >( "Face Detection : OpenVINO" );

RealSenseSupport, 2020-08-26:
Hi @jackgrovery,

Please take a look at :
https://github.com/IntelRealSense/librealsense/blob/master/common/viewer.cpp#L1812
and let us know if you need any additional help with this question.

Thank you!

jack1099, 2020-08-26:
Thank You. I tried the following

                    glDisable(GL_DEPTH_TEST);
                     glEnable(GL_BLEND);
                     glBlendFunc(GL_ONE, GL_ONE);
                     glColor4f( 1.f, 1.f, 1.f, 1.f);
                     draw_rect( bbox );

Though I added the alpha blend (alpha =1.f) in glcolor4f function the opacity of bounding box remain unchanged
[image]
.

Can you please guide me further on where exactly the opacity of the bounding box is reflected?

jack1099, 2020-08-28:
Thank you. I was able to successfully modify the box. 

However, the modified box is not getting saved in .bag format.

In fact, none of the post processing is saved in .bag format. Even, the normal face identified box is not written to a a.bag file.

Any suggestions on which file or line number I need to look at.

RealSenseSupport, 2020-09-01:
@jackgrovery,

Since the original issue with drawing bounding box was solved and you don't need any additional clarifications regarding the original issue, this thread should be closed. Please close it and discuss the issue with the bag file in a separate thread.

jack1099, 2020-09-01:
Okay. Thanks

closed 2020-09-01 by RealSenseSupport