realsenseai/librealsense #7144: Raspberry Pi 4 and Intel RealSense SDK 2.0

netanelfar, 2020-08-20:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------| 
| Platform                                 | Raspberry Pi  |
| SDK Version                            |  {  2.36.0 }                          | 
| Language                            |  {C++,python }                          | 
| Segment			|  {installation}                          | 

### Issue Description

is there a way to install the sdk (not the viewer) on the raspberry pi 4? 
i tried following the instructions in the following link:
https://github.com/acrobotic/Ai_Demos_RPi/wiki/Raspberry-Pi-4-and-Intel-RealSense-D435
but its seems that the fact that the raspberry is using arm and not x isa dose not allow me to properly install the sdk.
i tried to look for a proper solution but all i could find is a work arounds at best.
let me know if its possible or if i need to change my sbc.
thank you
netanel

MartyG-RealSense, 2020-08-21:
Hi @netanelfar   The RealSense 400 Series cameras can work with any Intel or Arm processor.  Librealsense certainly can be installed for Arm devices such as the Pi.  The Acrobotic installation guide for Pi 4 is actually similar in content to the official Intel installation guide for Pi 3. 

For a quick setup, Intel have a pre-built SD card image for Pi 4 that can be downloaded and installed.  This can act as a reference when learning how to install librealsense on Pi 4 yourself.

https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras#section-2-3-preparing-the-sd-card

An easier method of installation than the guides mentioned above may be to use a method called **libuvc-backend**.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

Could you try this method please and see how it goes?

netanelfar, 2020-08-24:
thank you marty.
another question when using frame.get_data what is the type of each pixel?(what is the size of it).
if i want to split the frame bit by bit and put it in a matrix.
thanks again

MartyG-RealSense, 2020-08-25:
The question of size may be covered by **get_data_size()**, which is described in the C++ discussion in the link below.

https://github.com/IntelRealSense/librealsense/issues/5718

**get_data_size**() retrieves data size from the frame handle and returns the number of bytes in a frame.

MartyG-RealSense, 2020-09-03:
Hi @netanelfar   Do you require further assistance with this case, please?   Thanks!

netanelfar, 2020-09-06:
@MartyG-RealSense Thank you for now i am good.
thanks again for all the help!

closed 2020-09-06 by netanelfar

MartyG-RealSense, 2020-09-06:
Thanks @netanelfar for the update!

AndreV84, 2022-03-21:
@MartyG-RealSense 
Howdy Marty?
Does the rpi image get updated?
Or as for today there is no somewhat recent rpi image with realsense support pre-built by intel? Thanks

MartyG-RealSense, 2022-03-21:
@AndreV84   As far as I am aware, the pre-built Pi SD card image in the open source ethernet white paper does not get updated from its original support for librealsense 2.34.0.

AndreV84, 2022-03-21:
so we would just build like that https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md ?
will it address the metadata for d455 on raspberry?
e.g. on arm jetson we would apply Jetson librealsense L4T patch
how to address it on rpi?

AndreV84, 2022-03-21:
@MartyG-RealSense

MartyG-RealSense, 2022-03-21:
The build script 'libuvc_installation.sh' uses the CMake build flag **-DFORCE_LIBUVC=true** and so metadata support should be included in the build (as DFORCE_LIBUVC=true is the predecessor of FORCE_RSUSB_BACKEND=true)

AndreV84, 2022-03-21:
@MartyG-RealSense 
Yes, but on Jetson we wold aditionally patch the kernel for d455
with librealsense script L4T
in addition to building with RSUSB

MartyG-RealSense, 2022-03-21:
Intel's official Jetson installation instructions offer a choice of building librealsense with the RSUSB Backend by using  the 'libuvc_installation.sh' script so that you do not need to apply a patch, or using the Native Backend and patching the kernel with the 'patch-realsense-ubuntu-L4T.sh' script.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md#building-from-source-using-rsusb-backend

If you build with LIBUVC / RSUSB then you should not be running a kernel patch script.