realsenseai/librealsense #7146: How is depth calculated by internal algorithm?

zzmwfhy, 2020-08-21:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |   LiDAR L515  | 
| SDK Version                            |   2.0                         | 

### Issue Description
Hello, my lab is working on a distance test with LiDAR L515, and I was asked to know how to calculate the distance.
We use "depth.get_distance(X, Y)", as the code sample ["rs-hello-realsense"](https://dev.intelrealsense.com/docs/rs-hello-realsense) to get the distance(Z) from the center of image to the camera. 
I found the calculation method which might be similar with this code on #2348:

> using namespace rs2;
> pipeline p;
> pipeline_profile profile = p.start();
> device dev = profile.get_device();
> depth_sensor ds = dev.query_sensors().front().as<depth_sensor>();
> float scale = ds.get_depth_scale();
> frameset fs = p.wait_for_frames();
> depth_frame depth = fs.get_depth_frame();
> uint16_t* data = (uint16_t*)depth.get_frame_data();
> uint16_t pixel = data[0];
> float meters = pixel * scale;

It looks that we can find the number of depth by "depth.get_frame_data()". However, can we know how is the value of data[0] calculated? In other words, how does the camera calculate the distance between the camera and the center of image? 
At start I though it uses (light speed * time)/2, but this method need a reaction time near picosecond based on its precision. It is hard to believe. Does anyone has any idea about that? Thanks.

RealSenseSupport, 2020-09-09:
Hi @zzmwfhy 

We don't necessarily get to in depth with regards to providing exactly how depth is calculated by our algorithm.  

We do have some documentation around the camera and the technology that should be helpful but again it won't provide the exacts around our algo calculations.

https://www.intelrealsense.com/optimizing-the-lidar-camera-l515-range/

https://www.intelrealsense.com/download/13061/

zzmwfhy, 2020-09-09:
@RealSenseSupport 
Thanks for your reply. I know my question may be not suitable, but this answer help me a lot as well. I found the two documents before I post this issue, but I didn't find what we want. Now I can tell the Party A that we can not get that. Thank you very much!
I will close it after this. Thanks

closed 2020-09-09 by zzmwfhy