realsenseai/librealsense #7149: Is "remove IR pattern" preset supported for model D455?

zc08, 2020-08-21:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 05.12.06.00 | 
| Operating System & Version |   Ubuntu 18.04.3  | 
| Kernel Version (Linux Only)    |  5.4.0-42-generic | 
| Platform                                 | PC |
| SDK Version                            |   v2.36.0                          | 
| Language                            |  None                          | 
| Segment			| None     | 

### Issue Description
I have a latest D455, which seems to work fine with realsense viewer v2.36.0 under Ubuntu 18.04.3. However, when I choose the "remove IR pattern" preset, it gives me an error saying "the device doesn't support Remove IR Pattern Preset". 

Some previous threads indicate that this functionality only works for D415 
 (https://github.com/IntelRealSense/librealsense/issues/6248), but I'm wondering is it supported (or any plan for supporting) for the latest model D455? Thanks!

MartyG-RealSense, 2020-08-21:
Hi @zc08  Yes, the **Left Imager Color w/o IR pattern** is for D415 only

https://dev.intelrealsense.com/docs/d400-series-visual-presets#section-preset-table

You can though remove the pattern by disabling the **IR Emitter**.  In the RealSense Viewer's options side-panel, you can do this by left-clicking on the drop-down menu beside the **Emitter Enabled** option, under the Stereo Module section, and selecting the **Off** option from the menu.  To turn it back on, select **Laser** from the same menu.

[image]

Alternatively, the dot pattern can have its visibility reduced without disabling the IR Emitter by reducing the value of the **Laser Power** option under the **Controls** category of the Stereo Module options.  The lower the Laser Power value, the less visible the dot pattern will be.  Increasing the value increases the pattern's visibility.  A drawback of this method is that reducing Laser Power can make the depth image more sparse.

zc08, 2020-08-24:
Thank you all for the helpful answers! 
I'm still wondering when will this function be available (at least in the bleeding-edge repository so I could have a try) ? I'm really in bad need of this. Thank you!

MartyG-RealSense, 2020-09-03:
Hi @agrunnet   Was the "remove IR" color correction for D455 included in the new **5.12.7.100** firmware please?   Thanks!

agrunnet, 2020-09-03:
Sorry. Not in that version. Coming soon.

MartyG-RealSense, 2020-09-03:
Thanks very much for the information @agrunnet !

MartyG-RealSense, 2020-09-21:
Case closed due to no further comments received

closed 2020-09-21 by MartyG-RealSense

bs97, 2022-06-23:
Is this preset still being added?

MartyG-RealSense, 2022-06-23:
@bs97  I will ask Intel about your question.  Thanks for your patience!

MartyG-RealSense, 2022-06-23:
Hi @bs97  The **Remove IR pattern** preset was created to address an issue with the projector on the D415 model specifically that does not occur on the D455 model.  The D455 model therefore does not require this preset.

TucoFernandes, 2023-01-04:
@MartyG-RealSense I'm having the same issue with D435. When collecting images from objects at close range (0.3 m) the depth frame gets distorted if there is too much light in the scene. But, if I reduce the light to remove glare and improve depth the IR pattern shows up on the color image. Any ideas?

MartyG-RealSense, 2023-01-04:
Hi @TucoFernandes  The IR pattern should not be leaking through to the color image, though it can happen in very rare cases.  If the scene is well lit then you could try disabling the IR emitter, as RealSense 400 Series cameras can alternatively use ambient light in a scene to analyze surfaces for depth detail instead of analyzing the IR dot pattern.

You could also reduce the visibility of the dot pattern instead of removing it completely by reducing the value of the **Laser Power** option.

A third possibility would be to fit a **thin-film linear polarization filter** over the lenses on the outside of the camera to significantly reduce glare from reflections.  These filters can be purchased affordably by searching stores such as Amazon for the term **linear polarizing film sheet**

Polarizing filters are discussed in section 4.3 of Intel's white-paper guide about using optical filter products with 400 Series cameras.

https://dev.intelrealsense.com/docs/optical-filters-for-intel-realsense-depth-cameras-d400#4-the-use-of-optical-filters

The image below taken from this guide section demonstrates the benefits of using this filter type to negate glare.

[image]

TucoFernandes, 2023-01-04:
@MartyG-RealSense thanks for the fast reply, I will try to buy and try the filter. But do you have any idea why the IR pattern shows for the pyrealsense color image but not for the realsense viewer color image? I had the idea that the D345 had a separate color image and the IR would not show on it.

MartyG-RealSense, 2023-01-04:
Yes, D435 has a separate color sensor, and the infrared dot pattern is not supposed to appear on the RGB image.  It is highly rare and not clear why it occurs.  

The RealSense Viewer applies a range of post-processing filters by default, whilst self-created scripts do not apply these settings and they have to be manually programmed into a script in order to generate an image that more closely resembles those of the Viewer.  You could try left-clicking on the blue icon next to the **Post-Processing** option under 'Stereo Module' in the Viewer's options side-panel to disable all post-processing filters and see the effect that this has on the image.

TucoFernandes, 2023-01-04:
@MartyG-RealSense  I'm generating the camera config json file using the intel realsense viewer and than loading it on my custom code. To me, it looks like some of the parameters are not being loaded. I will check them one by one then.

MartyG-RealSense, 2023-01-04:
You are vorrect.  Not all Viewer options are exported to a json file.