realsenseai/librealsense #7160: Pointcloud Filtering

upasana27, 2020-08-23:
How to proceed with filtering the point cloud obtained from the point cloud example? Since the data structure belongs to rs2 library, how can we access the data? 
Thanks and Regards

MartyG-RealSense, 2020-08-24:
Hi @upasana27   Are you referring to applying post-processing filters to the point cloud data please?  If you are, then the Intel guide to post-processing should be helpful.

https://dev.intelrealsense.com/docs/depth-post-processing

These filters can be applied in the RealSense Viewer program.  If you are planning to do filtering using programming code, could you also please let us know which programming language (C++, C# or Python) you are using so that script references can be suggested.   Thanks!

upasana27, 2020-08-24:
Hi @MartyG-RealSense, thanks for replying.  I am using the following pointcloud example in C++: https://github.com/IntelRealSense/librealsense/tree/master/examples/pointcloud 
Regards

MartyG-RealSense, 2020-08-24:
Thanks very much for the information.  The SDK has a C++ tutorial for defining post-processing filters at the link below:

https://github.com/IntelRealSense/librealsense/tree/master/examples/post-processing

Further documentation about post-processing filters is here:

https://github.com/IntelRealSense/librealsense/blob/master/doc/post-processing-filters.md

upasana27, 2020-08-28:
Hi @MartyG-RealSense, thank you for the help. I am looking forward to filter the generated point cloud using voxel leaf filtering snf noise removal. I wanted to use PCL library for the same. Is there any recommended wrapper or code for converting the point cloud generated in the example into pcl format?
https://community.intel.com/t5/Items-with-no-label/Converting-RealSense-Pointcloud-to-PCL-Pointcloud/td-p/541957 
This is a code which is available, however it generates a single point cloud rather than a stream.  
Please let me know if there is any script which has been used reliably before for the use case.

Kind regards
Upasana

MartyG-RealSense, 2020-08-28:
I have handled a couple of cases in the past where a voxelized point cloud has been generated using PCL.  The links are below.  I hope that you find some useful insights within them.

https://github.com/IntelRealSense/librealsense/issues/6411
https://github.com/IntelRealSense/librealsense/issues/6802

upasana27, 2020-08-30:
Hi @MartyG-RealSense, thank you for the guidance. The issue has been solved. Closing the thread.

closed 2020-08-30 by upasana27