realsenseai/librealsense #7162: Jetpack 4.4  no pyrealsense2 in /usr/local/lib

jysah, 2020-08-24:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435i}                                                   | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Jetson Nano Jetpack 4.4}                     | 
| Kernel Version (Linux Only)    |  (Kernel 4.9(Jetpack 4.4))                         | 
| Platform                                 | NVIDIA Jetson Nano                               |
| SDK Version                            |  { legacy / 2.<31>.<0> }                          | 
| Language                               |  {python}                                                 | 
| Segment		  	         |  {Autonomous Driving}                           | 

### Issue Description
<question>
I'd like to build Librealsense on Ubuntu 18.04 (Kernel 4.9) of Jetson Nano with Jetpack 4.4.
My problem is that there is no pyrealsense2.so in /usr/local/lib directory after build.
realsense-viewer and librealsense2.so(in /usr/local/lib) are fine. 

Instead of pyrealsense2.so, pyrealsense2.cpython-36m-aarch64-linux64-linux-gnu.so is created in /usr/local/lib.

I did "export PYTHONPATH=$PYTHONPATH:/usr/local/lib".

Without pyrealsense2.so (even if with pyrealsense2.cpython-36m-aarch64-linux64-linux-gnu.so) in /usr/local/lib,  
pyrealsense2 module does not work.

I followed http://github.com/Jetsonhacks/installRealSenseSDK.
I used the modified buildLibrealsense.sh with NVCC_PATH=/usr/local/cuda-10.2/bin/nvcc on Line 11 (10.2 not 10.0).

Fortunately, a few days ago, I managed to get it right. (I have pyrealsense2.so in /usr/local/lib).
But I can't remember the exact process. 
......(maybe some setups...... ex: sudo apt-get install (??? such as libxinerama-dev etc)
installLibrealsense.sh
buildLibrealsense.sh

I tried so many cases, but I couldn't reproduce the success.

MartyG-RealSense, 2020-08-24:
Hi @jysah   If you are building pyrealsense2 for Python 3, it is normal for the pyrealsense2.so file to have a long and unusual filename that is auto-generated based on the version number of Python that you are using (e.g using Python 3.6 may result in the filename containing the term **cpython-36m**).   It is simply called pyrealsense2.so if you are building the python wrapper for Python 2.7.  The Linux section of the wrapper's installation instructions provide the following comment about this:

"Python 3 module filenames may contain additional information, e.g. pyrealsense2.cpython-35m-arm-linux-gnueabihf.so)"

If the .so filename in your successful earlier build was **pyrealsense2.so** then that suggests to me that pyrealsense2 had been compiled for Python 2.7 instead of 3.6 on that occasion.  So Python 3.6. is likely installed on your Nano if the installation process gave the .so file that auto-generated name in the more recent attempt.

The installation method that you used - build with the JetsonHacks script and then build the pyrealsense2 wrapper - has worked for other Jetson Nano users.  So let's stick with that method, as it is the one that you are most familiar with and so most likely to replicate your earlier success with.

It looks as though the JetsonHacks guide that you linked to is for Jetsons in general, whilst the guide in the link below, which has similar instructions, is specific to Jetson Nano:

https://github.com/JetsonHacksNano/installLibrealsense

This guide confirms that the librealsense build instructions with the JetsonHacks method are:

**installLibrealsense.sh
buildLibrealsense.sh**

I believe that the instructions from that point on to install the Python wrapper are the ones in the official wrapper installation instructions for building from source:

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python#building-from-source

**Ensure apt-get is up to date**
sudo apt-get update && sudo apt-get upgrade

**I would assume the next step can be skipped if you already have Python 3 installed**
sudo apt-get install python3 python3-dev

**Build the pyrealsense2 wrapper**
mkdir build
cd build
cmake ../ -DBUILD_PYTHON_BINDINGS:bool=true -DPYTHON_EXECUTABLE=/usr/bin/python

if  -DPYTHON_EXECUTABLE=/usr/bin/python does not work for you then you could try just defining it as one of these:

-DPYTHON_EXECUTABLE=...

or

-DPYTHON_EXECUTABLE=/usr/bin/python3

**Update your PYTHONPATH**
export PYTHONPATH=$PYTHONPATH:/usr/local/lib

jysah, 2020-08-24:
Hi @MartyG-RealSense You are right. I did not realize it even though I was successful.

I add the followings in .bashrc.
alias python='python3'
export PYTHONPATH=$PYTHONPATH:/usr/local/lib

When I run rs-imu-calibration.py, it works fine.  
python rs-imu-calibration.py

I has been almost there. 

Tanks a lot.

MartyG-RealSense, 2020-08-25:
Great news that it is working for you - thanks for the update!

closed 2020-08-25 by jysah