realsenseai/librealsense #7166: D435i not getting connected in Realsense Viewer.

AteequeFox, 2020-08-24:
Hi,
        I have purchased T265 and D435i, After getting the cameras I have followed the get started procedure. I am able to see the T265 camera in realsense viewer but unable to see the D435i.
        Even after following many solutions i am still unable to connect the D435i.
I am using Windows 10 64 bits OS. I am able to see depth and RGB cameras under the device manager. But Theirs error in HID. I have updated the drivers also but nothing solved my issue.

xenc0d3r, 2020-08-24:
@AteequeFox can you share the realsense error logs and warnings when you try to start the d435.

AteequeFox, 2020-08-24:
> ATTENTION: This email is sent by an external sender. Do not click
> links or open attachments unless you recognize the sender and know the
> content is safe.
>
>
> @AteequeFox <https://github.com/AteequeFox> can you share the
> realsense error logs and warnings when you try to start the d435.

Hi Greetings,

           PFA screen shot of the realsense viewer and device manager 
when D435i camera is connected.

Kind regards

*Ateequrrehman KHAN*

*Asst. Manager Projects (SPM) *

Cell : +919423076665

“PLAN”TO ACHIEVE- ACHIEVE“THE” PLAN

Fox.jpg

FOX Solutions

www.foxindia.net

Part A,Shed no. 7, Gat No. 439

Mumbai-Nashik Highway

Gonde Dumala,

Igatpuri, Nashik-422403

Tel          : +91-253-6621210

Fax         : +91-253-6618108 / 100

xenc0d3r, 2020-08-24:
You did not post the screenshot.

AteequeFox, 2020-08-25:
I have mailed the screen shots. Please it find attached here also.
[image]
[image]

AteequeFox, 2020-08-25:
Hi xenc0d3r,
         Any update after checking the shared screen shots?

RealSenseSupport, 2020-08-31:
@AteequeFox Please refer to https://github.com/IntelRealSense/librealsense/issues/3644#issuecomment-479043117 to see if it helps on your issue. Thanks!

RealSenseSupport, 2020-09-09:
@AteequeFox  Did you get chance to try if https://github.com/IntelRealSense/librealsense/issues/3644#issuecomment-479043117 helps? Looking forward to your update. Thanks!

RealSenseSupport, 2020-09-15:
@AteequeFox Any update from your side? Thanks!

AteequeFox, 2020-09-15:
#3644 also didn't resolve the issue. Seems some kind of a hardware issue. Hardware needs to be returned.

RealSenseSupport, 2020-09-15:
@AteequeFox Could you please type "winver" to check which build version of your windows OS? Please note that you need to update Build 15063 or later. Thanks!

AteequeFox, 2020-09-15:
Yes Winver is 1909 (OS build 18363.1082)
As per attached screen shot.
[image]

RealSenseSupport, 2020-09-15:
@AteequeFox Then can you run sensordiagnostictool.exe successfully on your PC? Just as https://github.com/IntelRealSense/librealsense/issues/3644#issuecomment-479043117 mentioned?

AteequeFox, 2020-09-21:
I have ran the Sensor Explorer in that I am unable to see any sensor connected.

RealSenseSupport, 2020-09-21:
@AteequeFox So I suppose you already enabled location services by "Settings -> Privacy -> Location", right? Any other PC do you have to try the camera? Let's check it's camera issue or OS-related issue.

AteequeFox, 2020-09-21:
Ok I will check and update here.

RealSenseSupport, 2020-09-27:
@AteequeFox Any update from your side? Looking forward to your reply. Thanks!

RealSenseSupport, 2020-10-10:
@AteequeFox Did you get it through? Looking forward to your reply. Thanks!

RealSenseSupport, 2020-10-15:
@AteequeFox Could you please update? Please note that this will be closed if we don't hear from you for 7 days.

RealSenseSupport, 2020-10-21:
@AteequeFox  Sorry that we didn't get response from you for weeks. This issue will be closed at this point. Please feel free to create new one if you still have issues or questions. Thanks!

closed 2020-10-21 by RealSenseSupport