realsenseai/librealsense #7168: Connection issues with D415

smits, 2020-08-24:
I've been looking at similar reports already, but none seem to entirely fix my problem

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | N/A | 
| Operating System & Version |  Linux (Ubuntu 16.04.7) | 
| Kernel Version (Linux Only)    |  4.4.0-186-generic  | 
| Platform                                 | PC  |
| SDK Version                            |  2.34.0 }                          | 
| Language                            |  -                          | 
| Segment			|  -                        | 

### Issue Description

`rs-enumerate-devices` does not show my connected RealSense camera:
```
$ rs-enumerate-devices
No device detected. Is it plugged in?
```

I also see no `/dev/video0|1|2` devices

`lsusb` shows that the camera is connected:
```
$ lsusb
Bus 002 Device 002: ID 8086:0ad3 Intel Corp.
...
```
And I verified if it is connected to a USB 3 port:
```
$ lsusb -t
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/10p, 10000M
    |__ Port 2: Dev 2, If 0, Class=Video, Driver=, 5000M
    |__ Port 2: Dev 2, If 1, Class=Video, Driver=, 5000M
    |__ Port 2: Dev 2, If 2, Class=Video, Driver=, 5000M
    |__ Port 2: Dev 2, If 3, Class=Video, Driver=, 5000M
    |__ Port 2: Dev 2, If 4, Class=Video, Driver=, 5000M
```

But the kernel/syslog shows the following error:
```
Aug 24 11:54:36 pc-50779 kernel: [ 4927.156989] usb 2-2: new SuperSpeed USB device number 2 using xhci_hcd
Aug 24 11:54:36 pc-50779 kernel: [ 4927.173601] usb 2-2: New USB device found, idVendor=8086, idProduct=0ad3
Aug 24 11:54:36 pc-50779 kernel: [ 4927.173609] usb 2-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Aug 24 11:54:36 pc-50779 kernel: [ 4927.173616] usb 2-2: Product: Intel(R) RealSense(TM) Depth Camera 415
Aug 24 11:54:36 pc-50779 kernel: [ 4927.173621] usb 2-2: Manufacturer: Intel(R) RealSense(TM) Depth Camera 415
Aug 24 11:54:36 pc-50779 kernel: [ 4927.173626] usb 2-2: SerialNumber: 933623024506
Aug 24 11:54:36 pc-50779 kernel: [ 4927.174797] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 415  (8086:0ad3)
Aug 24 11:54:36 pc-50779 kernel: [ 4927.174802] ------------[ cut here ]------------
Aug 24 11:54:36 pc-50779 kernel: [ 4927.174816] WARNING: CPU: 9 PID: 4293 at /build/linux-OgLptz/linux-4.4.0/drivers/media/media-device.c:380 __media_device_register+0xf0/0x140 [media]()
Aug 24 11:54:36 pc-50779 kernel: [ 4927.174820] Modules linked in: uvcvideo(OE) videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_core v4l2_common videodev(OE) media hid_sensor_accel_3d(OE) hid_sensor_gyro_3d(OE) hid_sensor_trigger industrialio_triggered_buffer kfifo_buf industrialio hid_sensor_iio_common hid_sensor_hub xt_conntrack iptable_filter xt_REDIRECT nf_nat_redirect xt_tcpudp iptable_nat nf_conntrack_ipv4 nf_defrag_ipv4 nf_nat_ipv4 nf_nat nf_conntrack ip_tables x_tables nvidia_uvm(OE) ipmi_ssif snd_hda_codec_hdmi intel_rapl x86_pkg_temp_thermal snd_hda_codec_realtek intel_powerclamp snd_hda_codec_generic ppdev coretemp kvm_intel snd_hda_intel snd_hda_codec kvm snd_hda_core snd_hwdep joydev snd_pcm irqbypass input_leds snd_timer snd shpchp soundcore serio_raw wmi parport_pc 8250_fintek parport ipmi_si mac_hid acpi_pad autofs4 jitterentropy_rng drbg ansi_cprng algif_skcipher af_alg uas usb_storage dm_crypt hid_generic usbhid hid nvidia_drm(POE) nvidia_modeset(POE) drm_kms_helper syscopyarea sysfillrect sysimgblt fb_sys_fops drm nvidia(POE) crct10dif_pclmul crc32_pclmul ghash_clmulni_intel aesni_intel aes_x86_64 lrw gf128mul glue_helper ablk_helper cryptd igb dca psmouse ptp pps_core ahci i2c_algo_bit libahci ipmi_msghandler fjes video
Aug 24 11:54:36 pc-50779 kernel: [ 4927.174970] CPU: 9 PID: 4293 Comm: kworker/9:0 Tainted: P           OE   4.4.0-186-generic #216-Ubuntu
Aug 24 11:54:36 pc-50779 kernel: [ 4927.174974] Hardware name: Default string Default string/Default string, BIOS 5.13 08/29/2018
Aug 24 11:54:36 pc-50779 kernel: [ 4927.174984] Workqueue: usb_hub_wq hub_event
Aug 24 11:54:36 pc-50779 kernel: [ 4927.174989]  0000000000000286 04cab61c63652a8a ffff880267ff7710 ffffffff8140ebbb
Aug 24 11:54:36 pc-50779 kernel: [ 4927.174996]  0000000000000000 ffffffffc1844380 ffff880267ff7748 ffffffff810864b2
Aug 24 11:54:36 pc-50779 kernel: [ 4927.175003]  ffff88026db4b070 ffff88026db4b070 ffff88026db4b000 0000000000000004
Aug 24 11:54:36 pc-50779 kernel: [ 4927.175009] Call Trace:
Aug 24 11:54:36 pc-50779 kernel: [ 4927.175022]  [<ffffffff8140ebbb>] dump_stack+0x6d/0x92
Aug 24 11:54:36 pc-50779 kernel: [ 4927.175033]  [<ffffffff810864b2>] warn_slowpath_common+0x82/0xc0
Aug 24 11:54:36 pc-50779 kernel: [ 4927.175040]  [<ffffffff810865fa>] warn_slowpath_null+0x1a/0x20
Aug 24 11:54:36 pc-50779 kernel: [ 4927.175049]  [<ffffffffc1842a40>] __media_device_register+0xf0/0x140 [media]
Aug 24 11:54:36 pc-50779 kernel: [ 4927.175061]  [<ffffffffc18d6b30>] uvc_probe+0x490/0x2a70 [uvcvideo]
Aug 24 11:54:36 pc-50779 kernel: [ 4927.175070]  [<ffffffff8164953d>] usb_probe_interface+0x11d/0x300
Aug 24 11:54:36 pc-50779 kernel: [ 4927.175079]  [<ffffffff8157d4d2>] driver_probe_device+0x1c2/0x4b0
Aug 24 11:54:36 pc-50779 kernel: [ 4927.175087]  [<ffffffff8157d8c4>] __device_attach_driver+0x74/0xa0
Aug 24 11:54:36 pc-50779 kernel: [ 4927.175094]  [<ffffffff8157d850>] ? __driver_attach+0x90/0x90
[... 64 lines cut ...]
Aug 24 11:54:36 pc-50779 kernel: [ 4927.176245]  [<ffffffff8157c4b5>] bus_probe_device+0x95/0xa0
Aug 24 11:54:36 pc-50779 kernel: [ 4927.176251]  [<ffffffff81579fbe>] device_add+0x3de/0x640
Aug 24 11:54:36 pc-50779 kernel: [ 4927.176257]  [<ffffffff81541600>] ? random_poll+0x60/0x80
Aug 24 11:54:36 pc-50779 kernel: [ 4927.176265]  [<ffffffff8163c087>] usb_new_device+0x277/0x4b0
Aug 24 11:54:36 pc-50779 kernel: [ 4927.176272]  [<ffffffff8163d867>] hub_port_connect+0x5e7/0xaf0
Aug 24 11:54:36 pc-50779 kernel: [ 4927.176279]  [<ffffffff8163e441>] hub_event+0x6d1/0xb20
Aug 24 11:54:36 pc-50779 kernel: [ 4927.176286]  [<ffffffff810a0d5b>] process_one_work+0x16b/0x4e0
Aug 24 11:54:36 pc-50779 kernel: [ 4927.176291]  [<ffffffff810a111e>] worker_thread+0x4e/0x590
Aug 24 11:54:36 pc-50779 kernel: [ 4927.176297]  [<ffffffff810a10d0>] ? process_one_work+0x4e0/0x4e0
Aug 24 11:54:36 pc-50779 kernel: [ 4927.176303]  [<ffffffff810a7857>] kthread+0xe7/0x100
Aug 24 11:54:36 pc-50779 kernel: [ 4927.176310]  [<ffffffff81864a51>] ? __schedule+0x301/0x810
Aug 24 11:54:36 pc-50779 kernel: [ 4927.176317]  [<ffffffff810a7770>] ? kthread_create_on_node+0x1e0/0x1e0
Aug 24 11:54:36 pc-50779 kernel: [ 4927.176325]  [<ffffffff81869da5>] ret_from_fork+0x55/0x80
Aug 24 11:54:36 pc-50779 kernel: [ 4927.176331]  [<ffffffff810a7770>] ? kthread_create_on_node+0x1e0/0x1e0
Aug 24 11:54:36 pc-50779 kernel: [ 4927.176335] ---[ end trace e95729b208659ac8 ]---
```

I verified if the uvcvideo kernel module is the one provide by `librealsense2-dkms`:
```
$ modinfo uvcvideo | grep "version:"
version:        1.1.2.realsense-1.3.11
srcversion:     AF7B446C80291E3076A1144
```

I have no clue how to get this fixed, has anyone seen this before?

MartyG-RealSense, 2020-08-24:
Hi @smits  I would recommend using a librealsense SDK version newer than 2.34.0, as that particular version was affected by a continuously generating error called ** get_device_time_ms() took too long (more then 2 mSecs)**.  Even if it is unrelated to your camera detection problem, it is a good idea from a stability viewpoint to use a newer SDK version.

https://github.com/IntelRealSense/librealsense/issues/6189

You could test whether your detection problem is related to the kernel by using a librealsense method called **RSUSB backend**.  It requires an internet connection but is not dependent on Linux versions or kernel versions and does not require patching.

https://github.com/IntelRealSense/librealsense/issues/6845#issuecomment-659540316

smits, 2020-08-25:
Using the RSUSB backend fixed my issue (using just a newer version from debians, 2.36.0 was not),  since we want to use this in production systems I was wondering where I can find the debian files that generate the debian packages so we could easily rebuild them with the RSUSB flag.

MartyG-RealSense, 2020-08-25:
Debian packages are obtained from an Amazon AWS server, as described by the **Linux distribution** installation instructions.

https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

If you are seeking the source code for the packages, the instructions state that "The packages include binaries and configuration files only.  Use the github repository to obtain the source code".

smits, 2020-08-25:
FYI, if I update the system to use a more recent kernel by installing `linux-generic-hwe-16.04` (which provides 4.15 over 4.4) I can use the provided debians out of the box. I do see some messages that indicate something is still not quite right (like UVC settings that fail), but it does not seem to prevent us from using the sensor as we want.

MartyG-RealSense, 2020-08-25:
Might you have a DKMS package installation and a manual source-code installation installed on the same computer and there is a conflict?

smits, 2020-08-25:
Both tests were on different clean systems, so I'm pretty sure that is not the case.

MartyG-RealSense, 2020-08-25:
You could try updating the DKMS packages if you have not done so already.

https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#upgrading-the-packages

MartyG-RealSense, 2020-09-03:
Hi @smits   Do you require further assistance with this case, please?  Thanks!

smits, 2020-09-07:
@MartyG-RealSense , for now we have a proper solution (use a more recent supported kernel) so no further assistance required.  I can tell you that the dkms packages were installed using apt and up to date.  It was installed on a clean system without a manual source-code installation in place.

MartyG-RealSense, 2020-09-07:
Okay, I will close the case if you do not require further assistance.  Thanks so much for the update and the details of what worked for you!

closed 2020-09-07 by MartyG-RealSense