realsenseai/librealsense #7170: D435i Color Camera FOV

01717, 2020-08-24:
<question>
In D435i Color Sensor, I’m aware that OV2740 supports 1920x1080 (16:9) as active pixels.
Regarding the spec "RGB Sensor FOV (H × V × D): 69.4° × 42.5° × 77° (±3°)" , does this spec assume 1920x1080 (16:9) ?
If RGB image captured resolution is changed from 1920x1080 (16:9) to 640x480 (4:3) on D435i, does FOV change?
If Yes, what value is it?
In RGB image 640x480 (4:3) it seems that FOV angle is lower than H:69.4°.

bmegli, 2020-08-27:
If you have the camera you may get intrinsics and calculate FOV for current resolution using librealsense `rs2_fov` function.

See example in [librealsense API-How-To](https://github.com/IntelRealSense/librealsense/wiki/API-How-To#get-field-of-view).

RealSenseCustomerSupport, 2020-08-28:
----------------------------------------------

Hi @01717
Yes, your finding is correct that D435i's RGB has smaller HFOV  (cropped) at 640x480 and 320x240 resolution, the HFOV is 55 degree

closed 2020-09-01 by RealSenseCustomerSupport