realsenseai/librealsense #7172: obtaining colorized depth data for further manipulation.

arothenberg, 2020-08-24:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |   D435  | 
| Firmware Version                   | 05.12.02.100 | 
| Operating System & Version |   Win (10)  | 
| Platform                                 | PC |
| SDK Version                            | 2.32                      | 
| Language                            |  C++                      | 
| Segment			|desktop app  | 

### Issue Description
Adjusting the rs-align-advanced code:
I use the realsense colorizer on a depth depth frame and it returns a frame object.

```
...                  	
rs2::colorizer c;                     // Helper to colorize depth images
 ...
rs2::depth_frame aligned_depth_frame = processed.get_depth_frame();
   
//new code
rs2::frame **depth_image** = c.process(aligned_depth_frame); 
 ...		
```

I then try and use and manipulate the **depth_image** frame. iow I want a depth mapping(depth to colors) and then 
I want to "draw" (change certain color values) in the new frame/table. 

I look at the frame data returned from **depth_image** and I'm not sure what I am looking at ex. 25 '\x19' 

What is an way to to obtain a table of color values representing depths that I can then manipulate? Do I have to use custom mapping algorithms?

MartyG-RealSense, 2020-08-24:
Another way to look at it may be to get a depth value from an aligned color.  If that is acceptable to you then the C++ code in the link below may provide useful insights.

https://github.com/IntelRealSense/librealsense/issues/2231#issuecomment-412804368

arothenberg, 2020-08-24:
That's probably more complicated than I need. I just need to map distance to color, any scheme will do. I was hoping not to use matlab  and just use realsense. I haven't worked much with matlab.  Realsense does the translation with the texture class's render function(Is that right?) consuming a frame created by using the realsense colorizer on a depth table. 

I was hoping maybe to get the color values through realsense since it somehow does that translation. 

I suppose I could render the colormap using the rs-align code as an example and then just change the resulting texture. I'm not sure how slow that is. 
I know in rs-align-advanced that memset is used when looping through the video_frame data in order to 
to shade the "background". I was planning on doing a similar type of thing with a table of the depth map color values.

arothenberg, 2020-08-24:
ie I want to access the depth image data that the colorizer creates. Is that possible or even make sense?

arothenberg, 2020-08-24:
I'll just use the distance as an intensity for a table of  fixed rgb values. I'll leave this open for a day or 2 just to see if there are other 
suggestions.  I'll re open if I can't make the solution work. Thanks.

MartyG-RealSense, 2020-08-25:
I'm not sure if this matches what you were asking for, but Intel's white-paper document on depth image compression with colorization has a section that discusses recovering the depth image from a colorized image using C++.

https://dev.intelrealsense.com/docs/depth-image-compression-by-colorization-for-intel-realsense-depth-cameras#section-3-2-depth-image-recovery-from-colorized-depth-images-in-c

arothenberg, 2020-08-25:
I'll check it but I need the opposite.  I need to create a simple color map of the depth values. One that I can then draw on .
Realsense sdk does this mapping using their "colorizer" function. I was hoping that, since realsense already does the translation 
from depth to some color scheme, I could  use their translation algorithm  or , even better, I could use the return value(a frame) from the function. 
rs2::frame depth_image = c.process(aligned_depth_frame);

But, i have another solution I'll try, which is to create a simple depth image myself by using the depth as  intensity and I'll have the color be constant.  If I can't do that maybe I'll open another issue and try and be clearer. Thanks for the help. I'll check the article.

arothenberg, 2020-08-25:
Yeah, I was wrong, that article seems to be what I'm looking for. 
Thanks.

MartyG-RealSense, 2020-08-25:
I'm pleased to hear that the link was helpful.  Thanks very much   :)

arothenberg, 2020-08-27:
Before I close this I looked into the texture class in example.hpp and it helped resolve my problem.

closed 2020-08-27 by arothenberg

MartyG-RealSense, 2020-08-27:
Excellent - thanks very much for the update!