realsenseai/librealsense #7173: Gathering IR stream intensity values

JRads, 2020-08-24:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  L515 | 
| Firmware Version                   | 01.04.01.02| 
| Operating System & Version |   Win 10 | 
| Platform                                 | PC |
| SDK Version                            |  {2.36.0 }                          | 
| Language                            |  {C++}                          | 
 

Hello,

I am currently working with the Intel L515 Lidar camera, and am interesting in accessing the intensity data for a given pixel in the IR stream.  I have gone through some of the sample code, but have not found anything that allows for extracting the intensity value from the IR stream.

Is there a function that contains this information?

I hope this is enough information, as I am new to programming,  and I can provide more if need be.

RealSenseSupport, 2020-09-14:
Hi @JRads 

We do not provide the intensity value for a given pixel in the IR stream.

Since you are new to the product, I do however want to point you to some documentation that may be of assistance as you are using the L515:

https://www.intelrealsense.com/optimizing-the-lidar-camera-l515-range/

https://www.intelrealsense.com/download/13061/

RealSenseSupport, 2020-10-22:
Hi

Is there anything else you'll be needing with regards to this thread?

*If we don’t hear from you in 7 days, this issue will be closed.

closed 2020-11-10 by RealSenseSupport

ScottYang1982, 2020-12-10:
> Hi @JRads
>
> We do not provide the intensity value for a given pixel in the IR stream.
>
> Since you are new to the product, I do however want to point you to some documentation that may be of assistance [...]

Feel really disappointed about this information. in documents intensity's referred to but no access? If it's there why no access to it?

yhecarpio, 2022-03-03:
> Hi @JRads
>
> We do not provide the intensity value for a given pixel in the IR stream.
>
> Since you are new to the product, I do however want to point you to some documentation that may be of assistance [...]

Hi @RealSenseSupport according to #6835 it's possible to get intensity as an additional stream. Is this a mistake? And if it's not, how could I get the intensity data? Thank you in advance.