realsenseai/librealsense #7174: L515 Hardware Issue?

jdmickney, 2020-08-24:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                        | L515 } | 
| Firmware Version                    | 01.04.01.02 | 
| Operating System & Version  |  Win 10 and Linux Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |         | 
| Platform                                  | PC  and NVIDIA Jetson Nano.  |
| SDK Version                             |   2.36.<?> }                          | 
| Language                                | | 
| Segment			          |  Robotics/others | 

### Issue Description
I've had the L515 camera for about 5 days now. It's been working fine on:
 - Windows 10 with Realsense Viewwr 2.36 (x64 based machine)
 - Ubuntu 18.04 with Realsense Viewer 2.36 and realsense-ros 2.15 (x64 based machine)
 - Ubuntu 18.04 with Realsense Viewer 2.35.2 (Jetson Nano)

When I installed and ran realsense-ros 1.14 on the Jetson Nano I started receiving hardware errors "Stream stoped" once every second. That's how the error indication is spelled "stoped" not "stopped". The depth image in rviz fluctuated at that same interval. Error output in ROS: "Hardware Notification:Stream stoped. Non recoverable. power reset may help,1.59827e+12,Error,Hardware Error".

I then stopped running ROS and checked out the L500 Depth Sensor in the Realsense Viewers on the Windows 10 and both Ubuntu 18.04 machines and received the same error: when depth sensor is selected a red error box in the upper right-hand corner shows "Stream stoped" about once a second and the depth image in the viewer fluctuates. The RGB Camera and Motion Module options seem to work fine.

USB 3 ports were used at all times and no environmental concerns. I tried reseating usb cord, powered on/off, different resolutions, tried the hardware reset option in the viewer, and the install recommended firmware option in the viewer to no avail. Is there a way to clear this error?

RealSenseSupport, 2020-08-25:
Hi @jdmickney 

Would you mind elaborating a bit on what exactly you're doing/mean by "when depth sensor is selected." We are trying to reproduce and don't see the error you're referring to. 

Thanks

jdmickney, 2020-08-25:
After opening RealSense Viewer 2.36.0 the user is presented with three buttons/switches on the left side to turn on/off. 
These are:
- L500 Depth Sensor
- RGB Camera
- Motion Module
After turning on L500 Depth Sensor the depth stream image is shown in the viewer but fluctuates on and off at a ~1sec interval.

jdmickney, 2020-08-25:
I should have added to my previous post that along with the image fluctuation I receive a red error box in the viewer indicating "Stream stoped" that increments every time error occurs.

RealSenseSupport, 2020-09-26:
May I ask how you installed librealsense/realsense-ros on your jetson?

As an asside....in my experience I have found the most success on non-IA architectures by....

a) cleaning all realsense packages off the system using:
        dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge
b) installing the librealsense SDK packages using apt as outlines here:
        https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md
c) then building realsense-ros in a catkin workspace following the instructions here:
        https://github.com/IntelRealSense/realsense-ros#step-2-install-intel-realsense-ros-from-sources

Thanks

jdmickney, 2020-09-27:
After successfully testing the camera in realsense viewer on a Windows and Ubuntu desktop I followed the guidance provided on:
https://github.com/JetsonHacksNano/installLibrealsense and
https://github.com/JetsonHacksNano/installLibrealsenseROS
The depth topic in rviz on the Jetson Nano initially seemed to work but the errors showed up shortly thereafter. Then the errors followed the device in realsense viewer on even Windows and Ubuntu desktops. I RMA'd the original camera and received a replacement L515 but haven't gotten around to implementing it on the Nano yet (I checked it out in Windows though - works fine). When I do I'll follow your suggested instructions which are greatly appreciated!
Thank you

RealSenseSupport, 2020-10-22:
Hi @jdmickney 

Did you get a chance to try this on your new camera and/or need further assistance?

Thanks

jdmickney, 2020-10-24:
As of late, I haven't been able to get back to my RealSense/Nano project but I don't anticipate needing this thread to remain open. Thank you much for your assistance.

closed 2020-10-24 by jdmickney